theyosh/TerrariumPI #752: Remote buttons (door switches)

theyosh, 2022-12-01:
### Discussed in https://github.com/theyosh/TerrariumPI/discussions/751

<div type='discussions-op-text'>

<sup>Originally posted by **EvilHomerGD** November 27, 2022</sup>
@theyosh what are the chances of adding some additional remote (JSON) functionalities? A little background for you: I have multiple enclosures setup throughout my house (or is it a zoo?). I've been looking at ways to add those remote sensors and come up with a fairly simple solution. I have setup a Pi Pico W to spit out sensor information (temp and humidity) to a JSON page. I've been toying with a reed switch for the doors and would love to be able to read that state into my main TerrariumPi interface. </div>

EvilHomerGD, 2022-12-04:
maybe I'm missing something here but when I add the button, I get "Could not update remote button: int() argument must be a string, a bytes-like object or a number, not 'NoneType'". I have the button address set to "http://10.50.2.176/#door1/isOpen" to traverse the JSON to the value that will be 0 if the door is open and 1 if closed (I can easily swap my logic).

theyosh, 2022-12-04:
Hmm,

Not sure what has changed, but for me it works now. I made a logic error 5 years ago about door statuses, so I have to do some inverse logic juggling... ;) But with you JSON example it does work for me now.

Do a new `git pull` and it should work...

theyosh, 2022-12-04:
And I expect a 0 for door closed in default mode. If you cannot change that, you can use the inverse toggle in the calibration section to change that.

A '0' is converted to closed / not pressed. Any number higher then 0 will be then open / pressed

EvilHomerGD, 2022-12-04:
That seems to have fixed the error. I'll fix my logic to change the way the doors report their open/closed status. That's a simple if/else.

EvilHomerGD, 2022-12-05:
So it would seem that when I did my git pull, it didn't pull down __init__.py which is what was giving me fits (I'm more well versed with subversion so it could be something I did). I manually pushed the updated files and can confirm that everything is working as expect. Now I just need to address my logic which is easy enough. Thanks for the quick turnaround and patience with me.

EvilHomerGD, 2022-12-08:
I would say it is safe to close. It is absolutely working wonderfully for me.

theyosh, 2022-12-09:
Nice, and thanks for testing!

theyosh, 2022-12-09:
Oh, almost forgotten... Could post some pictures at: #210

closed 2022-12-09 by theyosh

EvilHomerGD, 2022-12-09:
I absolutely will! I have to take some new pics since I've added 2 more enclosures since I last posted pics.