theyosh/TerrariumPI #753: Atlas Scientific errorn 22

theyosh, 2022-12-02:
### Discussed in https://github.com/theyosh/TerrariumPI/discussions/749

<div type='discussions-op-text'>

<sup>Originally posted by **Cinett87** November 22, 2022</sup>
getting error sensor [errorn 22] "temp1" at address "66" invalid argument I have also attempted to add it as  0x66, 66,1 and 0x66,1 all have returned invalid argument I have rebooted and restarted many times. 
[image]
[image]
</div>

theyosh, 2022-12-02:
Need to test with: https://github.com/AtlasScientific/Raspberry-Pi-sample-code in order to see if the file `ic2.py` does work with the hardware. So we can try to mimic that.

Cinett87, 2022-12-05:
Absolutly I can help with the testing I will have access to my PI tomorrow I will update as soon as I receive data.

theyosh, 2022-12-05:
Ah, that is nice. I would like to ask if you are able to check out the library https://github.com/AtlasScientific/Raspberry-Pi-sample-code.

So somewhere on your pi do: `git clone https://github.com/AtlasScientific/Raspberry-Pi-sample-code.git`

And try to run the application i2c.py: `python i2c.py` and see what happens. I think you need to add some parameters. Try the List and Poll options. I have found that a Raspberry Pi 4 is some issues. https://github.com/AtlasScientific/Raspberry-Pi-sample-code/issues/9

So do you use a RPI 3 or 4?

Cinett87, 2022-12-06:
I am currently using a PI 4 and the sample code is working for me. it gives me a reading when I prompt it. I notice the code is using the i2c address 102 but not the 0x66 version. attached is showing the code working, I got the reading based on the datasheets code by typing "r" and it returned a value.  the attached pics are the i2c addresses from atlas as well as the screenshot of it working
[image]
[image]

Cinett87, 2022-12-06:
the commands used to access the readings of change the temp scale is found here mind you this is for the temp (EZO-RTD) chip only each chip might have different command codes. https://www.atlas-scientific.com/files/EZO_RTD_Datasheet.pdf

Cinett87, 2022-12-06:
[image]

theyosh, 2022-12-06:
I think I found my error. Could you do a `git pull` and restart TP4?

Cinett87, 2022-12-06:
yes but because I am very beginner whats the full git pull command I need

Cinett87, 2022-12-07:
nevermind had to cd to terrariumpi then git pull

Cinett87, 2022-12-07:
Still receiving errorno 22

theyosh, 2022-12-08:
Hmm, are you sure? First make sure TerrariumPI is stopped: `sudo service terrariumpi stop`

The following actions needs to be done in the `/home/pi/TerrariumPI` folder.

1. `git status` -> It should say on the first line: 'On branch 4.x.y.z'
2. If not, then try `git checkout 4.x.y.z` -> if this fails... we are back at the beginning. If you are on the 4.x.y.z branch, continue to 3
3. `git pull` -> Should update some files again
4. `source venv/bin/activate` -> This should load the python3 environment. You should see '(venv)' in front of your prompt.
5. `python terrariumPI.py` -> This should start TerrariumPI in debug mode.

During the startup, you should see a message about: 'Loading sensor at address: ' with a number. That number should be 102.

When TerrariumPI is started, and the number is 102, you should see some values in the graph coming in.

And then press `Ctrl+C` in the console to stop TerrariumPI.

So, now, it comes down if you get the 102 address number, and if it will read the sensor.... Else I have to think a bit more what could be the issue.

theyosh, 2022-12-08:
Ah crap, I was confused with another issue... Ok, here are the steps to get on the 4.x.y.z branch:

1. `git remote set-branches origin '*'`
2. `git fetch`
3. `git checkout 4.x.y.z`

Cinett87, 2022-12-08:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-08 16:37:57,393 - INFO    - terrariumEngine       - Starting TerrariumPI                                      4.5.0 ...
2022-12-08 16:37:57,449 - INFO    - terrariumEngine       - Loaded 30 settings i                                     n 0.05 seconds.
2022-12-08 16:37:57,483 - INFO    - terrariumEngine       - Loading weather data                                      from source https://api.openweathermap.org/data/2.5/weather?q=Hawaii,us&APPID=3                                     50a2ae63d7c6ba46baafbcaa5c27c3c
2022-12-08 16:37:58,870 - INFO    - weather.openweathermap_org_weather - Using O                                     penweathermap Free API
2022-12-08 16:37:58,871 - INFO    - weather               - Loaded new weather d                                     ata in 1.384 seconds.
2022-12-08 16:37:58,871 - INFO    - terrariumEngine       - Loading Meross cloud                                      connection.
2022-12-08 16:37:58,874 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:37:59,157 - ERROR   - terrariumCloud        - You have issued too                                      many tokens without logging out and your account might have been temporarly disa                                     bled.
2022-12-08 16:37:59,171 - INFO    - terrariumCloud        - Stopping Meross clou                                     d ...
2022-12-08 16:37:59,875 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:00,875 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:01,876 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:02,876 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:03,877 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:04,877 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:05,878 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:06,878 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:07,879 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:08,879 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:09,880 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:10,880 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:11,881 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:12,881 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:13,882 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:14,882 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:15,883 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:16,883 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:17,884 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:18,884 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:19,885 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:20,885 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:21,886 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:22,886 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:23,887 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:24,887 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:25,888 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:26,888 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
2022-12-08 16:38:27,889 - INFO    - terrariumCloud        - Waiting for Meross c                                     loud connection ...
[... cut ...]
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:
(venv) pi@raspberrypi:~/TerrariumPI $

Cinett87, 2022-12-08:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-08 16:41:23,948 - INFO    - terrariumEngine       - Starting TerrariumPI 4.5.0 ...
2022-12-08 16:41:24,002 - INFO    - terrariumEngine       - Loaded 30 settings in 0.05 seconds.
2022-12-08 16:41:24,035 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Hawaii,us&APPID=350a2ae63d7c6ba46baafbcaa5c27c3c
2022-12-08 16:41:25,662 - INFO    - weather.openweathermap_org_weather - Using Openweathermap Free API
2022-12-08 16:41:25,664 - INFO    - weather               - Loaded new weather data in 1.625 seconds.
2022-12-08 16:41:25,665 - INFO    - terrariumEngine       - Loading Meross cloud connection.
2022-12-08 16:41:25,671 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:25,976 - ERROR   - terrariumCloud        - You have issued too many tokens without logging out and your account might have been temporarly disabled.
2022-12-08 16:41:25,990 - INFO    - terrariumCloud        - Stopping Meross cloud ...
2022-12-08 16:41:26,672 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:27,673 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:28,673 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:29,674 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:30,674 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:31,675 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:32,675 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:33,676 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:34,676 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:35,677 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:36,677 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:37,678 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:38,678 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:39,679 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:40,679 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:41,680 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:42,680 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:43,681 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:44,681 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:45,682 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:46,682 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:47,683 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:48,683 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:49,684 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:50,684 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:51,685 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:52,685 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:53,686 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:54,686 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
2022-12-08 16:41:55,687 - ERROR   - terrariumCloud        - Could not login to Meross cloud within 30 seconds. Restarting...
2022-12-08 16:41:55,688 - WARNING - terrariumCloud        - Reconnecting to Meross cloud. Somehow the connection was lost ...
2022-12-08 16:41:55,689 - INFO    - terrariumCloud        - Stopping Meross cloud ...

Cinett87, 2022-12-08:
it won't progress after the stopping meross cloud

theyosh, 2022-12-08:
Yeah, somehow something went wrong with the logins to Meross. Mostly it will fix after 24 hours :(

> 2022-12-08 16:41:25,976 - ERROR - terrariumCloud - You have issued too many tokens without logging out and your account might have been temporarly disabled.

theyosh, 2022-12-08:
So sorry, for now, there is not a quick fix but to wait till the Meross logins start to work again.

Cinett87, 2022-12-09:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-08 22:18:39,822 - INFO    - terrariumEngine       - Starting TerrariumPI                                                                          4.5.0 ...
2022-12-08 22:18:39,876 - INFO    - terrariumEngine       - Loaded 30 settings i                                                                         n 0.05 seconds.
2022-12-08 22:18:39,909 - INFO    - terrariumEngine       - Loading weather data                                                                          from source https://api.openweathermap.org/data/2.5/weather?q=Hawaii,us&APPID=3                                                                         50a2ae63d7c6ba46baafbcaa5c27c3c
2022-12-08 22:18:41,549 - INFO    - weather.openweathermap_org_weather - Using O                                                                         penweathermap Free API
2022-12-08 22:18:41,551 - INFO    - weather               - Loaded new weather d                                                                         ata in 1.637 seconds.
2022-12-08 22:18:41,551 - INFO    - terrariumEngine       - Loading Meross cloud                                                                          connection.
2022-12-08 22:18:41,555 - INFO    - terrariumCloud        - Waiting for Meross c                                                                         loud connection ...
2022-12-08 22:18:42,556 - INFO    - terrariumCloud        - Waiting for Meross c                                                                         loud connection ...
2022-12-08 22:18:43,557 - INFO    - terrariumCloud        - Meross cloud is conn                                                                         ected! Found 1 devices in 2.00 seconds.
2022-12-08 22:18:44,081 - INFO    - terrariumEngine       - Loading existing sen                                                                         sors from database.
2022-12-08 22:18:44,087 - INFO    - terrariumEngine       - Scanning for new sen                                                                         sors ...
2022-12-08 22:18:53,538 - INFO    - terrariumEngine       - Loaded 0 sensors in 9.46 seconds.
2022-12-08 22:18:53,539 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-12-08 22:18:54,028 - INFO    - terrariumEngine       - Loaded relay meross relay named 'Lights' at address '220                                     6011978162254060248e1e994f197,1' value 0.00 in 0.46 seconds.
2022-12-08 22:18:54,096 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2206011                                     978162254060248e1e994f197 [False, False, False, False, False, False]
2022-12-08 22:18:54,200 - INFO    - terrariumEngine       - Loaded relay meross relay named 'Channel Switch 2' at ad                                     dress '2206011978162254060248e1e994f197,2' value 0.00 in 0.17 seconds.
2022-12-08 22:18:54,375 - INFO    - terrariumEngine       - Loaded relay meross relay named 'Channel Switch 3' at ad                                     dress '2206011978162254060248e1e994f197,3' value 0.00 in 0.17 seconds.
2022-12-08 22:18:54,448 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2206011                                     978162254060248e1e994f197 [False, False, False, False, False, False]
2022-12-08 22:18:54,552 - INFO    - terrariumEngine       - Loaded relay meross relay named 'Heating pad' at address                                      '2206011978162254060248e1e994f197,4' value 0.00 in 0.18 seconds.
2022-12-08 22:18:54,617 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2206011                                     978162254060248e1e994f197 [False, False, False, False, False, False]
2022-12-08 22:18:54,726 - INFO    - terrariumEngine       - Loaded relay meross relay named 'Channel USB' at address                                      '2206011978162254060248e1e994f197,5' value 0.00 in 0.17 seconds.
2022-12-08 22:18:54,732 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-12-08 22:18:55,785 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2206011                                     978162254060248e1e994f197 [False, False, False, False, False, False]
2022-12-08 22:18:56,067 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2206011                                     978162254060248e1e994f197 [False, False, False, False, False, False]
2022-12-08 22:19:05,803 - INFO    - terrariumEngine       - Loaded 5 relays in 12.26 seconds.
2022-12-08 22:19:05,803 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-12-08 22:19:05,815 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.01 seconds.
2022-12-08 22:19:05,816 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-12-08 22:19:05,825 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2022-12-08 22:19:05,826 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2022-12-08 22:19:05,841 - INFO    - terrariumArea         - Refreshing timer table for Lighting named Lighting. peri                                     od: day
[... cut ...]
2022-12-08 22:19:05,844 - INFO    - terrariumEngine       - Loaded Enclosure Vanilla orchid Terrarium with 2 areas i                                     n 0.01 seconds.
2022-12-08 22:19:05,845 - INFO    - terrariumEngine       - Loaded 1 enclosures in 0.02 seconds.
2022-12-08 22:19:05,922 - INFO    - terrariumEngine       - TerrariumPI is up and running at address: http://0.0.0.0                                     :8090 in 26.81 seconds.
2022-12-08 22:19:05,000 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090

Cinett87, 2022-12-09:
alright it ran but it is not collecting the sensor attached

theyosh, 2022-12-09:
I get the feeling you did not switch banches. So I added the code in the main branch.

So. if you do now a `git pull` and try again to manually start it, I think it should work.

theyosh, 2022-12-09:
But you did not add the sensor to the system. 

>  2022-12-08 22:18:53,538 - INFO - terrariumEngine - Loaded 0 sensors in 9.46 seconds.

You need to add the sensor first

Cinett87, 2022-12-09:
I did switch the branch and can show you the status that its in that branch, where am I supposed to add the sensor? cause I did on the dashboard and it said no data

Cinett87, 2022-12-09:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-09 14:54:04,084 - INFO    - terrariumEngine       - Starting TerrariumPI                                                                               4.5.0 ...
2022-12-09 14:54:04,139 - INFO    - terrariumEngine       - Loaded 30 settings i                                                                              n 0.05 seconds.
2022-12-09 14:54:04,173 - INFO    - terrariumEngine       - Loading weather data                                                                               from source https://api.openweathermap.org/data/2.5/weather?q=Hawaii,us&APPID=3                                                                              50a2ae63d7c6ba46baafbcaa5c27c3c
2022-12-09 14:54:05,776 - INFO    - weather.openweathermap_org_weather - Using O                                                                              penweathermap Free API
2022-12-09 14:54:05,778 - INFO    - weather               - Loaded new weather d                                                                              ata in 1.601 seconds.
2022-12-09 14:54:05,778 - INFO    - terrariumEngine       - Loading Meross cloud                                                                               connection.
2022-12-09 14:54:05,783 - INFO    - terrariumCloud        - Waiting for Meross c                                                                              loud connection ...
2022-12-09 14:54:06,783 - INFO    - terrariumCloud        - Waiting for Meross c                                                                              loud connection ...
2022-12-09 14:54:07,785 - INFO    - terrariumCloud        - Meross cloud is conn                                                                              ected! Found 1 devices in 2.00 seconds.
2022-12-09 14:54:08,298 - INFO    - terrariumEngine       - Loading existing sen                                                                              sors from database.

===========================================
Loading sensor at address: 258
=============================================

2022-12-09 14:54:08,307 - WARNING - hardware.sensor       - Unable to load senso                                                                              r AtlasScientific I2C sensor temperature named 'Temp1' at address '102': [Errno                                                                               22] Invalid argument, retrying in 0.5 seconds...

**===========================================
Loading sensor at address: 258
=============================================

2022-12-09 14:54:08,812 - WARNING - hardware.sensor       - Unable to load senso                                                                              r AtlasScientific I2C sensor temperature named 'Temp1' at address '102': [Errno                                                                               22] Invalid argument, retrying in 0.5 seconds...

===========================================
Loading sensor at address: 258
=============================================

2022-12-09 14:54:09,317 - ERROR   - terrariumEngine       - Error loading atlass                                                                              cientific temperature named 'Temp1' at address '102' with error: Unable to load                                                                               sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '102': [E                                                                              rrno 22] Invalid argument.
2022-12-09 14:54:09,319 - INFO    - terrariumEngine       - Scanning for new sen                                                                              sors ...
2022-12-09 14:54:18,796 - INFO    - terrariumEngine       - Loaded 0 sensors in                                                                               10.50 seconds.**
2022-12-09 14:54:18,797 - INFO    - terrariumEngine       - Loading existing rel                                                                              ays from database.
2022-12-09 14:54:19,341 - INFO    - terrariumEngine       - Loaded relay meross                                                                               relay named 'Lights' at address '2206011978162254060248e1e994f197,1' value 100.0                                                                              0 in 0.52 seconds.
2022-12-09 14:54:19,379 - INFO    - terrariumCloud        - Got an update from t                                                                              he Meross Cloud. Relay state 2206011978162254060248e1e994f197 [True, True, False                                                                              , False, False, False]
2022-12-09 14:54:19,492 - INFO    - terrariumEngine       - Loaded relay meross                                                                               relay named 'Channel Switch 2' at address '2206011978162254060248e1e994f197,2' v                                                                              alue 0.00 in 0.15 seconds.
2022-12-09 14:54:19,671 - INFO    - terrariumEngine       - Loaded relay meross                                                                               relay named 'Channel Switch 3' at address '2206011978162254060248e1e994f197,3' v                                                                              alue 0.00 in 0.18 seconds.
[... cut ...]
2022-12-09 14:55:16,039 - INFO    - terrariumEngine       - Stopped Meross power switches relay named 'Channel USB' at address '2206011978162254060248e1e994f197,5'
2022-12-09 14:55:16,040 - INFO    - terrariumCloud        - Stopping Meross cloud ...
2022-12-09 14:55:16,477 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2206011978162254060248e1e994f197 [True, True, False, False, False, False]
2022-12-09 14:55:16,777 - INFO    - terrariumCloud        - Closed Meross cloud connection
2022-12-09 14:55:16,779 - INFO    - terrariumEngine       - Stopped TerrariumPI 4.5.0 after running for 0:01:13. Bye bye.

theyosh, 2022-12-09:
Ok, now, you have entered the address 102 in the form. That is not correct. You have to enter the address `0x66` which the software will translate to 102. So, could you try to enter the address 0x66?

Cinett87, 2022-12-09:
when I entered 0x66 it autosaved it as zero and won't accept 0x66 it says can't be saved

Cinett87, 2022-12-09:
its says error sensor cant be saved is there a way to save the info cause at this point I have a bunch of sensors saved but none in my list so it also gives me a unique ID error

Cinett87, 2022-12-09:
[image]

Cinett87, 2022-12-09:
at this point would it be best to just reset my pi and start from scratch since it will then download all current

theyosh, 2022-12-09:
Wait!

Cinett87, 2022-12-09:
okay

theyosh, 2022-12-09:
You can 'reset' your setup just by deleting the database files.

`rm data/terrariumpi*`

and start TP4 in debug mode again. That would give you a clean setup again

Cinett87, 2022-12-09:
do I add the sensor via the command line or via the dashboard sorry super PI noob here.
 also do I `cd TerrariumPI` inorder to do the rm command

theyosh, 2022-12-09:
Euh, the command needs to be done in the TerrariumPI folder. So yes, do first a `cd TerrariumPI`. This is what I call the default location to start working or doing commands. So if I do not mention that, than try to keep that in mind.

And you should just be adding the sensor through the web gui. You need to login into the gui. Default login is user 'admin' and password is 'password'. And then use the menu on the left, and use the add sensor menu item.

Just to be complete, the valid address is officially `0x66` which I advise to use. But just 66 should also work. But by adding '0x' in front. you (can) know that it is an I2C device. If you need to debug it on the console with the `i2cdetect -y 1` command.

Cinett87, 2022-12-09:
everytime I use 0x66 (zeroXsixsix) it auto changes the value to 0 (zero) so I am really not sure what to do

Cinett87, 2022-12-09:
[image]

theyosh, 2022-12-09:
Oh, that is strange. I need to look into that. But that will than tomorrow. As it is late over here.

theyosh, 2022-12-09:
And thanks for testing! We will get to work!

Cinett87, 2022-12-09:
okay 66 worked. here is the lastest

Cinett87, 2022-12-09:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-09 16:19:13,542 - INFO    - terrariumEngine       - Starting TerrariumPI 4.5.0 ...
2022-12-09 16:19:13,598 - INFO    - terrariumEngine       - Loaded 29 settings in 0.05 seconds.
2022-12-09 16:19:14,122 - INFO    - terrariumEngine       - Loading existing sensors from database.

===========================================
Loading sensor at address: 102
=============================================

Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 125, in __init__
    self.__load_existing_sensors()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 505, in __load_existing_sensors
    value = self.sensors[sensor.id].update()
  File "/home/pi/TerrariumPI/hardware/sensor/__init__.py", line 312, in update
    current = None if data is None or self.sensor_type not in data else data[self.sensor_type]
TypeError: argument of type 'float' is not iterable

Cinett87, 2022-12-09:
sleep well

Cinett87, 2022-12-10:
(venv) pi@raspberrypi:~/TerrariumPI $ git pull
remote: Enumerating objects: 37, done.
remote: Counting objects: 100% (36/36), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 20 (delta 16), reused 18 (delta 14), pack-reused 0
Unpacking objects: 100% (20/20), done.
From https://github.com/theyosh/TerrariumPI
   c91b2c5d..dea83739  4.x.y.z    -> origin/4.x.y.z
   e19bddb5..7515145a  main       -> origin/main
Updating c91b2c5d..dea83739
Fast-forward
 gui/pages/Dashboard.svelte                |   2 +-
 hardware/sensor/atlasscientific_sensor.py |   5 ++---
 public/main.js                            |   2 +-
 public/main.js.gz                         | Bin 708853 -> 708855 bytes
 4 files changed, 4 insertions(+), 5 deletions(-)
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-09 18:41:01,392 - INFO    - terrariumEngine       - Starting TerrariumPI 4.5.0 ...
2022-12-09 18:41:01,446 - INFO    - terrariumEngine       - Loaded 29 settings in 0.05 seconds.
2022-12-09 18:41:01,966 - INFO    - terrariumEngine       - Loading existing sensors from database.

===========================================
Loading sensor at address: 102
=============================================

2022-12-09 18:41:03,481 - ERROR   - hardware.sensor       - Sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66' has exception: could not convert string to float: '24.902\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2022-12-09 18:41:03,488 - WARNING - hardware.sensor       - Invalid reading from sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66', retrying in 0.5 seconds...
2022-12-09 18:41:05,496 - ERROR   - hardware.sensor       - Sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66' has exception: could not convert string to float: '24.893\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2022-12-09 18:41:05,497 - WARNING - hardware.sensor       - Invalid reading from sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66', retrying in 0.5 seconds...
2022-12-09 18:41:07,504 - ERROR   - hardware.sensor       - Sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66' has exception: could not convert string to float: '24.888\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2022-12-09 18:41:07,506 - ERROR   - hardware.sensor       - Error updating sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66'. Check your hardware! Invalid reading from sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66'
2022-12-09 18:41:07,508 - WARNING - terrariumEngine       - atlasscientific temperature named 'Temp1' at address '66' had problems reading a new value during startup in 5.54 seconds. Will be updated in the next round.
2022-12-09 18:41:07,509 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-12-09 18:41:16,724 - INFO    - terrariumEngine       - Loaded 1 sensors in 14.76 seconds.
2022-12-09 18:41:16,725 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-12-09 18:41:16,740 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-12-09 18:41:27,811 - INFO    - terrariumEngine       - Loaded 0 relays in 11.09 seconds.
2022-12-09 18:41:27,812 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-12-09 18:41:27,827 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.01 seconds.
2022-12-09 18:41:27,827 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-12-09 18:41:27,839 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2022-12-09 18:41:27,839 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2022-12-09 18:41:27,846 - INFO    - terrariumEngine       - Loaded 0 enclosures in 0.01 seconds.
2022-12-09 18:41:27,905 - INFO    - terrariumEngine       - TerrariumPI is up and running at address: http://0.0.0.0:8090 in 27.23 seconds.
2022-12-09 18:41:27,000 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090
2022-12-09 18:41:28,188 - WARNING - terrariumEngine       - Could not take a new measurement from sensor atlasscientific temperature named 'Temp1' at address '66'. Tried for 0.00 seconds. Skipping this update.
2022-12-09 18:41:59,690 - ERROR   - hardware.sensor       - Sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66' has exception: could not convert string to float: '24.887\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2022-12-09 18:41:59,693 - WARNING - hardware.sensor       - Invalid reading from sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66', retrying in 0.5 seconds...
2022-12-09 18:42:01,701 - ERROR   - hardware.sensor       - Sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66' has exception: could not convert string to float: '24.884\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2022-12-09 18:42:01,703 - WARNING - hardware.sensor       - Invalid reading from sensor AtlasScientific I2C sensor temperature named 'Temp1' at address '66', retrying in 0.5 seconds...
[... cut ...]
2022-12-09 18:42:09,472 - INFO    - terrariumEngine       - Stopping TerrariumPI 4.5.0 ...
2022-12-09 18:42:09,473 - INFO    - terrariumEngine       - Stopped main engine thread
2022-12-09 18:42:09,475 - INFO    - terrariumEngine       - Stopped log tailing.
2022-12-09 18:42:09,476 - INFO    - terrariumEngine       - Stopped AtlasScientific I2C sensor temperature named 'Temp1' at address '66'
2022-12-09 18:42:09,477 - INFO    - terrariumEngine       - Stopped TerrariumPI 4.5.0 after running for 0:01:08. Bye bye.

theyosh, 2022-12-10:
I think and hope with the last code update it should work... Do a `git pull` and try again in debug mode.

Cinett87, 2022-12-10:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-12-10 02:53:56,224 - INFO    - terrariumEngine       - Starting TerrariumPI                                                 4.5.0 ...
2022-12-10 02:53:56,279 - INFO    - terrariumEngine       - Loaded 29 settings i                                                n 0.05 seconds.
2022-12-10 02:53:56,793 - INFO    - terrariumEngine       - Loading existing sen                                                sors from database.

===========================================
Loading sensor at address: 102
=============================================

2022-12-10 02:53:58,333 - INFO    - terrariumEngine       - Loaded sensor atlass                                                cientific temperature named 'Temp1' at address '66' with value 24.48C in 1.53 se                                                conds.
2022-12-10 02:53:58,337 - INFO    - terrariumEngine       - Scanning for new sen                                                sors ...
2022-12-10 02:54:07,551 - INFO    - terrariumEngine       - Loaded 1 sensors in                                                 10.76 seconds.
2022-12-10 02:54:07,552 - INFO    - terrariumEngine       - Loading existing rel                                                ays from database.
2022-12-10 02:54:07,566 - INFO    - terrariumEngine       - Scanning for new rel                                                ays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-12-10 02:54:18,639 - INFO    - terrariumEngine       - Loaded 0 relays in 1                                                1.09 seconds.
2022-12-10 02:54:18,640 - INFO    - terrariumEngine       - Loading existing but                                                tons from database.
2022-12-10 02:54:18,656 - INFO    - terrariumEngine       - Loaded 0 buttons in                                                 0.02 seconds.
2022-12-10 02:54:18,656 - INFO    - terrariumEngine       - Loading existing web                                                cams from database.
2022-12-10 02:54:18,667 - INFO    - terrariumEngine       - Loaded 0 webcams in                                                 0.01 seconds.
2022-12-10 02:54:18,668 - INFO    - terrariumEngine       - Loading existing enc                                                losures from database.
2022-12-10 02:54:18,675 - INFO    - terrariumEngine       - Loaded 0 enclosures                                                 in 0.01 seconds.
2022-12-10 02:54:18,736 - INFO    - terrariumEngine       - TerrariumPI is up an                                                d running at address: http://0.0.0.0:8090 in 23.22 seconds.
2022-12-10 02:54:18,000 - INFO    - terrariumWebserver    - Running webserver at                                                 0.0.0.0:8090

Cinett87, 2022-12-10:
its taking readings I think we have it fixed.

Cinett87, 2022-12-10:
by we I mean you!!! rock on!! thank you for working on this glad I could help

theyosh, 2022-12-10:
Nice! Thanks for testing! And enjoy TerrariumPI

theyosh, 2022-12-10:
FYI: I found the bug that made 0x66 became 0. So that will also be fixed in the next release probably tomorrow.

closed 2022-12-11 by theyosh

theyosh, 2022-12-11:
I just released 4.6.0. So that means for you that you can switch back to the main branch and have the same functionality. This is also better for the future as new released will be done in the main branch.

So in the folder `TerrariumPI` do the following commands:

1. `git checkout main`

And then follow the steps at: https://github.com/theyosh/TerrariumPI/releases/tag/4.6.0

Thanks for testing.

commit 13441a99 (2022-12-09): Readout fix #753
commit 7515145a (2022-12-09): Readout fix #753