theyosh/TerrariumPI #754: [FEATURE]: Discussion platform

1ucket, 2022-12-12:
**Is your feature request related to a problem? Please describe.**
I'm completely new to TerrariumPI amd have lots and lots of questions. I couldn't find any discussion platform to ask them (creating a github issue for each might be a bit over the top). Is there already one? Then ignore below.

**Describe the solution you'd like**
A platform to discuss TerrariumPI and get help. My suggestion would be a Discord server with theme specific channels (e.g. software help, electronics help, general discussions, enclosure sharing and SW development). But I'd also be totally fine with a subreddit, custom forum or whatever.
If you want any support in setting something up I'd be willing to help.

**Additional context**
Add any other context or screenshots about the feature request here.

theophile, 2022-12-19:
Have you visited here? https://github.com/theyosh/TerrariumPI/discussions

theyosh, 2022-12-19:
@1ucket found the Mattix chat and that solved it. The discussions part is also not very active. But for now I have added some extra links in de readme to the matrix chat and the documentation page with multiple contact options.

That should be enough. Supporting another discord service or forum like systems is more work then the gives back.... :(

closed 2022-12-19 by theyosh