theyosh/TerrariumPI #758: [BUG]: Notification Setup

sni89, 2022-12-15:
**Setup:**
 - TerrariumPI version: 4.6.0
 - Python version: [e.g. 2 or 3]
 - Raspberry PI: 4

**Describe the bug**
It’s not possible to setup A new display or other notification due the setup area itself is missing 

**To Reproduce / Installation steps**
Went to notifications, wanted to setup a display. The setup area directly leads me to notification settings not to the setup itself. Ends with an error message that setup is missing. 

**Expected behavior**
Add of a display and messages on it 🙂

**Screenshots**

[image]

theyosh, 2022-12-15:
Hmm, this I need to look into it. But a bit of warning, the display functionality is not that good :(. It is difficult to make it all variable on a small screen.

So most people will end up creating their own script to display information in the way they want.

Nevertheless I will look this issue.

theyosh, 2022-12-16:
Ok, found the problem. Do a `git pull` and reload the GUI. That should fix this problem.

In the screenshot I miss some fields. This is due to a javascript error. That should now be fixed. For a display, you need to specify the hardware type and the I2C address.

theyosh, 2022-12-22:
Did you had time to test the fix? For me it is working.

sni89, 2022-12-23:
At least I have options like address and displaytype back. But when I want to set up the message it’s trapped in loading the page. I see the page behind the loading symbol but it won’t let me go on. This could be a „mobile user“ problem did not try on a desktop so far. 

[image]

theyosh, 2022-12-27:
Hmm, this is now fixed. Do a `git pull` and restart TP4. Than it should work again.

closed 2022-12-30 by sni89