theyosh/TerrariumPI #759: [BUG]:

chrRtg, 2022-12-15:
**More info**
Please look first at the [FAQ](https://github.com/theyosh/TerrariumPI/wiki/FAQ) or [Known Issues](https://github.com/theyosh/TerrariumPI/wiki/Known-issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.6.0
 - Python version: 2.7
 - Raspberry PI: 3B

**Describe the bug**
Webserver not running. 

I updated from 4.4 to 4.6. Afterwards the webserver on port 8090 were no longer responding. 
netstat -pln does not indicate any service listening on that port. 

Checked all 3 logfiles in ./log but it looks like if TerrariumPI is still polling data and works as expected. Lights in the Terrarium switch on / off according to the settings.

Access by ssh works fine without issues.

While I had some issues 1 1/2 years ago with TerrariumPI I additionally ran fix_db.py to be sure it's not again my database.

Any suggestion?

theyosh, 2022-12-15:
Oh, this is strange..

What you could try is to manual debug it. The steps are described here: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

At a certain point there should be a message that the webserver is running at ip 0.0.0.0 at port 8090. If that is the case it should all be good. If not, then we have to look what log you get on the screen in debug mode.

chrRtg, 2022-12-16:
Debug told my problem with data in terrariumArea.py. The application hung in the startup.

Finally I deleted both entries from Area table in my database. 
Afterwards TerrariumPy started but I was not able to enter any entries to the Area, see screemshots:

[image]
[image]
[image]

I'll roll back now to 4.5.

theyosh, 2022-12-16:
I am just working on these bugs :(. The problem is that some fields are staying empty, which is somehow valid, but should contain the number 0.

If you have about 1 hour time left... wait.. I think I am close to a fix.

chrRtg, 2022-12-16:
Whow, OK :)

theyosh, 2022-12-16:
So, I think it should be fixed now. Do a `git pull` and start TerrariumPI in debug mode. It should start, and you should be able to add the areas again.

chrRtg, 2022-12-16:
Update went well. No problems visible in the console but frontend gave me an error. I'm about to add this area as my first one.

[image]

theyosh, 2022-12-16:
Did you reload the interface holding the shift? Because the gui code is also changed. So also reload the gui holding the shift key

chrRtg, 2022-12-16:
Yes I did. Double checked with Dev Console and Network log. Also checked with a fresh incognito browser.

git pull from master was successful, saw a lot of file been update, in particular the .js files. But it remains 4.6.0, right?

theyosh, 2022-12-16:
Nice!

And yes, this is still version 4.6.0 But I am thinking of releasing a new version this weekend due to all the bugs that were found. :(

chrRtg, 2022-12-16:
OK, but my problem remain as seen above in the screenshot.

theyosh, 2022-12-16:
Strange, as I am able to add a new light area based on a timer:

[image]

I hope you can figure out the Dutch ;)

So I do not understand why you can add your area... Is the error message exactly the same?

chrRtg, 2022-12-16:
no problem with dutch, at least if written :)

The error message remain the same.

On network level :
Request URL: http://agame:8090/api/areas/
Request Method: POST
Status Code: 500 Internal Server Error

with a payload of 
[image]

response is: {"message": "Area could not be added. could not convert string to float: "}

I'm running the log files in DEBUG with a tail -f on the logfiles. Nothing appears there except the err 500 from the access log.

theyosh, 2022-12-16:
There are some differences with the post data. So I am not sure if you are on the latest code.

[image]

But I can just add a new light area with a timer.

So, do another `git pull` and then there should be a line line:

```
From github.com:theyosh/TerrariumPI
   0131f170..a4a9b004  main       -> origin/main
```

After the git pull make sure you restart TerrariumPI. Either with manual debug or through the service. And then load the GUI, making sure you do not cache the old GUI. Holding the shift while reloading works for me (Firefox)

chrRtg, 2022-12-16:
for me the pull looks like this:
From https://github.com/theyosh/TerrariumPI
   6d697656..91abf9d7  4.x.y.z    -> origin/4.x.y.z
Updating 6d697656..91abf9d7
Fast-forward
 gui/locale/ca.json                |  22 +++++++++++-----------
 gui/modals/WebcamFormModal.svelte |   4 ++++
 public/main.js                    |  12 ++++++------
 public/main.js.gz                 | Bin 707577 -> 707685 bytes
 terrariumArea.py                  |  35 ++++++++++++++++++++---------------
 terrariumEngine.py                |  12 +++++++-----
 6 files changed, 48 insertions(+), 37 deletions(-)

But more the less it's working now. 

After I added an area I had to wait roughly a minute until it appears in enclosures/ or the area view on the dashboard.
Changing a area from a day/night to a timer does not work. Instead I had to delete the area and create it from the scratch.

Usually our terrarium run on day/night mode all over the year. But in the winter we have to offer our bearded dragons a hibernation. To initiate the hibernation we have to decrease the time we offer heat and light day by day. 

From my perspective we may close the issue. From what I understood the migration to svelte brought a lot of improvement (and a new technology stack for you) but some challenges as well ;)

Lot of thanks for your outstanding support!

closed 2022-12-16 by chrRtg

theyosh, 2022-12-16:
You are on the 4.x.y.z branch. That was not updated. I think I just synced both branches a few minutes ago..

theyosh, 2022-12-16:
Hmm or not... not sure..  `git status` should show:

```
On branch main
Your branch is up to date with 'origin/main'.
```

chrRtg, 2022-12-16:
OK, did a "git reset --hard main", git pull, restart service and loaded the interface. I won't perform additional tests, instead fetch some food and get some rest :)