theyosh/TerrariumPI #760: [BUG]: Relay dials misaligned at certain screen widths

theophile, 2022-12-17:
At certain screen widths, the numerical text inside the dial on the relays screen is misaligned, showing up underneath the row. I spent a long time trying to figure out why, assuming code modifications I made on my system were probably to blame. But then I noticed that the exact same thing happens on your installation at https://terrarium.theyosh.nl/#/relays/

Here are two screenshots. Top one looks like it should but the bottom one shows the issue:
[image]

[image]

As you can see, the presence of the Android nav bar on the second one narrows the screen width enough to trigger the issue. The same thing happens on Chrome. I haven't tested exhaustively but I have noticed that you can see the bug show up in Chrome developer tools if you set the device to responsive and switch back and forth between screen widths of 800px and 799px.

The most I've been able to figure out is that whatever is calculating the positioning of the "input" class object is suddenly applying a style rule of "top: 181px" when the bug occurs, whereas that value is 0px normally. I don't know enough about the UI framework to dig deeper than that.

theyosh, 2022-12-19:
Damm, was hoping nobody would see it ;). But somehow it does not play nice on some resolutions. This will take some more time to figure out. As I am more a backend developer then frontend....

theophile, 2022-12-19:
I'm sort of in the same boat, except that I am really only a hobbyist. For the life of me I can't figure out where or how the positioning of these elements occurs.

theophile, 2022-12-27:
Can confirm this fixed the issue I was seeing. Thanks!

closed 2022-12-27 by theophile