theyosh/TerrariumPI #764: [BUG]: Dimming not working

Jexkin, 2022-12-28:
**More info**
Hey, this morning I noticed that my LED lights were fully on after about a minute or two. (Seems like it did the same last night as well)
Normally they dimm to 100% (or 0%) within 80 minutes just like I set it up in the Area settings.
A few days earlier it still did dimm just like it should so I'm not sure why there's a problem now, I already checked my hardware.

**Setup:**
 - TerrariumPI version: 4.6.0
 - Raspberry PI: 4
 - LED Warrior 18 Dimmer

**Describe the bug**
LED lights are dimming to 100% (or 0%) within a minute or so, check picture below.

**Expected behavior**
Lights slowly dimm from 0% to 100% (and vice versa) within 80 minutes.

**Screenshots**
[image]

**Additional context**
Just a side note: When clicking the "?" in the Area settings, it says that the max duration for the "Dimmer duration" is 180 minutes but the sliders only go to 100 minutes.

theyosh, 2022-12-28:
Hi, there was a bug with the dimmer duration. Could you do a `git pull` to make sure you have the latest version. And then I think it should work again after a restart of TerrariumPI.

About the 100 min. vs. 180 min. I have to look into it...

Jexkin, 2022-12-28:
Did a `git pull` and restarted TPi. Will give you an update in about 8 hours when the lights should dimm down :)

theyosh, 2022-12-28:
Haha, I thought already of that :). So lets see if it works like expected tomorrow.

theyosh, 2022-12-28:
Now I have also fixed the GUI. If you do another `git pull` but than just reload your GUI, you should be able to delay till 180 minutes again if needed.

Jexkin, 2022-12-28:
Did another `git pull` but it seems like that fix didn't work. Also did another restart just to be sure but that didn't do anything either
[image]
Created another enclosure and area just to see if my existing enclosure/area was causing the fix not to work but seems like that's not the case.

theyosh, 2022-12-28:
Than you did not refresh the GUI. Make sure you hold the shift key while reloading. I just tested and I have 180 minutes.

Jexkin, 2022-12-28:
Yep, I just found the issue on my side. I never switched back to the main branch after the LED Warrior 18 testings.
Works now, thanks!
Lights should dimm down in roughly an hour so I'll update later on that one.

theyosh, 2022-12-29:
And still working? Than we can close this issue?

Jexkin, 2022-12-29:
Yes, thank you!

closed 2022-12-29 by theyosh