theyosh/TerrariumPI #765: [BUG]: Save Error - Add Relay

ewiandr73, 2023-01-02:
G'day,

Happy New Year....

**Setup:**
 - TerrariumPI version: 4.6.0
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**

Save Error
Relay could not be added. Object sonoff relay named 'Fridge 2 - Light (94-1)' at address 'http://192.168.86.94/#1' cannot be stored in the database. IntegrityError: UNIQUE constraint failed: Relay.id

**To Reproduce / Installation steps**
Not sure

**Expected behavior**
Add the relay

**Screenshots**
[image]

**Additional context**
The address of the relay was used before and deleted. 

**http://192.168.86.94/#1  Save Error
http://192.168.86.94/#2  Save Error
http://192.168.86.94/#3  Save Error**

http://192.168.86.94/#4  OK

I suspect the entry is still in the database even though it's not showing in the GUI.
I tried to clean up and fix the database, still no change.

_(venv) pi@raspberrypi:~/TerrariumPI/contrib$ python db_cleanup.py
This script will cleanup your terrariumpi.db file. We will keep 420 days, 0:00:00 of data from now. If you want to make a backup first, please enter no and make your backup.
Database size: 850.29 MB, free diskspace: 9.97 GB
Would you like to continue? Enter yes to start. Anything else will abort.
yes
Starting cleaning up table 'SensorHistory'. Deleting data older then 2021-11-08 16:34:50.594917 in batches of 1000 records. This could take some time...
Total rows:
 4940229 from 2022-04-06 11:34:00 till 2023-01-02 16:34:00. Took (40.18503999710083)
Clean up rows: No data, nothing to do!

(venv) pi@raspberrypi:~/TerrariumPI/contrib$ python fix_db.py 
Found broken database at: /home/pi/TerrariumPI/data/terrariumpi.db. Size: 850.3 MiB. Remaining disk space: 10.0 GiB
Would you like to continue to fix your database? Enter 'yes' to start. Anything else will abort.
yes
Copied 217674(100.00%) of 217674 pages...
Recovery done in 45.15 seconds. New database is located at: /home/pi/TerrariumPI/data/terrariumpi.db
(venv) pi@raspberrypi:~/TerrariumPI/contrib$_

Any Ideas?

Regards

Andy

theyosh, 2023-01-02:
I think you have ignored some relays? Because the unique key constrained is due to the fact that this address is already used for another relay. So I think you have ignored some.

In the settings screen you can see the ignored devices and by removing there, they will become visible again. But it needs a restart to make them fully work in updates. Because ignored devices will not be loading during a startup.

ewiandr73, 2023-01-02:
G'day,

I don't see any listed?

[image]

Regards

Andy

theyosh, 2023-01-03:
Hmm, what do you get if you open the url: `http://[raspberry_pi:8090/api/relays/` ?

This should give you a list of all relays in the database. Every relay should have an address field. And look if there are addresses with the value `http://192.168.86.94/#[1-4]` Those are the existing Sonoff relays.

Else, is it possible to upload the database somewhere? Not sure how big the database is. And then I need all the `terrariumpi.db` files in the `/data` folder. Then I can look into it see what you have and why you have this issue.

ewiandr73, 2023-01-03:
They are not in the api list.

[image]

Interestingly I can edit the address form the #4 to #1 and it allows me to save it.

I have copied, zipped and uploaded the terrariumpi.db to the following location
https://sftp.ericsson.net/public/folder/0g-XgMq6GESCXZd8EuFsXw/TerrariumPi

Password has been sent via the Matrix

Regards

Andy

ewiandr73, 2023-01-03:
I guess I could also change the IP address of the relay, might be easier?
Unless there is some underlying issue.

theyosh, 2023-01-03:
This is strange. I need some time to think this over. Changing the IP is an option if that is not to much work for you.

Could it be that you have updated some relays addresses in the past? Because, the ID is generated the first time when added. But then you update the address it will NOT update the ID, as that will break the enclosure areas. And history. So when you have added the relay at address `http://192.168.86.94/#2` in the past, and renamed it later to `http://192.168.86.216/#2` then you cannot add the relay at address `http://192.168.86.94/#2` again. The new generated ID will be the same as the first time.

So, yes, the only solution is to change the IP. Sorry for that. Technically you could make a script that can rename the ID based on the current addresses. But that will take a lot of time because that will rewrite the relay history actions...

theyosh, 2023-01-03:
And when added on the new IP, you can than change the IPs back to what you have right now... As updating the address is possible.

ewiandr73, 2023-01-03:
OK, must have been from when my other relay died, I didn't know it would keep the original ID.

Is that the only one that has a miss match?
I could delete and recreate the mismatched ID now 216?

theyosh, 2023-01-03:
Euh, I did not look into your IDS. This is just how the logic of the code works. So for now, just change the hardware IP and try again. That is for now the easiest to do.

ewiandr73, 2023-01-03:
Ok, thanks

ewiandr73, 2023-01-03:
I just tried 2 new addresses .96 & .98 and both show the same error?
I don't think I have used those addresses before.

theyosh, 2023-01-03:
Damm, than I need some more time... sorry for that. And it is getting late, so I will look into this tomorrow morning and will get back to you

ewiandr73, 2023-01-03:
All good, thanks

theyosh, 2023-01-06:
So, I found the problem. The issue is that I use the MAC address of the sonoff to make a unique ID. But this gives errors when hardware is used in the past and now being added again. So the only solution for now is to renumber your IDs. And the code has changed, that it now uses the IP address for unique ID, as I need something to make a unique ID off, which is rather stable source. And an IP will not change that much.

So, in your case, first I would like to ask to update the code to the latest version. `git pull`. This will update a lot, but also add a new script in the contrib folder called `fix_sonoff_relay_ids.py`. This script is created to renumber sonoff relay IDs.

In order to use this script, make sure TP4 is stopped! This is very important. Than I advise to make a copy of the database yourself. Just to be sure you can go back. Store the DB somewhere secure, like you own desktop.

Than enter the folder TerrariumPI and activate the python environment with the command `source venv/bin/activate` this will load the correct python environment. Than enter the folder `contrib` and run the command `./fix_sonoff_relay_ids.py`. This will start the renumber action. You should not do anything. It can take some time, as all data has to be checked.

The script will change the IDs of the relays itself. In the relay history table, all IDs will be updated, and in the areas where the relays are used, will also be updated with the new IDs. After that, it should all be good, and you can start TP4 again.

Look at the GUI if al relays are still there and have still the 24 hour history graphs. And check the enclosure areas if all relays are still correct. If this is the case, the renumbering was successful. Else I recommend to restore your backup database. There is also an extra backup at the `data` folder as well which can be used.

ewiandr73, 2023-01-06:
The script worked a treat....All present and correct :)

[image]

Thanks, let me know if you want anything tested.

Regards

Andy

closed 2023-01-06 by ewiandr73

theyosh, 2023-01-06:
And you were able to add the other relays as well? I see 28 relays :)

ewiandr73, 2023-01-06:
Yep, new one working too....

Before the change the new relay wasn't configure in the system, but it would turn on and off with the others, which was very weird, but I guess it was because of the MAC addresses in the ID somehow