theyosh/TerrariumPI #766: [BUG]: Upgrade from 4.6.0 to 4.6.1

ewiandr73, 2023-01-02:
G'day,

**Setup:**
 - TerrariumPI version: 4.6.0
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
Upgraded from 4.6.0 to 4.6.1 everything seemed to work ok without any errors, but the GUI is still showing the version as 4.6.0 and telling me there is a new upgrade.

Regards

Andy

theyosh, 2023-01-02:
Hi,
it needs a restart, so make sure you have restarted after the update. Then reload the GUI. And I think your browser is caching the GUI. So hold the shift key while reloading. That should do the trick.

ewiandr73, 2023-01-02:
Hi,

I tried the restart the RPI, GUI and PC.  Cleared the cache, but still no change

Also, I can see in the CLI that it still indicating that an update is required.

[image]

Regards

Andy

theyosh, 2023-01-02:
Are you on the main branch or the 4.x.y.z branch? You should be on the main branch. 4.x.y.z is a bit behind.

ewiandr73, 2023-01-02:
I thought I was on the main branch, how do I check?

theyosh, 2023-01-02:
run the command `git status`. this should produce:
```
On branch main
Your branch is up to date with 'origin/main'.
```

To switch to the main branch if you are on the 4.x.y.z branch, use the command: `git checkout main` and then `git pull`. There is no need to run the installer, as there are no new modules. After this, restart TP4. and you should be on the latest version.

theyosh, 2023-01-02:
And what setup needs 22 relays ;P I do not even test with that much ;P

ewiandr73, 2023-01-02:
Looks like I'm on the wrong branch :(

[image]

The 'git checkout main' doesn't seem to work?

22 Relays: I currently have 8 enclosures running, most have 4 relays on each

ewiandr73, 2023-01-02:
Let me know if you want anything tested with the 20+ relays :)

theyosh, 2023-01-02:
If you did the `git checkout main` and if you now run the command `git status` the message should change that you are on branch main on line 2. Than do `git pull` and that should do a lot of update I guess. And then just restart. That should do it.

And 8 enclosures is the max I have seen yet. That is very nice. I did not expect to be able to handle that much enclosures. The fact that it works, and you did not had any issues, is a best test ever! So I am happy to hear that this works out well.

ewiandr73, 2023-01-02:
It doesn't seem to like the 'main' ?

[image]

PS: I have another 7 enclosures I'm slowly adding, I'll let you know how they go.
Please let me know if you want anything tsted.

theyosh, 2023-01-02:
ok, I synced the 4.x.y.z branch. So for now, do a `git pull` and restart TP4. Than you should be on the same version as main.

If you have a Matrix account, we can discuss the change to the main branch through matrix. Is a bit handier then using this issue.
https://matrix.to/#/#terrariumpi:theyosh.nl

For today, I would like to end... tomorrow there is a new day.

theyosh, 2023-01-02:
And then also discuss the unique database error.... #765

theyosh, 2023-01-03:
So I found the correct steps to go back to the main branch:

1. `git remote set-branches origin '*'`
2. `git fetch`
3. `git checkout main`
4. `git pull`

And now you should be up to date in the main branch.

ewiandr73, 2023-01-03:
That worked...Thanks

closed 2023-01-03 by ewiandr73