theyosh/TerrariumPI #768: Relays keep turning keep dropping

res2cpu, 2023-01-02:
Hello and happy new year!

I recently had my TerrariumPI install die after a reboot on my rasp-pi 3 (I think I was running version 3.5) which I'm very happy to say has worked very well for for a long time.

Unable to get it working again I completed a clean install of 4.5, I've got 3 switches, 2 relays (lights and backup heater control) and a PWM dimmer). The relays (might be both or just the lights) keep turning off and then back on within less then half a second.  This only happens during the day when the light is supposed to be on.

This issue has only just started post upgrade, no hardware changes have been made.

I've attached my log file.  Are there any checks I can do?  Any known issues, I did check the closed issues but didn't see anything.

Thanks for any help you can provide.
[terrariumpi.log](https://github.com/theyosh/TerrariumPI/files/10332883/terrariumpi.log)

res2cpu, 2023-01-03:
I'm not sure if this helps but I noticed that TerrariumPI is restarting approximately every 17 minutes log segment below and log file attached:
2023-01-03 07:39:19,343 - INFO    - terrariumArea         - Updating NextEVO Universal AC MAINS Dimmer (MPDMv4.1) dimmer named 'PWM Heater' at address '32' to value 100%. Current: 43.38C, target: 46.50C
2023-01-03 07:39:19,353 - INFO    - terrariumEngine       - Updated Enclosure Uri with 2 areas in 0.03 seconds.
2023-01-03 07:39:19,524 - INFO    - terrariumEngine       - Engine update done in 4.56 seconds. Waiting for 25.44 seconds for the next round.
2023-01-03 07:39:44,969 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 5 relays, 0 buttons and 0 webcams.
2023-01-03 07:41:56,218 - INFO    - terrariumEngine       - Starting TerrariumPI 4.5.0 ...
2023-01-03 07:41:56,313 - INFO    - terrariumEngine       - Loaded 28 settings in 0.09 seconds.
2023-01-03 07:41:56,351 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Norwich, GB&appid=*********
[terrariumpi.log](https://github.com/theyosh/TerrariumPI/files/10335617/terrariumpi.log)

theyosh, 2023-01-03:
The restarting is due to the fact that for 2 minutes nothing happens. The line above is more than 2 minutes earlier. So somehow, when a new update rounds starts, it stops or hangs or there is a strange error.

I am afraid there is an error which is not showing right now. Could you do the following: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/ and keep it running until there is an error. Keep your terminal open and wait for an error to happen...

Not sure why this happens after 17 minutes.... I think it is 15 minutes and 2 minute timeout.

res2cpu, 2023-01-03:
I've done as you asked, I have the below so far.

The sensor reading I'm aware of from the old setup, but it looks like an issue with weather?

I'll leave it running, I didn't see a restart from the error but I only just noticed the restarting after it had been up a while.

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2023.01.03 10:33:36 =~=~=~=~=~=~=~=~=~=~=~=
/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py:3380: DeprecationWarning: Absolute template path names are deprecated.
  fname = self.search(name, self.lookup)
2023-01-03 10:40:56,048 - WARNING - hardware.sensor       - Invalid reading from sensor AM2302 sensor humidity named 'Humidity' at address '15', retrying in 0.5 seconds...
2023-01-03 10:41:56,050 - WARNING - hardware.sensor       - Invalid reading from sensor AM2302 sensor humidity named 'Humidity' at address '15', retrying in 0.5 seconds...
Exception in thread Thread-37:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1109, in __engine_loop
    self.weather.update()
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 74, in retry_decorator
    logger)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 33, in __retry_internal
    return f()
  File "/home/pi/TerrariumPI/weather/__init__.py", line 55, in update
    if self._load_data():
  File "/home/pi/TerrariumPI/weather/openweathermap_org_weather.py", line 202, in _load_data
    if self.__load_forecast_data():
  File "/home/pi/TerrariumPI/weather/openweathermap_org_weather.py", line 153, in __load_forecast_data
    if not self.__load_forecast_data_one_call():
  File "/home/pi/TerrariumPI/weather/openweathermap_org_weather.py", line 98, in __load_forecast_data_one_call
    if not data:
UnboundLocalError: local variable 'data' referenced before assignment

theyosh, 2023-01-03:
Ok, I think I found the issue.

The above error will stop everything. So nothing will be updated or changed when there is an error. And now you are running in debug mode, so there will be no restart.
So. press `Ctrl-C` (a couple of times) to stop everything.

Than do a `git pull` and restart in debug mode once more an let it running again.

res2cpu, 2023-01-03:
Update applied, No error's as yet and also the PWM dimmer for the heater looks to be working as expected now as it wasn't ramping up/down as it used to. 
[image]

I have a second viv running as a 2nd setup which I updated yesterday which also seems to have the same issue (running via docker).

I'll leave the main viv in diag until I get home tonight an then update the issue, I can't see if the light/relay's still playing up but I'm thinking it's caused by the constant restarting.

Thank's for looking into this so quickly and for the work you do.

res2cpu, 2023-01-03:
I notice under system status it has a system uptime, would a "service" uptime be a potential idea?

theyosh, 2023-01-03:
> I notice under system status it has a system uptime, would a "service" uptime be a potential idea?

Yeah, but it is not for 5 years the OS uptime, and I would leave it that way.

With the weather fix you relays should now be stable. I also updated the docker image with the latest code. Version number is not changed, so make sure you do `docker compose pull` to get the latest image from online.

res2cpu, 2023-01-03:
I've done the "docker compose pull" on my viv running the docker install and It's still having the same issue with "Starting TerrariumPI 4.6.1 ..." showing in the logs every 17 ish minutes.

Log attached, Do the same commands work to debug the docker install?
[terrariumpi_logfile (4).txt](https://github.com/theyosh/TerrariumPI/files/10339898/terrariumpi_logfile.4.txt)

theyosh, 2023-01-03:
Which version did you pull? Line 4 at https://theyosh.github.io/TerrariumPI/install/#docker

res2cpu, 2023-01-03:
My docker compose file:
version: "3.7"
services:
  terrariumpi:
    image: theyosh/terrariumpi:latest # Or use a specific version
    volumes:
      - /opt/terrariumpi/logs:/TerrariumPI/log
      - /opt/terrariumpi/data:/TerrariumPI/data
      - /opt/terrariumpi/media:/TerrariumPI/media
      - /opt/terrariumpi/scripts:/TerrariumPI/scripts
      - /opt/terrariumpi/webcam-archive:/TerrariumPI/webcam/archive
      - /opt/terrariumpi/DenkoviRelayCommandLineTool:/TerrariumPI/3rdparty/DenkoviRelayCommandLineTool
      - /boot/config.txt:/boot/config.txt
      - /boot/cmdline.txt:/boot/cmdline.txt
      - /etc/modules:/etc/modules
      - /dev:/dev
    network_mode: host
    restart: unless-stopped
    privileged: true
    cap_add:
      - NET_ADMIN
    stop_grace_period: 1m
    stop_signal: SIGINT
    init: true
    environment:
      TZ: "Europe/London" # timezone list can be found here https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
      ENABLE_I2C: "true"
      ENABLE_1_WIRE: "true"
      ENABLE_CAMERA: "true"
      ENABLE_SERIAL: "true"
      ENABLE_CO2_SENSORS: "true"
      AUTO_REBOOT: "true"

res2cpu, 2023-01-03:
As it didn't seem to have fixed the issue I rebooted the PI to be sure.

Just re-run both commands.  Should know in the next 20 or so min if it's changed anything.

res2cpu, 2023-01-03:
It's looking good at the moment, I'll leave it over night and report back in the morning.

theyosh, 2023-01-04:
Still all OK?

res2cpu, 2023-01-05:
Afternoon.

I can confirm that all looks to be working well.

Thank you for your assistance and all you do.

closed 2023-01-05 by res2cpu