theyosh/TerrariumPI #769: [FEATURE]: Request for new hardware support for shelly 1 plus web interface

Dave0408, 2023-01-04:
Hi @theyosh , 
Since I still have problems with the Meross Cloud Connection that I can't explain, I would like to replace the Meross power strip. 
I still have 3 [Shelly 1 Plus](https://www.shelly.cloud/de/products/product-overview/shelly-plus-1-1) from my home automation system [Shelly ](https://www.shelly.cloud/de) and have made sockets out of them. 

The Shellies can be easily addressed via a web interface:

e.g. :
```
http://ip-of-shelly-device/relay/<index>?turn=on
http://ip-of-shelly-device/relay/<index>?turn=off
http://ip-of-shelly-device/relay/<index>?turn=toggle
```
index is always 0 for the Shelly 1 Plus, as it only has one physical switch. The [Shelly 2PM](https://www.shelly.cloud/de/products/product-overview/splus2pm-3), for example, would have two switches.

See also manufacturer API documentation: [docs](https://shelly-api-docs.shelly.cloud/gen1/#shelly1-pm-overview)

I have now tried to set it up as a remote power switch. However, this does not seem to be correct due to an incorrect json response. It does not return the expected key:values pairs.

From the log:
```
2023-01-04 20:26:42,277 - ERROR - hardware.relay - Error changing state for relay Remote power switch (API) relay named 'Shelly Beleuchtung' at address 'http://<ip>/relay/0?turn=toggle' to 0.0 :Error changing relay Remote power switch (API) relay named 'Shelly Beleuchtung' at address 'http://<ip>/relay/0?turn=toggle' to state 0.0. Error: 'terrariumRelayRemote' object has no attribute '_set_hardware_value'.
2023-01-04 20:26:42,309 - ERROR - hardware.relay - Error getting new data from relay Remote power switch (API) relay named 'Shelly lighting' at address 'http://<ip>/relay/0?turn=toggle'. Error: float() argument must be a string or a number, not 'dict'.
2023-01-04 20:26:42,310 - WARNING - hardware.relay - Error getting new data from relay Remote power switch (API) relay named 'Shelly Lighting' at address 'http://<ip>/relay/0?turn=toggle'. Error: unknown, retrying in 0.5 seconds...
```

The json response of the shelly for above links is like:
##### if relay on:
`{"ison": true, "has_timer":false, "timer_started_at": 0, "timer_duration": 0.00, "timer_remaining": 0.00, "source": "http"}`

##### if relay off: 
`{"ison": false, "has_timer":false, "timer_started_at": 0, "timer_duration": 0.00, "timer_remaining": 0.00, "source": "http"}`

Would it be possible to set up another hardware type for "Shelly Web API" or "Shelly 1Plus Web API" ?

Shelly also offers connection options for Bluetooth and MQTT in addition to the web interface, but these are not interesting for me at the moment. 

Thank you very much. 

Kind regards,
David

Dave0408, 2023-01-04:
If it should be of interest, here is a log extract concerning Meross. 
But as I said, I would like to replace it. 
If you would like to analyse this in more depth, you are welcome to open a separate issue. I will be happy to support you as far as I can. 

```
pi@raspberrypi:~ $ tail -f /var/log/syslog
Jan  4 21:15:37 raspberrypi python[755]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/http_api.py", line 417, in <listcomp>
Jan  4 21:15:37 raspberrypi python[755]:     return [HttpDeviceInfo.from_dict(x) for x in result]
Jan  4 21:15:37 raspberrypi python[755]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/model/shared.py", line 27, in from_dict
Jan  4 21:15:37 raspberrypi python[755]:     obj = cls(**new_dict)
Jan  4 21:15:37 raspberrypi python[755]: TypeError: __init__() missing 1 required positional argument: 'skill_number'
Jan  4 21:15:37 raspberrypi python[755]: 2023-01-04 21:15:37,383 - WARNING - terrariumCloud        - Reconnecting to Meross cloud. Somehow the connection was lost ...
Jan  4 21:15:37 raspberrypi python[755]: 2023-01-04 21:15:37,385 - INFO    - terrariumCloud        - Stopping Meross cloud ...
Jan  4 21:15:38 raspberrypi python[755]: 2023-01-04 21:15:38,334 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:39 raspberrypi python[755]: 2023-01-04 21:15:39,336 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:40 raspberrypi python[755]: 2023-01-04 21:15:40,338 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:41 raspberrypi python[755]: 2023-01-04 21:15:41,340 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:42 raspberrypi python[755]: 2023-01-04 21:15:42,342 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:43 raspberrypi python[755]: 2023-01-04 21:15:43,344 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:44 raspberrypi python[755]: 2023-01-04 21:15:44,346 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:45 raspberrypi python[755]: 2023-01-04 21:15:45,348 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:46 raspberrypi python[755]: 2023-01-04 21:15:46,350 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:47 raspberrypi python[755]: 2023-01-04 21:15:47,352 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:48 raspberrypi python[755]: 2023-01-04 21:15:48,354 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:49 raspberrypi python[755]: 2023-01-04 21:15:49,356 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:50 raspberrypi python[755]: 2023-01-04 21:15:50,358 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:51 raspberrypi python[755]: 2023-01-04 21:15:51,360 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:52 raspberrypi python[755]: 2023-01-04 21:15:52,361 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:53 raspberrypi python[755]: 2023-01-04 21:15:53,363 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:54 raspberrypi python[755]: 2023-01-04 21:15:54,365 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:55 raspberrypi python[755]: 2023-01-04 21:15:55,367 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:56 raspberrypi python[755]: 2023-01-04 21:15:56,369 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:57 raspberrypi python[755]: 2023-01-04 21:15:57,372 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:58 raspberrypi python[755]: 2023-01-04 21:15:58,374 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:15:59 raspberrypi python[755]: 2023-01-04 21:15:59,375 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:00 raspberrypi python[755]: 2023-01-04 21:16:00,377 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:01 raspberrypi python[755]: 2023-01-04 21:16:01,379 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:02 raspberrypi python[755]: 2023-01-04 21:16:02,381 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:03 raspberrypi python[755]: 2023-01-04 21:16:03,383 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:04 raspberrypi python[755]: 2023-01-04 21:16:04,385 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:05 raspberrypi python[755]: 2023-01-04 21:16:05,387 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:16:06 raspberrypi python[755]: 2023-01-04 21:16:06,389 - ERROR   - terrariumCloud        - Could not login to Meross cloud within 30 seconds. Restarting...
Jan  4 21:16:06 raspberrypi python[755]: 2023-01-04 21:16:06,392 - INFO    - terrariumCloud        - Meross cloud is connected! Found 0 devices in 30.06 seconds.
```

theyosh, 2023-01-05:
Ok, I added the Shelly relay. Works the same as a Sonoff. So do a `git pull` and restart TP4. Refresh the browser GUI (holding shift while reloading). Then you should be able to add a Shelly relay. The address is: `http://[host_or_ip]#[relay_nr_from_1_till_x]`

If you are able to test if the relay works, that would be great. Than I will add it to the documentation as well.

Dave0408, 2023-01-05:
Hello @theyosh , 
Thank you very much for the quick integration.
Unfortunately, I am coming up with an error: 

[image]

#### http://192.168.1.224#1
`2023-01-05 20:21:39,121 - WARNING - hardware.relay        - Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224]#1': Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224]#1': Unknown error, retrying in 0.5 seconds...`

#### http://192.168.1.224#0
`2023-01-05 20:21:32,674 - WARNING - hardware.relay        - Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224]#0': Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224]#0': Unknown error, retrying in 0.5 seconds...`

Same with: 
http://192.168.1.224/#0
http://192.168.1.224/#1

It should be relay nr 0. 
[image]

theyosh, 2023-01-05:
The relay numbering starts at 1. That is easier to remember for users. In code I reduce it by 1 to get the right number.

But you have entered an invalid address: `http://192.168.1.224]#0` there is a `]` in it. That is not valid...

So in this case the address `http://192.168.1.224#1` or `http://192.168.1.224/#1` should be valid and working.

Dave0408, 2023-01-05:
Sorry, 
copied the wrong attempt.

```
Jan  5 20:44:32 raspberrypi python[581]: 2023-01-05 20:44:32,389 - WARNING - hardware.relay        - Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Unknown error, retrying in 0.5 seconds...
Jan  5 20:44:32 raspberrypi python[581]: 2023-01-05 20:44:32,918 - WARNING - hardware.relay        - Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Unknown error, retrying in 0.5 seconds...
```

[image]

theyosh, 2023-01-05:
Ok, could you run in debug mode and than try to add the relay? There should be some debug which maybe explains what I did wrong.

https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

theyosh, 2023-01-05:
Sorry, but you also need to do a `git pull` to make sure you have the latest code

Dave0408, 2023-01-05:
> Sorry, but you also need to do a `git pull` to make sure you have the latest code

```
(venv) pi@raspberrypi:~/TerrariumPI $ ls -al hardware/relay/shelly_relay.py
-rw-r--r-- 1 pi pi 3255 Jan  5 20:08 hardware/relay/shelly_relay.py
```
So i did a git pull as mentioned. 

Here's the debug log:
```
Address
{'protocol': 'http', 'user': None, 'passwd': None, 'host': '192.168.1.224', 'nr': 0}
Device
http://192.168.1.224/status
2023-01-05 21:01:15,971 - WARNING - hardware.relay        - Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Unknown error, retrying in 0.5 seconds...
Address
{'protocol': 'http', 'user': None, 'passwd': None, 'host': '192.168.1.224', 'nr': 0}
Device
http://192.168.1.224/status
2023-01-05 21:01:16,500 - WARNING - hardware.relay        - Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Could not load hardware for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1': Unknown error, retrying in 0.5 seconds...
Address
{'protocol': 'http', 'user': None, 'passwd': None, 'host': '192.168.1.224', 'nr': 0}
Device
http://192.168.1.224/status
```

[http://192.168.1.224/status](url)   -> page not found

If i use [http://192.168.1.224/relay/0](url) i get:
`{"ison": false, "has_timer":false, "timer_started_at": 0, "timer_duration": 0.00, "timer_remaining": 0.00, "source": "WS_in"}`

theyosh, 2023-01-05:
Strange, the API says it should be there: https://shelly-api-docs.shelly.cloud/gen1/#shelly1-1pm-status

And I want to use the status page, so that I can load all relay status at once. You have only one, but there are relays with 2 switches.

Different version in your switch: https://shelly-api-docs.shelly.cloud/gen1/#changelog ?

Dave0408, 2023-01-05:
Oh I'm sorry. 
I seems to be a gen2 device now. Might be moved with a fimware update. 

One switch relay sample (which i use here):
[https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus1](https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus1)

`http://192.168.1.224/rpc/Input.GetStatus?id=0`
-> `{"id":0,"state":false}`

Two switch relay sample (which i also use in my house automation, but not at my terrarium):
[https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus2PM](https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus2PM)

[https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Switch#status](https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Switch#status)
> To obtain the status of the Switch component its **id must be specified**.

theyosh, 2023-01-05:
Ok, I updated the code so there is no caching. API is not clear, so I use the basic urls you gave me /relay/0 and /relay/0/?turn=on

So do a `git pull` and restart once more in debug mode. And see if you are now able to add the relay.

Dave0408, 2023-01-05:
Adding the relay was successfull:
```
Device
http://192.168.1.224/relay/0
```

When I switch on and off in manual mode, the following messages appear regarding a variable definition. But it already switches on and off successfully. 
```
2023-01-05 21:50:23,710 - WARNING - hardware.relay        - Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 100.0. Error: local variable 'data' referenced before assignment, retrying in 0.5 seconds...
Set hardware url: http://192.168.1.224/relay/0?turn=on
2023-01-05 21:50:24,237 - WARNING - hardware.relay        - Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 100.0. Error: local variable 'data' referenced before assignment, retrying in 0.5 seconds...
Set hardware url: http://192.168.1.224/relay/0?turn=on
2023-01-05 21:50:24,767 - ERROR   - hardware.relay        - Error changing state for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to 100.0 :Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 100.0. Error: local variable 'data' referenced before assignment
Set hardware url: http://192.168.1.224/relay/0?turn=on
2023-01-05 21:50:29,081 - WARNING - hardware.relay        - Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 100.0. Error: local variable 'data' referenced before assignment, retrying in 0.5 seconds...
Set hardware url: http://192.168.1.224/relay/0?turn=on
2023-01-05 21:50:30,054 - WARNING - hardware.relay        - Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 100.0. Error: local variable 'data' referenced before assignment, retrying in 0.5 seconds...
Set hardware url: http://192.168.1.224/relay/0?turn=on
2023-01-05 21:50:30,588 - ERROR   - hardware.relay        - Error changing state for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to 100.0 :Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 100.0. Error: local variable 'data' referenced before assignment
Set hardware url: http://192.168.1.224/relay/0?turn=off
2023-01-05 21:50:34,237 - WARNING - hardware.relay        - Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 0.0. Error: local variable 'data' referenced before assignment, retrying in 0.5 seconds...
Set hardware url: http://192.168.1.224/relay/0?turn=off
2023-01-05 21:50:34,762 - WARNING - hardware.relay        - Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 0.0. Error: local variable 'data' referenced before assignment, retrying in 0.5 seconds...
Set hardware url: http://192.168.1.224/relay/0?turn=off
2023-01-05 21:50:35,293 - ERROR   - hardware.relay        - Error changing state for relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to 0.0 :Error changing relay Shelly relay named 'Shelly Terrarium Licht' at address 'http://192.168.1.224#1' to state 0.0. Error: local variable 'data' referenced before assignment
```

theyosh, 2023-01-05:
another git pull please. I think we crossed some updates. If you will not get updates after your git pull, then I have to look further into the code

theyosh, 2023-01-05:
Cool! Yeah, could happen that we get old documentation ...;)

But with the current code, both type devices should be supported. Gen-1 and Gen-2. So that is nice.

So than I will add some documentation for this relay and we have new hardware we support. Nice!

theyosh, 2023-01-06:
Still working? Can we close this issue?

Dave0408, 2023-01-06:
Hi, 
it's still running. I don't see any problems in the logs either. Everything is fine. :)

theyosh, 2023-01-06:
Thanks. Than I close it now.

closed 2023-01-06 by theyosh