theyosh/TerrariumPI #770: [BUG]: TerrariumPi crashes during startup

Dave0408, 2023-01-04:
I have just updated to the latest TerrariumPi version 4.6.1 and now I can't get the software to start at all. 
Here is the log extract: 
```
pi@raspberrypi:~ $ tail -f /var/log/syslog
Jan  4 21:22:03 raspberrypi python[854]: 2023-01-04 21:22:03,299 - INFO    - terrariumEngine       - Loading Meross cloud connection.
Jan  4 21:22:03 raspberrypi python[854]: 2023-01-04 21:22:03,301 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
Jan  4 21:22:04 raspberrypi python[854]: 2023-01-04 21:22:04,291 - INFO    - terrariumCloud        - Closed Meross cloud connection
Jan  4 21:22:04 raspberrypi python[854]: 2023-01-04 21:22:04,293 - ERROR   - terrariumCloud        - Error in cloud run: __init__() missing 1 required positional argument: 'skill_number'
Jan  4 21:22:04 raspberrypi python[854]: Traceback (most recent call last):
Jan  4 21:22:04 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 49, in _run
Jan  4 21:22:04 raspberrypi python[854]:     data.result()
Jan  4 21:22:04 raspberrypi python[854]:   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
Jan  4 21:22:04 raspberrypi python[854]:     return self.__get_result()
Jan  4 21:22:04 raspberrypi python[854]:   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
Jan  4 21:22:04 raspberrypi python[854]:     raise self._exception
Jan  4 21:22:04 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 187, in _main_process
Jan  4 21:22:04 raspberrypi python[854]:     await self.manager.async_device_discovery()
Jan  4 21:22:04 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/manager.py", line 331, in async_device_discovery
Jan  4 21:22:04 raspberrypi python[854]:     http_devices = await self._http_client.async_list_devices()
Jan  4 21:22:04 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/http_api.py", line 417, in async_list_devices
Jan  4 21:22:04 raspberrypi python[854]:     return [HttpDeviceInfo.from_dict(x) for x in result]
Jan  4 21:22:04 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/http_api.py", line 417, in <listcomp>
Jan  4 21:22:04 raspberrypi python[854]:     return [HttpDeviceInfo.from_dict(x) for x in result]
Jan  4 21:22:04 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/model/shared.py", line 27, in from_dict
Jan  4 21:22:04 raspberrypi python[854]:     obj = cls(**new_dict)
Jan  4 21:22:04 raspberrypi python[854]: TypeError: __init__() missing 1 required positional argument: 'skill_number'
Jan  4 21:22:04 raspberrypi python[854]: 2023-01-04 21:22:04,316 - WARNING - terrariumCloud        - Reconnecting to Meross cloud. Somehow the connection was lost ...
Jan  4 21:22:04 raspberrypi python[854]: 2023-01-04 21:22:04,319 - INFO    - terrariumCloud        - Stopping Meross cloud ...
Jan  4 21:22:04 raspberrypi python[854]: 2023-01-04 21:22:04,324 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ...
```
after a short while:
```
Jan  4 21:23:32 raspberrypi python[854]: Exception in thread Thread-2:
Jan  4 21:23:32 raspberrypi python[854]: Traceback (most recent call last):
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 49, in _run
Jan  4 21:23:32 raspberrypi python[854]:     data.result()
Jan  4 21:23:32 raspberrypi python[854]:   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
Jan  4 21:23:32 raspberrypi python[854]:     return self.__get_result()
Jan  4 21:23:32 raspberrypi python[854]:   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
Jan  4 21:23:32 raspberrypi python[854]:     raise self._exception
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 187, in _main_process
Jan  4 21:23:32 raspberrypi python[854]:     await self.manager.async_device_discovery()
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/manager.py", line 331, in async_device_discovery
Jan  4 21:23:32 raspberrypi python[854]:     http_devices = await self._http_client.async_list_devices()
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/http_api.py", line 417, in async_list_devices
Jan  4 21:23:32 raspberrypi python[854]:     return [HttpDeviceInfo.from_dict(x) for x in result]
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/http_api.py", line 417, in <listcomp>
Jan  4 21:23:32 raspberrypi python[854]:     return [HttpDeviceInfo.from_dict(x) for x in result]
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/meross_iot/model/shared.py", line 27, in from_dict
Jan  4 21:23:32 raspberrypi python[854]:     obj = cls(**new_dict)
Jan  4 21:23:32 raspberrypi python[854]: TypeError: __init__() missing 1 required positional argument: 'skill_number'
Jan  4 21:23:32 raspberrypi python[854]: During handling of the above exception, another exception occurred:
Jan  4 21:23:32 raspberrypi python[854]: Traceback (most recent call last):
Jan  4 21:23:32 raspberrypi python[854]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Jan  4 21:23:32 raspberrypi python[854]:     self.run()
Jan  4 21:23:32 raspberrypi python[854]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Jan  4 21:23:32 raspberrypi python[854]:     self._target(*self._args, **self._kwargs)
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 52, in _run
Jan  4 21:23:32 raspberrypi python[854]:     self.reconnect()
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 145, in reconnect
Jan  4 21:23:32 raspberrypi python[854]:     self.stop()
Jan  4 21:23:32 raspberrypi python[854]:   File "/home/pi/TerrariumPI/terrariumCloud.py", line 137, in stop
Jan  4 21:23:32 raspberrypi python[854]:     self.__engine['thread'].join()
Jan  4 21:23:32 raspberrypi python[854]:   File "/usr/lib/python3.7/threading.py", line 1029, in join
Jan  4 21:23:32 raspberrypi python[854]:     raise RuntimeError("cannot join current thread")
Jan  4 21:23:32 raspberrypi python[854]: RuntimeError: cannot join current thread
```

I don't know if this is related to my Meross power strip or the incorrectly/erroneously defined remote switch. See #769 

Any idea how to get it up and running again? 

Thanks in advance.

theyosh, 2023-01-04:
This is just fixed in the latest code update. Meross did change their api: https://github.com/albertogeniola/MerossIot/issues/274

Stop TP4. Do a `git pull` and then rerun the installer `sudo ./install.sh` After that, it should startup again.

Dave0408, 2023-01-04:
Thanks. Fixed and is running again!

theyosh, 2023-01-05:
Closing due to a Meross API change.

closed 2023-01-05 by theyosh