theyosh/TerrariumPI #775: [BUG]:  Unable to save setting changes on New install

ewiandr73, 2023-01-19:
- TerrariumPI version: 4
 - Python version: 3
 - Raspberry PI: 3B 

**Describe the bug**
I performed a fresh install to setup a system for my Son.
When trying to change the system settings i was prompted to set the language and Authentication mode.

When I select either one it seems to accept the entry, but when I then select the second one the already set field indicates I need to make a choice again.

My already installed and working system works fine.

**Screenshots**
[image]

theyosh, 2023-01-19:
Yes, this is a bug. For the settings screen, it is possible to reload the page and try again. After an error, the form somehow does not reset it state. So if you reload this page again, and then select the correct values for Authentication and Language, than you can save it.

But this bug is in all forms with a select. So other forms will have this issue as well.

I hope the fix it this weekend.

ewiandr73, 2023-01-19:
I tried reloading the page also tried another browser (Edge & Firefox), but still no change.

Funny I noticed a similar issue when trying to add a sensor to an old install, so just thought I try a fresh install.
I didn't update the OS though just stopped service and deleted the directory and then performed a fresh install.
  
I'll wait for the fix and see if it fixes my issue too, no rush I'm still finishing the interior of the enclosure. :)

Andy

kaju666, 2023-01-22:
The same for me, my sdcard broke and i did new install and cant add webcams/sensors.

Waiting for fix.

ewiandr73, 2023-01-23:
Update:
I stopped the system and performed a git pull and install.  So far seems to be behaving...

Andy

ewiandr73, 2023-01-23:
Spoke too soon... 😔

The 'Add area' seems to be playing up now.

theyosh, 2023-01-23:
And found it! Unfortunately a 3rd party library was causing this... So I downgraded that library and created a new GUI build.

So if you all get the latest code, it should work again. For people that have used @ewiandr73 fix you need some more steps to get back on the main branch:

1. `git remote set-branches origin '*'`
2. `git fetch`
3. `git checkout main`
4. `git pull`

And that should do it.

ewiandr73, 2023-01-23:
The fix was all thanks to @kaju666 

I have performed a git pull and install again and everything looks to be working correctly again.

Thanks

closed 2023-01-23 by ewiandr73