theyosh/TerrariumPI #776: [FEATURE]: Actuator verification to detect sensor failure

1ucket, 2023-01-19:
**Is your feature request related to a problem? Please describe.**
In the potential case of sensor failure a too strong dimensioned heating source might heat the terrarium to unsafe temperatures. The same goes for water levels in aquariums or similar. An actuator (e.g. heater) failure is not as risky in the short term, but a warning to the user would still be beneficial to avoid too low temperatures.

**Describe the solution you'd like**
The easiest solution would be to monitor target and actual values. If there is no trend actual -> target visible (and they are not already within a certain proximity) the relais would have to be turned off and an alarm issued (e.g. a sensor_alarm or a new type of notification).

A configuration such as Klipper's check_gain_time & heating_gain (see link below) would allow for monitoring of quick actuators (such as water pumps) and still give slow actuators (such as heating stones) their time.
A simplification would be to include a time goal per relais. If less than 5% of the intended change happened in 20% of the time an alarm is sounded (values just an example).

**Additional context**
From looking at the source I could not identify any safety measures like this. In 3D printers this is called thermal runaway protection/ heater verification (for example those are the implementations in [Marlin](https://github.com/MarlinFirmware/Marlin/blob/f2b8942adc6ff85fe99c0bd5ebae83acdf4e1b75/Marlin/src/module/temperature.cpp#L2827) (C++) and [Klipper](https://github.com/Klipper3d/klipper/blob/0b918b357cb0c282d53cbdf59e1931a2054cd60a/klippy/extras/verify_heater.py) (Python)).

theyosh, 2023-01-23:
I need some thinking about this...

theyosh, 2023-03-12:
Hi, after some thinking of this, I came up with a simpler solution. In the areas you can select sensors that control the relays. If a sensor does not update for 5 minutes, it will get into an error state. So this error state is now checked in the area, and when all sensors are in an error state, the relays will shut down and stay down.

Using trend data is much harder, and takes more DB power. Which is not something I have much left of to keep it fast. So for now there is a somewhat security measurement in place when the sensors are failing.

closed 2023-04-10 by theyosh