theyosh/TerrariumPI #777: [BUG]: pcf8574 expander relais sets all other outputs to HIGH

1ucket, 2023-01-19:
**More info**

**Setup:**
 - TerrariumPI version: [4.6.1]
 - Python version: [2]
 - Raspberry PI: [4b]

**Describe the bug**
I am using a relay board which features an integrated PCF8574 IO expander ([this one](https://www.tindie.com/products/bugrovs2012/hd30a-4-channel-electromagnetic-i2c-relay-12v-30a/) to be exact). I added two relais, with ports 0 and 1. When setting one of the relais the other relays are turned on, independent of their state in the interface.

**To Reproduce / Installation steps**
1. Connect a PCF8574 IO expander via I2C
2. Add two relays for the expander
3. Set the relays individually

**Expected behavior**
The relays are set individually, setting one relay does not everwrite the other relay.

**Additional context**
[terrariumpi.log](https://github.com/theyosh/TerrariumPI/files/10459290/terrariumpi.log)

1ucket, 2023-01-20:
I've analyzed the code and it seems that every relay on the pcf8574 [gets it's own terrariumIOExpander instance ](https://github.com/theyosh/TerrariumPI/blob/a23c8410ef35a21ab5a2f767388f5733472fc6d7/hardware/relay/gpio_relay.py#L21).

Now on load_hardware the terrariumPCF8574IOExpander instance [creates a new PCF8574 instance](https://github.com/theyosh/TerrariumPI/blob/a23c8410ef35a21ab5a2f767388f5733472fc6d7/hardware/io_expander.py#L69) (btw. neat implementation with the lambda to set the class dynamically).

The setter calls [set_output](https://github.com/flyte/pcf8574/blob/c2843b9f20059f1600612dacdba55d8e493bcb57/pcf8574/__init__.py#L91) in the PCF8574 library object. Now this function maintains a current_state, which unfortunately is only for this single relay. So I would propose 2 solutions:
- Use a class variable to keep track of all pcf8574 relay states (which wouldn't be too clean/ redundant, but possible). Set them all at once using [port](https://github.com/flyte/pcf8574/blob/c2843b9f20059f1600612dacdba55d8e493bcb57/pcf8574/__init__.py#L79) instead of [set_output](https://github.com/flyte/pcf8574/blob/c2843b9f20059f1600612dacdba55d8e493bcb57/pcf8574/__init__.py#L91) to avoid relays quickly toggling (which might be detrimental to some actors such as mercury-vapor lamps, also flickering would be annoying).
- Read all pcf8574 relay states on that I2C address from the cache when setting a relay and write them all at once (again using using [port](https://github.com/flyte/pcf8574/blob/c2843b9f20059f1600612dacdba55d8e493bcb57/pcf8574/__init__.py#L79)). Similarly to how it's done in [terrariumRelayDenkoviV2](https://github.com/theyosh/TerrariumPI/blob/main/hardware/relay/denkovi_relay.py).

1ucket, 2023-01-20:
The issue should btw. also exist for [pcf8575](https://github.com/rp3tya/PCF8575/tree/master/pcf8575), from having a quick glance at the [library](https://github.com/rp3tya/PCF8575). But as terrariumPCF8575IOExpander is just a child class of terrariumPCF8574IOExpander that share the same setter the issue should be resolved when fixing pcf8574, so no separate ticket/ implementation is needed.

Btw. I think it would be cleaner to put the state getter and setter into terrariumIOExpander instead of terrariumPCF8574IOExpander and let both terrariumPCF8574IOExpander and terrariumPCF8575IOExpander be a child of terrariumIOExpander. IOExpander -> PCF8574 -> PCF8575 is a bit confusing.

theyosh, 2023-01-22:
Hi, thanks for feedback. This hardware is not tested by me, so I think the person that tested it, only used 1 relay. So we never had seen this issue.

And I think I made a fix for this. A bit different than proposed. Basically you should not reload the io expander multiple time like we do now. So I added some caching logic, to see if the IO expander has been loaded before, and will reuse that one. So that should mean that every IO expander should only be loaded once based on the I2C address.

At first I thought about using a singleton, but that does not look at the class init parameters. So I had to use some kind of caching.

I still depend on the internal state of the IO expander itself. So I still use the set_output call.

I hope this will fix the issue, and make the relays stable. Are you able to test this? And https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/ could be handy when testing.

1ucket, 2023-01-25:
Hi, thanks for the quick response and sorry for taking so long testing it. Unfortunately I think your commit broke something. In the startup log the following messages appear:
`2023-01-25 20:11:58,819 - WARNING - hardware.relay        - Could not load hardware for relay GPIO devices relay named 'Test Relais 0' at address 'pcf8574-0,0x26': 'list' object has no attribute 'join', retrying in 0.5 seconds...
2023-01-25 20:11:59,322 - WARNING - hardware.relay        - Could not load hardware for relay GPIO devices relay named 'Test Relais 0' at address 'pcf8574-0,0x26': 'list' object has no attribute 'join', retrying in 0.5 seconds...
2023-01-25 20:11:59,825 - ERROR   - terrariumEngine       - Error loading relay gpio relay named 'Test Relais 0' at address 'pcf8574-0,0x26' with error: Could not load hardware for relay GPIO devices relay named 'Test Relais 0' at address 'pcf8574-0,0x26': 'list' object has no attribute 'join'.`
 
When trying to toggle the relay the following message appears:
`Error getting relay gpio relay named 'Test Relais 0' at address 'pcf8574-0,0x26' detail. '5f198ac75a49e0c648aad1fca32de3b7'`

1ucket, 2023-01-25:
I also can't add a new pcf8574 relay, as the following error message appears:
`Relay could not be added. Could not load hardware for relay GPIO devices relay named 'Test Relais 0' at address 'pcf8574-0,0x26': 'list' object has no attribute 'join'`

Judging from the fact that you don't need the relay to be responsive when adding it (I could add it without having it connected) I think this makes it possible for you to test this.

theyosh, 2023-01-25:
Apparently I mixed up python and javascript code in syntax. So do a git pull and restart. That should fix the new bug, and hopefully also the loading and relay toggling.

1ucket, 2023-01-25:
Thanks for the quick fix! Unfortunately there's a new error message new when starting TerrariumPi/ adding a relay (both the same):
`Error loading relay gpio relay named 'Test Relais 1' at address 'pcf8574-1,0x26' with error: Could not load hardware for relay GPIO devices relay named 'Test Relais 1' at address 'pcf8574-1,0x26': sequence item 0: expected str instance, int found.`

theyosh, 2023-01-25:
Ok another attempt. Do a git pull and restart.. Lets hope I got it right this time ;P

1ucket, 2023-01-25:
The relay is working properly now! Thank you really much, I can't believe how fast you are! :D
But I had a look at the logging output (to see if there's any more issues during loading), and when trying to exit with ctrl+c this error came, which also seems to be related to the pcf8574:
`Traceback (most recent call last):
  File "terrariumPI.py", line 18, in <module>
    terrariumEngine.stop()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1460, in stop
    self.relays[relay].stop()
  File "/home/pi/TerrariumPI/hardware/relay/gpio_relay.py", line 58, in stop
    self.device.close()
AttributeError: 'terrariumPCF8574IOExpander' object has no attribute 'close'`

theyosh, 2023-01-25:
Ok, that I will fix later. For now I hope you can at least at the other relays and that they work nicely together. This closing issue is not that big of a problem right now.

theyosh, 2023-01-25:
Ah, forget that. I fixed the closed issue as well..

1ucket, 2023-01-26:
Thanks! I've verified that the shutdown is now successful, I guess I can close the ticket now.

1ucket, 2023-01-26:
I wanted to add an inverse relay. The issue seems to still exist for them, when setting one of them all other relays are reset to on again.

1ucket, 2023-01-26:
I've also deleted the normal (non inverse) relays and restarted TerrariumPi (to clear any eventual cache), the issue is still present. All inverse relays also start as turned on (I don't know if this is a separate issue, but I would suspect so).

1ucket, 2023-01-26:
I've just created a regular GPIO (inverse) relay (without pcf8574), that one is turned off on Terrarium startup. So it's probably also a pcf8574 issue.

theyosh, 2023-01-28:
This is very strange. But I refactored the code a bit. Maybe it is now working better. Now also using the .port array like the example of the pcf8574 library.

Could you try the new code? Do a `git pull` and start in debug mode.

 Else I think I should order the same relay board for testing. It is also interesting board to have for a future build.

commit 058cf946 (2023-03-10): Fixed IO expanders pcf857(4/5). Fixes #777
closed 2023-03-10 by theyosh