theyosh/TerrariumPI #778: [BUG]: Loading weather exception

kaju666, 2023-01-25:
**Setup:**
 - TerrariumPI version: 4.6.1
 - Install: Manual
 - Raspberry PI: 3B

**Describe the bug**
I always have this error, i tried several formats.
Link is working in browser giving me api data.
I lost ideas why this is not working

`2023-01-25 18:04:44,439 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://api.openweathermap.org/data/2.5/weather?q=Warsaw&appid=********* is not valid! Please check your source
2023-01-25 18:04:44,437 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Warsaw&appid=*********
`

**To Reproduce / Installation steps**
Fresh instalation

theyosh, 2023-01-25:
The url is not correct, even it works in the browser. I have a bit more precise check. You need to add your country letters to the url:

`https://api.openweathermap.org/data/2.5/weather?q=Warsaw,pl&appid=*********` So add `,PL` to the city name and it should work.

kaju666, 2023-01-25:
My got, thank you so much. I tried this with Poland... not PL.

closed 2023-01-25 by kaju666