theyosh/TerrariumPI #780: [FEATURE]: Set dimmer value for timer areas

1ucket, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
Timer controlled areas with a dimmer relay will always have the dimmer at 100% (outside of the turning on and off fading).

**Describe the solution you'd like**
For timer controlled areas a max dimmer percentage value is configurable.

theyosh, 2023-02-06:
I am not sure, but as far I know this is not happening.

When you set the max power of a dimmer to 50% it will never get higher. At the dimmer self, it will not allow a higher value than configured.

When used in an area, other than Lights or Waterrank, the dimmer will work following the PID principle https://en.wikipedia.org/wiki/PID_controller . This should try to get the dimmer at the best spot to keep an average temperature / humidity.
But also here, the max dimmer value is used for max action.

Are you sure you have setup the dimmer correctly? Adding screenshots can help. From both the dimmer relay as the area you are using it in.

1ucket, 2023-02-06:
For now I've setup the dimmers with a timer and a on/ off ramp. For some reason I couldn't get the one-wire sensors to work for now. But to be honest I haven't put much time into it as I've been quite busy.
The idea was to set a max value for timer controlled dimmers cause it's a tad bit too warm in my enclosure right now (countering this by setting the value manually). But yeah, it's a totally optional feature that would just add to a specific use case (I could also think about wanting to slow down water pumps or something, but again not the usual stuff).

closed 2023-03-10 by theyosh