theyosh/TerrariumPI #781: [BUG]: TerrariumPi crashing on startup

1ucket, 2023-02-06:
**Setup:**
 - TerrariumPI version: 4.6.1
 - Install: Manual
 - Raspberry PI: 4b 2Gb

**Describe the bug**
On startup Terrariumpi is crashing when loading the enclosures.

**To Reproduce / Installation steps**
I can only give clues how this issue might have been produced. I was turning off the Raspberry Pi by disconnecting it from the power source, so it might be that it didn't save everything properly. Other than that I connected a relay in between turning it off and on (on pin 12) but it wasn't yet added in the software. But I've disconnected said relay again and restarted TerrariumPi and the issue is still there.

**Expected behavior**
TerrariumPi starts up properly.

**Screenshots**
`2023-02-06 18:36:57,764 - INFO    - terrariumEngine       - Starting TerrariumPI 4.6.1 ...
2023-02-06 18:36:57,820 - INFO    - terrariumEngine       - Loaded 29 settings in 0.05 seconds.
2023-02-06 18:36:57,855 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Berlin,Germany&appid=9f6128a7dfcdcdc774fc953965fea377
2023-02-06 18:36:57,858 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://api.openweathermap.org/data/2.5/weather?q=Berlin,Germany&appid=9f6128a7dfcdcdc774fc953965fea377' is not valid! Please check your source
2023-02-06 18:36:58,352 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-02-06 18:36:58,361 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3ce10457f20f': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 28-3ce10457f20f: Invalid path., retrying in 0.5 seconds...
2023-02-06 18:36:58,863 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3ce10457f20f': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 28-3ce10457f20f: Invalid path., retrying in 0.5 seconds...
2023-02-06 18:36:59,366 - ERROR   - terrariumEngine       - Error loading 1wire temperature named '1-Wire sensor measuring temperature' at address '28-3ce10457f20f' with error: Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3ce10457f20f': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 28-3ce10457f20f: Invalid path..
2023-02-06 18:36:59,369 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor humidity named 'Testsensor' at address '28-3ce10457f20f': Unable to load sensor 1wire Testsensor at address 28-3ce10457f20f: Invalid path., retrying in 0.5 seconds...
2023-02-06 18:36:59,872 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor humidity named 'Testsensor' at address '28-3ce10457f20f': Unable to load sensor 1wire Testsensor at address 28-3ce10457f20f: Invalid path., retrying in 0.5 seconds...
2023-02-06 18:37:00,375 - ERROR   - terrariumEngine       - Error loading 1wire humidity named 'Testsensor' at address '28-3ce10457f20f' with error: Unable to load sensor 1-Wire sensor humidity named 'Testsensor' at address '28-3ce10457f20f': Unable to load sensor 1wire Testsensor at address 28-3ce10457f20f: Invalid path..
2023-02-06 18:37:00,377 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-02-06 18:37:09,606 - INFO    - terrariumEngine       - Loaded 0 sensors in 11.25 seconds.
2023-02-06 18:37:09,607 - INFO    - terrariumEngine       - Loading existing relays from database.
2023-02-06 18:37:10,058 - INFO    - terrariumEngine       - Loaded relay i2c_4ch-dimmer dimmer named 'Basking Spot links (75W)' at address '1,0x27' value 100.00 in 0.42 seconds.
2023-02-06 18:37:10,287 - INFO    - terrariumEngine       - Loaded relay tplinkkasa relay named 'Channel Voron' at address '192.168.178.25' value 100.00 in 0.23 seconds.
2023-02-06 18:37:10,377 - INFO    - terrariumEngine       - Loaded relay i2c_4ch-dimmer dimmer named 'Basking Spot Rechts (100W)' at address '2,0x27' value 100.00 in 0.09 seconds.
2023-02-06 18:37:10,380 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2023-02-06 18:37:21,425 - INFO    - terrariumEngine       - Loaded 3 relays in 11.82 seconds.
2023-02-06 18:37:21,426 - INFO    - terrariumEngine       - Loading existing buttons from database.
2023-02-06 18:37:21,593 - INFO    - terrariumEngine       - Loaded magnetic button 'Türkontakt links' at address '12' value 0.00 in 0.15 seconds.
2023-02-06 18:37:21,594 - INFO    - terrariumEngine       - Loaded 1 buttons in 0.17 seconds.
2023-02-06 18:37:21,595 - INFO    - terrariumEngine       - Loading existing webcams from database.
2023-02-06 18:37:21,600 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2023-02-06 18:37:21,601 - INFO    - terrariumEngine       - Loading existing enclosures from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 156, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1053, in __load_existing_enclosures
    list(enclosure.areas)))
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 22, in __init__
    self.load_areas(areas)
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 61, in load_areas
    area_setup
  File "/home/pi/TerrariumPI/terrariumArea.py", line 131, in __init__
    self.load_setup(setup)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 920, in load_setup
    relay = self.enclosure.relays[relay]
KeyError: '5f198ac75a49e0c648aad1fca32de3b7'`

theyosh, 2023-02-06:
I have added some extra checks. Do a `git pull` and restart. I think it should startup again.

1ucket, 2023-02-06:
I've already restored a backup (so that my loved beardie doesn't sit in the dark/ cold). So I won't be able to test it. But thank you very much for the quick response, and the fix so that it won't happen again!

closed 2023-02-06 by 1ucket