theyosh/TerrariumPI #784: Relays not switching

jezzalencko, 2023-02-12:
Hi,
Just started with my build and for some reason my relay is causing me pain. There is 0.14V coming out of every GPIO and none of them will change state when I initiate a change on the GPIO relay in TPi.
I'm running out of ideas. I thought it might be due to low Amps driving it from rPi (as I'm also using a PoE hat), but have now connected the relay to an AC brick delivering 5V/3A, connected direct to relay and it still doesn't work.
It seems to be the 0.14V coming out of every GPIO that's the culprit but I'm not sure what is going on with the software to be doing this. Is that a normal state of the rPi to be in when running TPi? 
I've restarting countless times to no avail.
Please help.

theyosh, 2023-02-15:
Hi,

I do not control any voltage of the PI. So I am not sure what you are trying to do.

But if you want to control some devices though GPIO, you really need some kind of device like this: https://theyosh.github.io/TerrariumPI/hardware/relay/gpio/

Or see if there is any other relay that could fit your needs: https://theyosh.github.io/TerrariumPI/hardware/#relays

But it is unclear what you are trying to do. Sorry.

theyosh, 2023-03-10:
Is this still an issue? Any updates?

jezzalencko, 2023-03-10:
@theyosh thanks for checking in, I ended up reinstalling and everything worked. Relays are switching like they should (albeit ‘inversed’). I’m halfway through the build at the moment, so far so good. Will report back if I have any more hiccups.
Thanks!

theyosh, 2023-03-10:
Ok, than I close this issue for now. You can open a new issue when you have new questions or issues.

closed 2023-03-10 by theyosh