theyosh/TerrariumPI #790: USB Camera Offline

Konrad84, 2023-03-18:
Hi there, thank you for this Genius Projekt First. So now the issue i have a Raspberry Cam already working and now i tried to Install a USB Camera on top and it worked already on /video11 but soon after i get a Message that the Camera ist offline since a few days ago. 

You git any Idea how to fix this?

Greets from Bavaria Turtle Pi

Konrad

Konrad84, 2023-03-21:
Ok now i have fixed it. There was the wrong video address. But now i got a other Problem:

- Since i get the Pictures of the USB Cam the Live Stream of the PI CAM is broken/frozen/on repeat.. got s IR 160° degree CSI Cam + USB Standard Cam on a Raspberry 3B

Any Idea how to fix it?

theyosh, 2023-03-21:
Hi,

I am not sure, but /dev/video11 looks like is not the correct device? My  usb cam is at /dev/video0 . So that could be a solution?

Also, you could do a `git pull` to get the latest code. This should try at least 3 times to get a picture before going in offline mode. The retry logic was wrong, and after 1 try it went already offline.

Another thing I noticed, is that my webcam became more stable when using a powered USB hub. So that the webcam does not use USB power from the PI but from the USB hub.

A restart of the software will sometimes help, but if it happens a lot, it is not solution.

And If it still not stable, than I am afraid there is not much I can do. Changing the webcam for a different once it the last solution.

theyosh, 2023-04-10:
Any updates on this?

Konrad84, 2023-04-10:
Camera works a time and then Show s offline till reboot. I Made the Update to newest Version. But its OK for me i got the Live Cam which works fine.Thank you ! xD

closed 2023-04-10 by theyosh