theyosh/TerrariumPI #794: AM2320 setup issues

jezzalencko, 2023-04-05:
Hi,
I've followed the setup here: https://theyosh.github.io/TerrariumPI/hardware/sensor/am2320-sensor/
but can't seem to get any reading from my AM2320 at all.
I have tried both physical pins (3 and 5) because it wasn't clear which was to be used for Temperature (assume one is temp and the other humidity).
I also see conflicting wiring configs for the AM2320 in various places. See here: 
http://www.pibits.net/code/am2320-temperature-and-humidity-sensor-and-raspberry-pi-example.php
it shows a fritzing image where the pin arrangement is completely different. So not sure which one is correct.
[image]
At the moment I'm using the pin arrangement from the schema shown here https://theyosh.github.io/TerrariumPI/hardware/sensor/am2320-sensor/
but I don't get a reading.
[image]
Any idea what might be going on?

theyosh, 2023-04-05:
>  Any idea what might be going on?

Yes, I made wrong documentation. So the webpage you linked as source is correct. So the image on my documentation is wrong.

In order to fix this, do the following. Hook up the sensor as described in the other source page. If you have done that, you should be able to see if the sensor is connected with the following command:

`sudo i2cdetect -y 1` - https://theyosh.github.io/TerrariumPI/hardware/#i2c-bus

That should produce an output, where you should see the number `5c`. If you see this in the output, your sensor is connected correctly. Then `5c` is also the correct address in the gui. That should fix it. The code is as far I can see correct.

I will update my documentation to make sure it is correct.

jezzalencko, 2023-04-06:
Thanks @theyosh. I have hooked up the AM2320 per the the other source page, but still not getting a temp reading. i2cdetect output is as follows, not sure if this looks correct or not?
[image]

5c isn't shown anywhere.
Is there something I'm doing wrong?

theyosh, 2023-04-06:
This is not good. This is a hardware failure. Is this the only sensor that you use? If so, maybe the connectors are different. I do not own this kind of hardware myself.

When you run that i2cdetect command the outcome should look similar as on page https://theyosh.github.io/TerrariumPI/hardware/#i2c-bus . But than in your case only 1 value is entered. As long that is not the case, the hardware is not working...

jezzalencko, 2023-04-06:
so this is interesting...  I put the pins back to the config you had on your original instruction page and got this...
[image]

jezzalencko, 2023-04-06:
Bizarrely I do get a 5c showing, but only on every second i2cdetect request. Is that normal?
[image]

jezzalencko, 2023-04-06:
I'm still getting no temp reading however. Just to be sure, is this the address setting in the GUI?
[image]

theyosh, 2023-04-07:
Hmm, the settings are ok. That should work.

But it is hard for me to see if there is a code issue or a hardware issue. I do not own this piece of hardware, so it is a bit hard testing it.

The device can be powered either bij 3.3volt or 5 volt. https://pinout.xyz/pinout/5v_power# or https://pinout.xyz/pinout/3v3_power#
Maybe different voltage will make it work better. i2cdetect should not run twice before the sensor is shown..

theyosh, 2023-04-10:
So, after some debugging and testing. I think I have now the correct code. But the downside is that this sensor is shutting itself down. And waking up is not always going well. That is also why you do not see the sensor with the first i2cdetect command. That wakes the sensor up. With the second i2cdetect call you see the sensor.

But I have bought one and tested it here. I do get some values out of it, but also a lot of I2C errors about sensor not available. Not sure if that is because of my setup.

So do a `git pull` and restart TP4. Now the sensor should work.

jezzalencko, 2023-04-10:
Git pulled and restarted and yep, seems to be working now. Thanks so much.
Can be closed now.

closed 2023-04-10 by theyosh