theyosh/TerrariumPI #797: [BUG]: Shelly Relais won't work - tried different options, but no success

wrsmcode, 2023-04-16:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.6.2
 - Install: Manual
 - Raspberry PI: 3B

**Error Description**
Shelly Integration not successful yet (actually all integrated to iobroker and TerrariumPi is Linked in the UI via iFrame)

From wiki first i've tried this: http://[PASSWORD]@[HOST]#[POWER_SWITCH_NR] - but with Shelly 2. Edition it wont work.
My successful html Test:
http://192.168.2.72/Relay/0?turn=toggle  <- as Toggle configured

and your Regex (shelly_relay.py) to proof the Link looks like:
 __URL_REGEX = re.compile(r'^(?P<protocol>https?):\/\/((?P<user>[^:]+):(?P<passwd>[^@]+)@)?(?P<host>[^#\/]+)(\/)?(#(?P<nr>\d+))?$',re.IGNORECASE)

possible Links with this Regex are:
https://host/
http://host/
https://user:passwd@host/
http://user:passwd@host/
https://host/nr
http://host/nr
https://user:passwd@host/nr
http://user:passwd@host/nr

the "new" Regex should be like this:

^(?P<protocol>https?):\/\/((?P<user>[^:]+):(?P<passwd>[^@]+)@)?(?P<host>[^#\/]+)(\/)?(#(?P<nr>\d+))?$|^http:\/\/\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}\/Relay\/\d\?turn=toggle$

something like that - i've just added the "pipe" for an or Logik

and there are again 2 posibilities:

2. Shelly Gen : different types of them you can use an username AND a password for http-Login credentials, and some types are only with password Login credentials... - not really strict programming from **Allterco** 

[image]

and
[image]

would be so nice, if you can fix this with a new comming version?

great work! and have a nice day...
Joko

theyosh, 2023-04-16:
Hi,

I guess the address was not clear. But it should be `http://192.168.2.72#0` based on your test url. And this is valid based on the regex.

The only difference I see is that you have the word `relay` with uppercase R and I do not. So does this url work for you:

http://192.168.2.72/relay/0

And for now, can we test first without a password? I do see there is a small bug with password only connection.

Dave0408, 2023-04-16:
Hi, 

all I needed to configure for Shelly Plus 1

http://<ip-address>#1

Maybe [this](https://github.com/theyosh/TerrariumPI/issues/769) will help.

wrsmcode, 2023-04-16:
Hi @ all,

@Dave0408 : i've 10 Shelly's here in action, so i won't test ur first Link :/ - but, this is a good advice for the next times...

@theyosh : it will now work properly... - tested it on different shellies... - but - with your advice - without password protection. that's no problem, but my HomeAutomation iobroker "needs" a password... so far, i can use it in the first time without credentials...

AND:  i'm really suprised about your response time! - really, really good! - could be closed here so far...

have a great evening...
Joko

http://192.168.2.72#1

theyosh, 2023-04-16:
I think the password option should also work. Make sure you have the latest code and do a `git pull`. And then the following url should work: `http://[PASSWORD]@192.168.2.72/#1`

theyosh, 2023-04-17:
Uhm... the thumbs up.. does that mean that it works with a single password?

wrsmcode, 2023-04-18:
oh, sorry, my fault... i paste the old url... - i've no password protection enabled... - i've tested, but won't work.

theyosh, 2023-05-08:
sorry for the long delay.

If you have password enabled on the relay, how can you then enter the web interface? Does your browser asks for a username and password? Do you get a sort of login screen? Can you describe how you can enter the web interface when it is password enabled?

Because there are different ways of authenticating....

theyosh, 2023-05-22:
Any updates about how to access an authenticated Shelly relais? If you could explain how you can login into the web interface when it has password enabled, I can see if I can fix that in the TP4 code.

The only thing I can find is that it uses MQTT which I cannot support. That makes it rather complex.

So without extra information, I am not able to trouble shoot this....

wrsmcode, 2023-05-26:
hello Yosh,

sry for the late answer, but i was in vacancies.... - and this (long)
weekend i'm again on tour with my wife.... - i'll answer you, when i'm back
again.
BUT:
I'm actually doesn't use the login credentials. but i have a "next goal": -
wanna create an adapter for the smarthome opensource software iobroker,
that i can integrate all your features in iobroker... - we can write about
it, when i'm back at coming tuesday.

have a great time!
Joko

theyosh, 2023-06-11:
Ok, than I will close this issue now, as it is not actual anymore. If you have more information about integrating it with IOBroker, please open a new issue.

closed 2023-06-11 by theyosh