theyosh/TerrariumPI #798: [BUG]: LEDs won't turn off after automatically dimming down to 0%

Jexkin, 2023-04-17:
**Setup:**
 - TerrariumPI version: 4.6.2
 - Install: [Manual]
 - Raspberry PI: 4

**Describe the bug**
LED won't turn off after automatically dimming to 0% using the LED Warrior 18 module.
The UI shows 0% but the LEDs are still on a very low dim and won't fully turn off.
If I manually set the LED to 0% it does turn off.

**Expected behavior**
LED turns off when automatically dimming down to 0%

**Additional context**
Didn't get the chance to fully test the LED Warrior 18 module until now so please excuse the very late report of that bug.
My other LEDs (connected via a IRF520 module) do turn off when automatically dimming down to 0%, so the problem should be the LED Warrior 18 module and nothing else.
Enclosure and Area settings are basically the same, so that shouldn't be the issue either.

I am unfortunately not able to test any fixes immediately since I will be gone for a few weeks at least starting tomorrow but I figured it'd still be good to let you know about that small bug :)

theyosh, 2023-04-18:
Could you explain how you did:

>  If I manually set the LED to 0% it does turn off.

Manual in the GUI? Or trough an external script?

Jexkin, 2023-04-18:
Manual in the GUI :)

theyosh, 2023-04-18:
Ok, at least when you put it to 0% power in the GUI, it does what you expect. So that give a good feeling that this should not be hard to fix.

I will have a look into it.

Jexkin, 2023-05-08:
Hey,

thank you! The earliest I am able to test is will be on May the 20th unfortunately.

I already did a `git pull` on April 19th/20th when you commited the `Force dimmer value at end` but that didn't solve anything - just in case you are referring to that change. If there's a new one I'll test it on May the 20th :)

theyosh, 2023-05-08:
Crap, that was the fix... Damm..

Ok then I will update the code this week, that when you go to 0% it will go up to 2% and then to 0%. This will introduce a short flash. In order to see if it needs some kind of a bigger change to go fully out.

theyosh, 2023-05-13:
Ok, changed the code once more. So when you have time, do a `git pull` and restart. Lets hope a 5% ramp up before going to 0% will fix it.

Jexkin, 2023-05-13:
Will do when I'm at home again next saturday! :)

Jexkin, 2023-05-20:
Hey, just a short update - I did a `git pull` and reinstalled it.
Previously when I plugged the LEDs into power they were still on despite being set to 0%, after the git pull they were completely off.
So things look good so far but I'll wait until tonight until it dimms down for a final update :)

Jexkin, 2023-05-20:
Okay so, the LEDs now automatically turn off when they dim to 0%!

I haven't seen it in person because I was at the dinner table but they're off and that's all that matters.
I could alternatively also try to be in the reptile room to see what exactly happens tomorrow night.
But so far the goal of turning off when it dimms down is achieved! :)
Thank you!

Jexkin, 2023-05-23:
Oops, sorry, I totally forgot to update --
it does seem like everything is working as intended now, LEDs dim down as usual! :)
Thanks a lot!

theyosh, 2023-05-23:
Nice, closing this issue. Thanks for the feedback

closed 2023-05-23 by theyosh