theyosh/TerrariumPI #800: [BUG]: invalid ELF header

kaju666, 2023-04-25:
**Setup:**
 - TerrariumPI version: 4.6.1
 - Install: Manual]
 - Raspberry PI: Raspberry Pi 2 Model B Rev 1.1

**Describe the bug**
For unknow reason terrarium pi stopped to work, error on screenshot.

**Screenshots**
[image]

**Additional context**
Didnt find similar problem on closed tickets, i think its something with python?

theyosh, 2023-04-25:
I am afraid that this could be a broken SD card.

If you have made [backups](https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/), you can just try to reformat the SD card with a clean OS. Maybe that could solve it.

kaju666, 2023-04-25:
i still can connect using winscp. so i made backup as you write.
buyed new sdcard, installed, restored backup and everything work.

closed 2023-04-25 by kaju666