theyosh/TerrariumPI #801: Can´t get into Web Interface after restart

Konrad84, 2023-05-17:
Yesterday i restarted via Web Interface. Then i get an Error (Red Writing) " Not possbile to start service for 1300 times check hardware".

I get Access via Putty to the Raspberry Pi 3B and tried to reinstall in the TerrariumPI folder but no success. Till Yesterday it worked fine.

Any Idea about this ?

Greets from Bavaria

theyosh, 2023-05-17:
First stop the TP4 https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/#stop

Then run it in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

I guess some hardware is failing and that takes more time to startup. And when a startup does not complete within 2 minutes it restarts. In the debug mode it should at least start. And that will hopefully show the issue.

Konrad84, 2023-05-17:
Thank you for this lightning speed answer i will try this today evening. Now i get no connection from office.

Konrad84, 2023-05-17:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2023-05-17 18:33:38,530 - INFO    - terrariumEngine       - Starting TerrariumPI 4.6.2 ...
2023-05-17 18:33:38,601 - INFO    - terrariumEngine       - Loaded 30 settings in 0.07 seconds.
2023-05-17 18:33:38,636 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Erding,de&APPID=de7a1fbd5f74afc83284ce53b8c9b504
2023-05-17 18:33:39,982 - INFO    - weather.openweathermap_org_weather - Using Openweathermap Free API
2023-05-17 18:33:39,985 - INFO    - weather               - Loaded new weather data in 1.343 seconds.
2023-05-17 18:33:41,145 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-05-17 18:33:41,717 - INFO    - terrariumEngine       - Loaded sensor dht11 humidity named 'Schaltkasten' at address '13' with value 52.00% in 0.56 seconds.
2023-05-17 18:33:42,621 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Schildkrötenhaus' at address '28-3c01f09637c2' with value 19.50C in 0.90 seconds.
2023-05-17 18:33:43,499 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Gehege' at address '28-3c01f096b371' with value 14.38C in 0.88 seconds.
2023-05-17 18:33:44,379 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Frühbeet' at address '28-3c01f096fb7e' with value 17.75C in 0.88 seconds.
2023-05-17 18:33:44,383 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-05-17 18:33:56,261 - INFO    - terrariumEngine       - Loaded 4 sensors in 15.12 seconds.
2023-05-17 18:33:56,263 - INFO    - terrariumEngine       - Loading existing relays from database.
2023-05-17 18:33:56,902 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Tageslicht' at address '32' value 0.00 in 0.60 seconds.
2023-05-17 18:33:56,914 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Türe' at address '36' value 0.00 in 0.01 seconds.
2023-05-17 18:33:56,925 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Dach' at address '38' value 0.00 in 0.01 seconds.
2023-05-17 18:33:56,935 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Primärheizung Haus' at address '40' value 0.00 in 0.01 seconds.
2023-05-17 18:33:56,937 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2023-05-17 18:34:07,146 - INFO    - terrariumEngine       - Loaded 4 relays in 10.88 seconds.
2023-05-17 18:34:07,147 - INFO    - terrariumEngine       - Loading existing buttons from database.
2023-05-17 18:34:07,169 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2023-05-17 18:34:07,170 - INFO    - terrariumEngine       - Loading existing webcams from database.
2023-05-17 18:34:12,488 - INFO    - terrariumEngine       - Loaded usbcam webcam 'Schildkrötenhaus' at address '/dev/video1' in 5.30 seconds.
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 150, in __init__
    self.__load_existing_webcams()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 988, in __load_existing_webcams
    self.webcams[webcam.id].update(relays)
  File "/home/pi/TerrariumPI/hardware/webcam/__init__.py", line 507, in update
    self.__raw_image = Image.open(image)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/PIL/Image.py", line 3298, in open
    raise UnidentifiedImageError(msg)
PIL.UnidentifiedImageError: cannot identify image file <_io.BytesIO object at 0x65735eb0>
^CKeyboardInterrupt
2023-05-17T16:35:14Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/monkey.py", line 912, in _shutdown
    orig_shutdown()
  File "/usr/lib/python3.9/threading.py", line 1428, in _shutdown
    lock.acquire()
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
  File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 442, in gevent._gevent_c_abstract_linkable.AbstractLinkable._AbstractLinkable__wait_to_be_notified
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:

Konrad84, 2023-05-17:
That is what happened in debug mode ..

theyosh, 2023-05-17:
Ok, I think I have an idea why this is happening. I updated a python library lately, and that looks not working well...

So can you do the following:
1. Enter the TerrariumPI folder: `cd /home/pi/TerrariumPI`
2. Enable python environment: `source venv/bin/activate`
3. Downgrade python library: `pip install Pillow==9.4.0`
4. Start manual: `python terrariumPI.py`

If it now works, then the Pillow library is the cause, and I will downgrade it in the installer. And you can start it again `sudo service terrariumpi start` and should work again.

Konrad84, 2023-05-17:
pi@raspberrypi:~ $ cd /home/pi/TerrariumPI
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ pip install Pillow==9.4.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Pillow==9.4.0
  Downloading https://www.piwheels.org/simple/pillow/Pillow-9.4.0-cp39-cp39-linux_armv7l.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 419.5 kB/s eta 0:00:00
Installing collected packages: Pillow
  Attempting uninstall: Pillow
    Found existing installation: Pillow 9.5.0
    Uninstalling Pillow-9.5.0:
      Successfully uninstalled Pillow-9.5.0
Successfully installed Pillow-9.4.0
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2023-05-17 19:39:26,599 - INFO    - terrariumEngine       - Starting TerrariumPI 4.6.2 ...
2023-05-17 19:39:26,672 - INFO    - terrariumEngine       - Loaded 30 settings in 0.07 seconds.
2023-05-17 19:39:26,708 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Erding,de&APPID=de7a1fbd5f74afc83284ce53b8c9b504
2023-05-17 19:39:28,485 - INFO    - weather.openweathermap_org_weather - Using Openweathermap Free API
2023-05-17 19:39:28,487 - INFO    - weather               - Loaded new weather data in 1.772 seconds.
2023-05-17 19:39:29,667 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-05-17 19:39:30,263 - INFO    - terrariumEngine       - Loaded sensor dht11 humidity named 'Schaltkasten' at address '13' with value 61.00% in 0.58 seconds.
2023-05-17 19:39:31,149 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Schildkrötenhaus' at address '28-3c01f09637c2' with value 17.81C in 0.88 seconds.
2023-05-17 19:39:32,028 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Gehege' at address '28-3c01f096b371' with value 12.56C in 0.88 seconds.
2023-05-17 19:39:32,908 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Frühbeet' at address '28-3c01f096fb7e' with value 14.25C in 0.88 seconds.
2023-05-17 19:39:32,919 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-05-17 19:39:44,688 - INFO    - terrariumEngine       - Loaded 4 sensors in 15.02 seconds.
2023-05-17 19:39:44,690 - INFO    - terrariumEngine       - Loading existing relays from database.
2023-05-17 19:39:45,347 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Tageslicht' at address '32' value 0.00 in 0.62 seconds.
2023-05-17 19:39:45,361 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Türe' at address '36' value 0.00 in 0.01 seconds.
2023-05-17 19:39:45,374 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Dach' at address '38' value 0.00 in 0.01 seconds.
2023-05-17 19:39:45,387 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Primärheizung Haus' at address '40' value 0.00 in 0.01 seconds.
2023-05-17 19:39:45,396 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2023-05-17 19:39:55,609 - INFO    - terrariumEngine       - Loaded 4 relays in 10.92 seconds.
2023-05-17 19:39:55,611 - INFO    - terrariumEngine       - Loading existing buttons from database.
2023-05-17 19:39:55,632 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2023-05-17 19:39:55,634 - INFO    - terrariumEngine       - Loading existing webcams from database.
2023-05-17 19:40:01,109 - INFO    - terrariumEngine       - Loaded usbcam webcam 'Schildkrötenhaus' at address '/dev/video1' in 5.45 seconds.
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 150, in __init__
    self.__load_existing_webcams()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 988, in __load_existing_webcams
    self.webcams[webcam.id].update(relays)
  File "/home/pi/TerrariumPI/hardware/webcam/__init__.py", line 507, in update
    self.__raw_image = Image.open(image)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/PIL/Image.py", line 3283, in open
    raise UnidentifiedImageError(msg)
PIL.UnidentifiedImageError: cannot identify image file <_io.BytesIO object at 0x655c8d20>
^CKeyboardInterrupt
2023-05-17T17:40:55Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/monkey.py", line 912, in _shutdown
    orig_shutdown()
  File "/usr/lib/python3.9/threading.py", line 1428, in _shutdown
    lock.acquire()
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
  File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
[... cut ...]
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:

Konrad84, 2023-05-17:
Hmm same issue i think :(

theyosh, 2023-05-17:
Hmm, then I am afraid it is hardware related... do not know why...

But I have made an update to make sure the system will keeping starting up. So do a `git pull` and restart in debug mode. I expect still to see an error (3 times) and then it should continue.

Konrad84, 2023-05-17:
_____          _   _       ______ _____
                  |_   _|        | | | |      | ___ \_   _|
                    | |_   _ _ __| |_| | ___  | |_/ / | |
                    | | | | | '__| __| |/ _ \ |  __/  | |
                    | | |_| | |  | |_| |  __/ | |    _| |_
                    \_/\__,_|_|   \__|_|\___| \_|    \___/

                                                Version: 4.6.2

  Durchschnitt temperatur: 12.77 C  Betriebszeit:     5 days, 3:51:21
                                    Systemauslastung: 0.31, 0.24, 0.19
                                    CPU Temperatur:   46.16 C
                                    Speicher:         8.4 GiB(28.91%) used of total 28.9 GiB
                                    RAM:              325.2 MiB(37.35%) used of total 870.8 MiB

  Aktuelle Tätigkeit (0/4)  0.0/160.00 Watt, 0.0/4.00 L/m

    Engine can't keep up! For 1137 times it could not finish in 30.0 seconds.      Please check your setup and hardware!

      Letztes update: Tuesday, 16-05-2023 19:35:53

Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed May 17 22:00:07 2023
pi@raspberrypi:~ $ sudo service terrariumpi stop
pi@raspberrypi:~ $ cd /home/pi/TerrariumPI/
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2023-05-17 22:02:17,805 - INFO    - terrariumEngine       - Starting TerrariumPI 4.6.2 ...
2023-05-17 22:02:17,879 - INFO    - terrariumEngine       - Loaded 30 settings in 0.07 seconds.
2023-05-17 22:02:17,916 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Erding,de&APPID=de7a1fbd5f74afc83284ce53b8c9b504
2023-05-17 22:02:19,182 - INFO    - weather.openweathermap_org_weather - Using Openweathermap Free API
2023-05-17 22:02:19,184 - INFO    - weather               - Loaded new weather data in 1.262 seconds.
2023-05-17 22:02:20,330 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-05-17 22:02:20,904 - INFO    - terrariumEngine       - Loaded sensor dht11 humidity named 'Schaltkasten' at address '13' with value 60.00% in 0.56 seconds.
2023-05-17 22:02:21,761 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Schildkrötenhaus' at address '28-3c01f09637c2' with value 15.06C in 0.86 seconds.
2023-05-17 22:02:22,639 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Gehege' at address '28-3c01f096b371' with value 10.19C in 0.88 seconds.
2023-05-17 22:02:23,519 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Frühbeet' at address '28-3c01f096fb7e' with value 10.00C in 0.88 seconds.
2023-05-17 22:02:23,523 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-05-17 22:02:35,332 - INFO    - terrariumEngine       - Loaded 4 sensors in 15.00 seconds.
2023-05-17 22:02:35,334 - INFO    - terrariumEngine       - Loading existing relays from database.
2023-05-17 22:02:35,976 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Tageslicht' at address '32' value 0.00 in 0.60 seconds.
2023-05-17 22:02:35,989 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Türe' at address '36' value 0.00 in 0.01 seconds.
2023-05-17 22:02:36,000 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Dach' at address '38' value 0.00 in 0.01 seconds.
2023-05-17 22:02:36,012 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Primärheizung Haus' at address '40' value 0.00 in 0.01 seconds.
2023-05-17 22:02:36,018 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2023-05-17 22:02:46,239 - INFO    - terrariumEngine       - Loaded 4 relays in 10.90 seconds.
2023-05-17 22:02:46,240 - INFO    - terrariumEngine       - Loading existing buttons from database.
2023-05-17 22:02:46,261 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2023-05-17 22:02:46,263 - INFO    - terrariumEngine       - Loading existing webcams from database.
2023-05-17 22:02:51,526 - INFO    - terrariumEngine       - Loaded usbcam webcam 'Schildkrötenhaus' at address '/dev/video1' in 5.25 seconds.
2023-05-17 22:02:57,376 - ERROR   - hardware.webcam       - Could not process webcam image Live Raspberry PI camera named 'TURTLE' at address 'rpicam_live': cannot identify image file <_io.BytesIO object at 0x65588be0>
2023-05-17 22:02:59,070 - ERROR   - hardware.webcam       - Could not process webcam image Live Raspberry PI camera named 'TURTLE' at address 'rpicam_live': cannot identify image file <_io.BytesIO object at 0x65588d70>
2023-05-17 22:03:00,737 - ERROR   - hardware.webcam       - Could not process webcam image Live Raspberry PI camera named 'TURTLE' at address 'rpicam_live': cannot identify image file <_io.BytesIO object at 0x65597a50>
2023-05-17 22:03:01,741 - WARNING - hardware.webcam       - Webcam TURTLE has errors!
2023-05-17 22:03:01,745 - ERROR   - hardware.webcam       - Webcam TURTLE has gone offline! Please check your webcam connections.
Traceback (most recent call last):
[... cut ...]
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:

Konrad84, 2023-05-17:
Is it possible that the Webcam can cause a failure?

Konrad84, 2023-05-17:
Setup = Raspberry 3B
3 Temperature Sensor
1 Humidity Sensor
1 CSI IR Cam
1 USB Webcam
4 Relais

theyosh, 2023-05-17:
Yeah, it is the webcam... But I was hoping I can fix in such a way that at least it will startup...

theyosh, 2023-05-17:
But now both camera's have issues... that is strange...

theyosh, 2023-05-17:
Oh wait.... I thought it was the usb webcam causing problems... but it is actually the RPI camera... could there be a broken wire / connector? Or that the camera cable is not fully connected on the board any more.?

theyosh, 2023-05-17:
I made another small update. Do `git pull` and restart in debug mode. I hope it will now continue.

But I do see that this problem is already longer an issue. The message

> Engine can't keep up! For 1137 times it could not finish in 30.0 seconds

Does indicate that there was already an issue yesterday and probably already earlier... Was the RPI camera working yesterday or before that?

Konrad84, 2023-05-17:
Ok i will disconnect and reconnect the Camera USB and CSI tomorrow. Thank you Lot GN

theyosh, 2023-05-17:
FYI: I see at my installation that I have also RPI Camera errors. But only when the lights are out :( So it looks like when there is low light, it takes more time to get a image from the camera... that is stupid...
But I have at least 1 success full picture every 2 a 3 minutes.. So it is just withing offline limit.. So I did not notice it..

So, for now, I will try to make the software cope with this, and keep on running... that is the main important. And then see if there is another way to fix low light issue...

Konrad84, 2023-05-18:
The RPI Cam was working Till restart the USB Cam worked but get a time Out from time to time. I wanted to fix the USB Cam with a restart and that caused the Problem. Now i disconnected both cameras and in the evening in will Run debug Mode again. Lets See what Happens then ;)

Konrad84, 2023-05-18:
_____          _   _       ______ _____
                  |_   _|        | | | |      | ___ \_   _|
                    | |_   _ _ __| |_| | ___  | |_/ / | |
                    | | | | | '__| __| |/ _ \ |  __/  | |
                    | | |_| | |  | |_| |  __/ | |    _| |_
                    \_/\__,_|_|   \__|_|\___| \_|    \___/

                                                Version: 4.6.2

  Durchschnitt temperatur: 12.77 C  Betriebszeit:     5 days, 3:51:21
                                    Systemauslastung: 0.31, 0.24, 0.19
                                    CPU Temperatur:   46.16 C
                                    Speicher:         8.4 GiB(28.91%) used of total 28.9 GiB
                                    RAM:              325.2 MiB(37.35%) used of total 870.8 MiB

  Aktuelle Tätigkeit (0/4)  0.0/160.00 Watt, 0.0/4.00 L/m

    Engine can't keep up! For 1137 times it could not finish in 30.0 seconds.      Please check your setup and hardware!

      Letztes update: Tuesday, 16-05-2023 19:35:53

Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu May 18 07:17:13 2023
pi@raspberrypi:~ $ sudo service terrariumpi stop
pi@raspberrypi:~ $ cd /home/pi/TerrariumPI/
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2023-05-18 21:51:10,322 - INFO    - terrariumEngine       - Starting TerrariumPI 4.6.2 ...
2023-05-18 21:51:10,398 - INFO    - terrariumEngine       - Loaded 30 settings in 0.07 seconds.
2023-05-18 21:51:10,437 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Erding,de&APPID=de7a1fbd5f74afc83284ce53b8c9b504
2023-05-18 21:51:11,648 - INFO    - weather.openweathermap_org_weather - Using Openweathermap Free API
2023-05-18 21:51:11,650 - INFO    - weather               - Loaded new weather data in 1.207 seconds.
2023-05-18 21:51:12,845 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-05-18 21:51:13,418 - INFO    - terrariumEngine       - Loaded sensor dht11 humidity named 'Schaltkasten' at address '13' with value 70.00% in 0.56 seconds.
2023-05-18 21:51:14,317 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Schildkrötenhaus' at address '28-3c01f09637c2' with value 14.94C in 0.90 seconds.
2023-05-18 21:51:15,196 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Gehege' at address '28-3c01f096b371' with value 11.19C in 0.88 seconds.
2023-05-18 21:51:16,076 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'Frühbeet' at address '28-3c01f096fb7e' with value 11.06C in 0.88 seconds.
2023-05-18 21:51:16,079 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-05-18 21:51:27,891 - INFO    - terrariumEngine       - Loaded 4 sensors in 15.05 seconds.
2023-05-18 21:51:27,893 - INFO    - terrariumEngine       - Loading existing relays from database.
2023-05-18 21:51:28,543 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Tageslicht' at address '32' value 0.00 in 0.60 seconds.
2023-05-18 21:51:28,555 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Türe' at address '36' value 0.00 in 0.01 seconds.
2023-05-18 21:51:28,566 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Dach' at address '38' value 0.00 in 0.01 seconds.
2023-05-18 21:51:28,577 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Primärheizung Haus' at address '40' value 0.00 in 0.01 seconds.
2023-05-18 21:51:28,579 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2023-05-18 21:51:38,711 - INFO    - terrariumEngine       - Loaded 4 relays in 10.82 seconds.
2023-05-18 21:51:38,713 - INFO    - terrariumEngine       - Loading existing buttons from database.
2023-05-18 21:51:38,735 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2023-05-18 21:51:38,737 - INFO    - terrariumEngine       - Loading existing webcams from database.
2023-05-18 21:51:39,778 - ERROR   - terrariumEngine       - Error loading usbcam webcam 'Schildkrötenhaus' at address '/dev/video1' with error: Unable to load webcam USB camera named 'Schildkrötenhaus' at address '/dev/video1': Did not return a device..
2023-05-18 21:51:44,848 - WARNING - hardware.webcam.rpilive_webcam - Webcam Live Raspberry PI camera named 'TURTLE' at address 'rpicam_live' is crashed. Restarting the webcam.
2023-05-18 21:51:50,908 - WARNING - hardware.webcam.rpilive_webcam - Webcam Live Raspberry PI camera named 'TURTLE' at address 'rpicam_live' is crashed. Restarting the webcam.
2023-05-18 21:51:56,979 - WARNING - hardware.webcam.rpilive_webcam - Webcam Live Raspberry PI camera named 'TURTLE' at address 'rpicam_live' is crashed. Restarting the webcam.
2023-05-18 21:52:03,032 - WARNING - hardware.webcam       - Webcam TURTLE has errors!
2023-05-18 21:52:03,036 - ERROR   - hardware.webcam       - Webcam TURTLE has gone offline! Please check your webcam connections.
Traceback (most recent call last):
[... cut ...]
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:
(venv) pi@raspberrypi:~/TerrariumPI $

Konrad84, 2023-05-18:
Above is the Debug Start -- i disconnected the both cameras today but the Web Interface still doesent work. May i disconnect all sensors ? hmm

theyosh, 2023-05-18:
Could be, but that is hard to judge from here. There is a line saying that it could not write data to memory.

> File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/PIL/Image.py", line 2428, in save
fp = builtins.open(filename, "w+b")
FileNotFoundError: [Errno 2] No such file or directory: '/dev/shm/webcam/4e96d2cfe49351a5d47209f6889f3bdd/tiles/tile_3_0_0.jpg'

But I updated the code to make that folder when writing does tile images. So I think it is strange that the folder does not exists, while in code I force to make that folder... So that means that I need some more time to test this all..

But, we can also manually remove the webcams from the database. This should start it up again, but than we still do not know why those errors are. For deleting the webcams, do the following while TP4 is stopped.

1. `cd /home/pi/TerrariumPI/data`
2. `sqlite3 terrariumpi.db`
3. `delete from webcam;`
4. Ctrl+D

This should bring you back and then try running in debug again.

Konrad84, 2023-05-18:
Ok now i can get into web interface :)

But don´t know what to to with the cams maybe reconnect them and reinstall via web interface .. hmm

Konrad84, 2023-05-18:
But now i left debug mode and can´t get into interface :(

theyosh, 2023-05-18:
Start the service: https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/#start

Konrad84, 2023-05-18:
Ok now it works after a restart piu :)

Konrad84, 2023-05-19:
Hi there now i reconnected the Raspberry Cam and added it in the Interface. Now i get a Blackscreen. I tried with Whitecorrection on and off but no screen

theyosh, 2023-05-19:
Hmm, I would not add the webcams back before you know for sure they work. Now you can get the same issue again when you restart, that it will not startup.

So for testing the RPI camera, you can use the following command. Run this in the folder `/home/pi`:
`/usr/bin/raspistill --quality 95 --timeout 3000 --encoding jpg --width 1920 --height 1080 --awb greyworld --output ./test.jpg`
This should produce an image with the name `test.jpg`. Copy that to your desktop/laptop and see what the image looks like. If the image is ok (ignore rotation) then you can add it to TP4.

I use the whitebalance `greyworld` which works better for me.

Testing the USB camera is a bit harder... 

If the image is not ok, you could try a reboot of the Pi itself. But make sure you have **deleted the webcam before rebooting**...

Konrad84, 2023-05-20:
Hmm i tried the following:

today i changed the Raspberry Camera with another working camera (the usb Camera is disconnected) didn´t get a live stream instead a black screen. so i tried to test if the cam is working and get 

pi@raspberrypi:~ $ /usr/bin/raspistill --quality 95 --timeout 3000 --encoding jpg --width 1920 --height 1080 --awb greyworld --output ./test.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

pi@raspberrypi:~ $

just for sure i did a reboot and then i get i live stream from the camera with wite correction (Black/White) i switched to auto then Black screen then swiched to none and blackscreen then swichted to black/white again STRG+F5 and there is the stream BUT Stream stops after short time so i tried to ping the rapspberry maybe bad wlan connection ping was good 55ms /7ms /7ms

There must be another issue maybe the connectors i got 2 adapters csi to hdmi may i disconnect them and then try again.

i will try that tomorrow / Complete disconnect both sides change to new camera again and test

theyosh, 2023-05-20:
Hmm, also make sure you have updated the OS with the latest packages: `apt-get update && apt-get -y full-upgrade`. So you will have the latest drivers.

Konrad84, 2023-05-22:
Thank you for the hint with OS i check this today evening .. so today morning i checked the connenctions and change the camera to the old (175 degree; 2592x1944). So the following happened:
- I get a stream in endless repetition but not in time
- In the options i can´t rotate the camera and the resolution won´t be safed 

something is still not working but today evening i update drivers then we see :)

Konrad84, 2023-05-22:
OMG now i ve seen that the humidity sensor makes failures and did´t work - i removed the sensor and puff got no more connection to the raspberry .. maybe there is a solder joint that makes a wrong connection :(

Konrad84, 2023-05-22:
[image]

ok seems like the camera is crashed ....

theyosh, 2023-05-22:
You can test the live rpi camera by hand. Run:
`/home/pi/TerrariumPI/hardware/webcam/rpilive_webcam.sh`
And that should produce an output like:

[image]

It should write some chunks every 2 seconds. If this is working, than the camera should work in TP4.

Konrad84, 2023-05-22:
pi@raspberrypi:~/TerrariumPI/hardware/webcam $ ls
fonts              local_webcam.py    rpilive_webcam.sh  usblive_webcam.sh
hlslive_webcam.py  __pycache__        rpi_webcam.py      usb_webcam.py
images             remote_webcam.py   rtsp_webcam.py
__init__.py        rpilive_webcam.py  usblive_webcam.py
pi@raspberrypi:~/TerrariumPI/hardware/webcam $ ^C
pi@raspberrypi:~/TerrariumPI/hardware/webcam $ rpilive_webcam.sh
-bash: rpilive_webcam.sh: command not found
pi@raspberrypi:~/TerrariumPI/hardware/webcam $ sudo rpilive_webcam.sh
sudo: rpilive_webcam.sh: command not found
pi@raspberrypi:~/TerrariumPI/hardware/webcam $

hmm no output for me

theyosh, 2023-05-22:
Copy past the exact line I posted. That works

Konrad84, 2023-05-22:
root@raspberrypi:/home/pi# /home/pi/TerrariumPI/hardware/webcam/rpilive_webcam.sh
Input #0, h264, from 'pipe:':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, 25 fps, 25 tbr, 1200k tbn, 50 tbc
Output #0, hls, to '/dev/shm/test/stream.m3u8':
  Metadata:
    encoder         : Lavf58.45.100
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, q=2-31, 25 fps, 25 tbr, 90k tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
[hls @ 0x1c434c0] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_000.ts' for writingpeed=1.05x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_001.ts' for writingpeed=1.02x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_002.ts' for writingpeed=1.01x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_003.ts' for writingpeed=1.01x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_004.ts' for writingpeed=1.01x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_005.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_006.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_007.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_008.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_009.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_010.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_011.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_012.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_013.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_014.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_015.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_016.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_017.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_018.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_019.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_020.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_021.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_022.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
frame= 1164 fps= 25 q=-1.0 size=N/A time=00:00:46.56 bitrate=N/A speed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_023.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_024.ts' for writingpeed=   1x
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
^Cmmal: Aborting program

[hls @ 0x1c434c0] Opening '/dev/shm/test/chunk_025.ts' for writing
[hls @ 0x1c434c0] Opening '/dev/shm/test/stream.m3u8.tmp' for writing
frame= 1252 fps= 25 q=-1.0 Lsize=N/A time=00:00:50.08 bitrate=N/A speed=   1x   
video:10193kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
Exiting normally, received signal 2.

Ok i did the update and copy paste and got this above

Konrad84, 2023-05-22:
Ok after a reboot i got same problem again manual remove camera and run debug etc. -- i send the camera back to the seller and change the camera -- i let you know if this solves the problem

Thanks for your patience and for the help :)

theyosh, 2023-05-22:
Wait, the above test looks like the camera is ok. So it could be that a setting is causing this. So I would not send it back.

You can also try to just take a picture: `raspistill -o ~/test_image.jpg` should make a file in your home folder /home/pi. If the image is ok, the camera is ok.

But it is hard to debug remotely.... It can also be that my GUI does not work well with updating webcams. There could also be a bug there... but again. It takes time to figure it out..

theyosh, 2023-05-22:
Make sure all webcams are deleted. Then I can just add the RPI Live camera:

[image]

I can even change the rotation a couple of times. So it should work. I do use the white balance `greyworld` which helps a lot. So I would like to ask to put your white balance on that.

theyosh, 2023-05-22:
Hell, it can even be that the SD card is broken... so make some backups just to be sure: https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/

Sometimes I need to just reformat the SDcard and reinstall OS and TP4 to get it all working... And then I can just restore the database. But, keep in mind, this takes me about 2- 3 hours. So plan it well and with enough time to finish it.

Konrad84, 2023-05-23:
Hi ok now i connected another Cam which i can rotate and set up but got the same Problem with the stream so your right maybe it isn´t the camera 

"I get a stream in endless repetition but not in time"

now i will set up a new sd for testing

theyosh, 2023-05-23:
> "I get a stream in endless repetition but not in time"

You mean the clock on the stream is jumping back in time? Sounds like old caching data. But after a reboot of the Pi, that should be gone.

theyosh, 2023-06-02:
Hi, any updates? Is it working again?

Konrad84, 2023-06-03:
Hi there i set IT Up complete new ;) but not finished camera ist working Temperatur sensors too.I ve seen a cold solder Joint in my setup... maybe this caused Problems I send an Update when all ist running again :)

theyosh, 2023-06-25:
Hi, any updates? Is it working again? I would like to close this issue...

Konrad84, 2023-06-25:
OK done Close the Ticket if there ist Something wrong i Open a new one. CSI Cam Works, openweather Works and the Temperatur sensors Work :) Thank you lot!

closed 2023-06-25 by theyosh