theyosh/TerrariumPI #802: Page crash

jezzalencko, 2023-05-18:
Hi @theyosh, apologies if this has been addressed before, but after my terrarium pi has been running for a while (eg overnight), when I back to access page I get this error.
Any ideas what might be going on here?

Critical error while processing request: /

Error:

OSError(24, 'Too many open files')
Traceback:

Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 968, in wsgi
    out = self._cast(self._handle(environ))
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 921, in _cast
    out = self.error_handler.get(out.status_code, self.default_error_handler)(out)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 242, in handle_error
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3664, in template
    TEMPLATES[tplid] = adapter(name=tpl, lookup=lookup, **settings)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3273, in __init__
    self.prepare(**self.settings)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3374, in prepare
    self.tpl = self.env.get_template(self.filename)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/jinja2/environment.py", line 1010, in get_template
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/jinja2/environment.py", line 969, in _load_template
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/jinja2/loaders.py", line 126, in load
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/jinja2/loaders.py", line 462, in get_source
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3385, in loader
    with open(fname, "rb") as f:
OSError: [Errno 24] Too many open files: '/home/pi/TerrariumPI/public/error.html'

jezzalencko, 2023-05-22:
Thanks for that @theyosh. I've had a read and saw that "lsof -pi" could yield some clues?
I've attached the PDF of this here. Does that point towards anything in particular?
[lsof-pi_log.pdf](https://github.com/theyosh/TerrariumPI/files/11527046/lsof-pi_log.pdf)

I only have one thing connected to with i2C, and that is an AM2320 sensor. Would I be saving a lot of work by just getting rid of this and using a temp/humidity sensor that is less buggy? Any recommendations?
Thanks.

theyosh, 2023-05-22:
I do get the feeling you have installed a Desktop version of Raspberry PI. That is not supported as it behaves differently than a headless server (lite).

https://theyosh.github.io/TerrariumPI/install/ and https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy (use the lite legacy version)

It could have worked, but I do know there are issues with the desktop versions.

theyosh, 2023-05-23:
Yes, a reinstall is wiping it all. So you can make a backup by copying the following files: https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/

Then install it all over using the **Raspberry Pi OS Lite (legacy)** image. You can find it at the 'Other image' menu option.
https://theyosh.github.io/TerrariumPI/install/

Restore your date like this: https://theyosh.github.io/TerrariumPI/faq/restoring-backup-files/

jezzalencko, 2023-05-26:
@theyosh, sorry if this is a stupid question. I pulled out the SD card and can't find those folders on it, see attached.
Are they stored on the rPi itself? If so, how do I transfer them to my PC? I'm a bit of a newb. If I ssh to the tPi, what commands do I use to copy the /data and /log contents to my external PC?
[image]

theyosh, 2023-05-26:
Ah, that should be `/home/pi/TerrariumPI/data`  there you can find the files.

jezzalencko, 2023-05-26:
Thanks. I mean, once I ssh to that directory, how do I then transfer the files to another computer?

theyosh, 2023-05-26:
I am afraid not. The SD card needs to be reformatted, and reinstalled. I would suggest buy a new SD card, and start with that one. When you swap the SD cards, you also have an extra backup of the data on the old SD card.

For downloading files you can also use https://filezilla-project.org/download.php?type=client And you can use the quick connect like here https://wiki.filezilla-project.org/Using#Connecting_to_an_FTP_server but do use port **22** which is ssh.
Then you can browse on your PI, and just select files to download.

jezzalencko, 2023-05-28:
Thanks, I have done as you suggested, bought a new SD card, fresh install and transferred all files in /data over from old setup (over rid all files in the folder with the backed up files).
Have a couple of new problems now unfortunately. 
1.) My distance sensor doesn't show up at all anymore its disappeared
2.) I can't seem to login. Another login window with admin as username after I enter the rPi credentials. Do you know what the password is here?
3.) All my relays are show as 'error' and I can't click on any of them. 
Any ideas what might be going on here? See screenshots below.

Thanks.

[image]

[image]

jezzalencko, 2023-05-29:
All good, I re-started and it all seems to have returned.

theyosh, 2023-05-29:
You can also login again?

So everything is running again. So now we have to wait if it keeps running for a couple of days? Correct?

theyosh, 2023-06-02:
Is it still working? If so, can we close this issue?

jezzalencko, 2023-06-02:
I’ve been monitoring it the last few days and strange thing happens that
when I try to access it, I still get the “too many open files” error,
however it seems just “act” of trying to access it, wakes it up somehow? Is
this normal?
Once woken, I get the notification that the tPi has started and then
accessing the page works after I refresh? Does that sound like normal
behaviour?

theyosh, 2023-06-02:
Hi,

this is not ok. I do not know what is happening with your installation, but it is not working like it should. So, could you do some tests?

What is the outcome of `sudo lsof -u pi | wc -l` . It should produce a number in the 500-600 range.

What is the outcome of `sudo lsof -u pi | grep -i i2c`. It should produce 1 or 2 lines

What kind or Raspberry PI are you using?

Are you sure that you now are running with the Legacy lite OS version?? https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

Else try again with a fresh SD card, clean OS, and just reinstall it according to: https://theyosh.github.io/TerrariumPI/install/#manual and do not put back your backup. Just test it clean and see if it will just run for 24 hours. From there you could putting back your backuped database and calendar. But if you are just starting, I would suggest not to restore the backup, but just continue and reconfigure it.

jezzalencko, 2023-06-12:
> What is the outcome of `sudo lsof -u pi | wc -l` . It should produce a number in the 500-600 range.
I get 497 for this.

> What is the outcome of `sudo lsof -u pi | grep -i i2c`. It should produce 1 or 2 lines
I get 11 lines for this as follows:

python   20378   pi   18u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   19u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   20u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   31u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   33u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   34u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   35u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   39u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   46u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   53u      CHR       89,1      0t0        227 /dev/i2c-1
python   20378   pi   57u      CHR       89,1      0t0        227 /dev/i2c-1

> What kind or Raspberry PI are you using?
I'm using a Raspberry Pi 4 Model B Rev 1.1.
I also have a PoE hat installed. Not sure if this is related to the issue somehow. e.g. I wonder whether the data connection goes to "sleep". Have no way of knowing. Interestingly the Uptime is reporting as being uninterrupted. i.e. its now at 14 days, despite me having to 'wake' the connection whenever I try to access the IP address.

> Are you sure that you now are running with the Legacy lite OS version?? https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy
Yes, 100%.
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

I REALLY want to leave new install as a last resort. Setting up all my relays and sensors again would be worst outcome solution for me. i.e. I would rather "live" with having to wake the connection up every-time (assuming this 'access' issue is the only problem and that all scheduled activities were still able to take place).

theyosh, 2023-06-12:
Hmm, the output of `lsof` looks not bad. So my guess is that the initial error `OSError(24, 'Too many open files')` is gone. As 497 open files is not much. I have more on my development platform.

Also the I2C lines are not to much.

So, then I am wondering why you are not able to login to the web gui, as you are able to login to your PI with ssh. As you give me some technical info.

1. So lets try the following. Make sure TerrariumPI is stopped: https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/#stop
2. Go the the folder of TerrariumPI: `cd /home/pi/TerrariumPI/`
3. Update to the latest code: `git pull`
4. Run TerrariumPI in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/
5. And then try to visit the web gui. If there are any errors, they should show in the terminal.

If the above steps works, it should all be good, and stop the debug mode with `Ctrl+C` and start TerrariumPI service https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/#start

Because your description about waking up and starting sounds strange. I does not bring up any ideas. So else try to make some screenshots of the terminal when you think you see errors.

And when you login with ssh/terminal, doe you see this kind of message; https://raw.githubusercontent.com/theyosh/TerrariumPI/main/screenshots/terrariumpi-motd.webp

jezzalencko, 2023-06-13:
Think we might be onto something here. Sounds like it might be the AM2302 sensor. See debug errors here. Happens every couple of minutes constantly. Obviously these errors will be building up.
Is it best to replace this sensor and if so, what is least likely to cause me headache?

_2023-06-13 17:40:48,930 - WARNING - hardware.sensor       - Invalid reading from sensor AM2302 sensor temperature named 'Air Temperature' at address '15', retrying in 0.5 seconds...
2023-06-13 17:40:51,933 - WARNING - hardware.sensor       - Invalid reading from sensor AM2302 sensor temperature named 'Air Temperature' at address '15', retrying in 0.5 seconds...
2023-06-13 17:40:56,192 - ERROR   - hardware.sensor       - Sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c' has exception: [Errno 121] Remote I/O error
2023-06-13 17:40:56,194 - WARNING - hardware.sensor       - Invalid reading from sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c', retrying in 0.5 seconds...
2023-06-13 17:40:56,701 - ERROR   - hardware.sensor       - Sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c' has exception: [Errno 121] Remote I/O error
2023-06-13 17:40:56,704 - WARNING - hardware.sensor       - Invalid reading from sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c', retrying in 0.5 seconds...
2023-06-13 17:40:57,212 - ERROR   - hardware.sensor       - Sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c' has exception: [Errno 121] Remote I/O error
2023-06-13 17:40:57,215 - ERROR   - hardware.sensor       - Error updating sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c'. Check your hardware! Invalid reading from sensor AM2320 sensor temperature named 'Air Temperature (AM2320)' at address '5c'
2023-06-13 17:40:57,218 - WARNING - terrariumEngine       - Could not take a new measurement from sensor am2320 temperature named 'Air Temperature (AM2320)' at address '5c'. Tried for 1.03 seconds. Skipping this update._

theyosh, 2023-06-13:
This is not correct, and looks like a broken sensor or bad cabling. But this should **not** stop you from accessing the webgui.

So, did you try to run TerrariumPI in debug mode and did you try to access the webgui?

theyosh, 2023-06-15:
Hi, I made a code change that should work better with this AM2320 sensor. As the sensor is not stable, or it need just some start up time. But when I start TP4, I get some errors, but after a while they disappear. And the sensor does produce values.

But in the old code, every error cased an open file. And that could in the end get to much. This now not the case anymore.

So do a `git pull` to get the latest code and restart TP4. I think it should work. The sensor looks to need some more time to work properly.

jezzalencko, 2023-06-16:
okay thanks. Have done a git pull. Let me know what you want me to check to make sure its working. Seems to not be creating the error anymore.

On another matter (let me know if I need to start a new thread) but I'm lost as to how you select heater to come on and maintain a temperature in a feedback loop from temp sensor. i.e. with a target temp of say 25degC, do I need to use high and low alarms on the sensor itself? Bit confused sorry.

jezzalencko, 2023-06-20:
I think you can close this thread because I no longer get the GUI access error. Will start a new topic on my other question.

closed 2023-06-23 by theyosh