theyosh/TerrariumPI #805: Placeholders and notification syntax - distance alarm

jezzalencko, 2023-05-29:
Hi @theyosh. Hoping you can assist with how we can use the placeholder syntax for notification of a low water level alarm?
I have a distance measurement for a water tank with a high distance (low water level).
If I set up a 'sensor_alarm', how do I write the syntax to send me a notification to indicate the alarm_max for that particular sensor has been triggered? Can you please send me example syntax for this?
I've been trying to piece it together from the instructions (e.g. the button one) but am struggling.
Thanks.
[image]

theyosh, 2023-06-11:
Hi, my documentation was wrong. You have to add a `$` in front of the placeholder. https://theyosh.github.io/TerrariumPI/notifications/#messages

Only use `{[placoholders]}` from the list on the right side. Self named placeholders will not work.

theyosh, 2023-06-11:
But as you also have issue #802 I was wondering if you could give an update on that...

theyosh, 2023-06-23:
And is the documentation clear now? It is not possible to get a message for a single sensor. The alarm message is sent for every sensor that is in alarm state.

theyosh, 2023-06-30:
Is it clear how to use the placeholders? I would like to close this issue.

theyosh, 2023-07-07:
Closing due to no response

closed 2023-07-07 by theyosh