theyosh/TerrariumPI #809: [BUG]: some features do not work (for me)

cjhudlin, 2023-06-19:
**Setup:**
 - TerrariumPI version: 4.6.3
 - Install: Docker
 - Raspberry PI: 3a

**Describe the bug**
Shutdown, restart and the replace hardware button does not work

**To Reproduce / Installation steps**
click on the buttons

**Expected behavior**
to shutdown or restart the device and modify the hardware

theyosh, 2023-06-19:
Hi, the restart will/should only restart TerrariumPI. Not the Raspberry PI itself. And that is working for me. Also the shutdown, that will shutdown my Raspberry PI.
But, it can happen that the software is busy and therefore the restart is not working. The shutdown should work, as that is just a normal Raspberry Pi shutdown.
Did you do at least one reboot after installation? That is needed in order to make at least the shutdown action work.

The variation part in the area is not really tested. It was a wish of a person, when it was ready for testing, did not needed it anymore. As I did like the option, I left it. But I think you need a 'no source' the first option so it can be 'ignored'.

But the ignore slider is not for the variation part. In the variation part you change the sensor alarm values, which will cause different alarm values on which the relays than react. But some times you do not want ignore an alarm, in order to get a bit different relay toggling. At this time I am not sure what the effect is, but this is also something a user asked for.
So I will add an empty option value in the variation part. Than you can at least save the form.

The replace hardware is not implemented somehow. I have to look into that.

theyosh, 2023-06-21:
An update: Reboot and Shutdown do not work with a Docker image. That will never work. So only the restart button should work.

cjhudlin, 2023-06-21:
thank you for the feedback. yes, i use the docker version, but im thinking to install the git version as i have twice hafd to reinstall the docker version because it crashed due to some float issue with the area values, which i cant fix because i dont have access to the database to change the values:

[image]

if its possible to access the database to fix this then that would be great, so far i havent figured it out

cjhudlin, 2023-06-21:
please cold you also confirm that this time in the period box equates to 8am and not 800 minutes? as it isnt possible to place the ":" charachter here

[image]

theyosh, 2023-06-23:
Euh, I got the feeling that you need the option day/night difference with a value of -3.

That way the sensor values will drop with 3 degrees when the night begins and goes back to normal when the day starts.

But as suggest in other issue, it is easier to start over with a clean database. I am working on code updates, but I will not update the docker images much. You can try the development image tag, to get the latest code.

cjhudlin, 2023-06-24:
thank you, after changing database setting as reccomended in the FAQ for stability, this seemed to make the system more responsive and much more stable and worked in the way as intended, thank you. might be worth making that the default!

theyosh, 2023-06-28:
The database settings in the FAQ are for more stability, but it will be slower in the end. As more data is added to the database it will get slower. At least that is my experience. So it could be that the responsiveness is reducing over time.

But do you still need the day/night difference you had in the variation tab? This van be entered in the day/night difference field.

I have also updated the development docker image once more, and it should now always start up, even when the variation is not correctly entered. It should then just ignore that and continue.

And are there still issues? Other than the other open issues?

cjhudlin, 2023-07-13:
Hi,
i use the day / night difference to drop the temps in the terrarium at night, although this wasnt clear to me in the beginning, but this is working. for variation, i didnt try it as at the time i didnt seem able to get it working, but i think this was due to the DB issues i had, but i think that the variation tab is good to have because it allows for conditional changes when you want them and not determined by only day and night

theyosh, 2023-07-20:
So can  we close this issue? I think the docker issues where you started with are all fixed.

cjhudlin, 2023-07-21:
yep, i thin kthis can be closed, thank you

closed 2023-07-22 by theyosh