theyosh/TerrariumPI #810: Manual on-off time schedule, multiple events

jezzalencko, 2023-06-20:
Hi @theyosh, I've been over your instructions a few times but can't seem to work out how to setup a timing schedule for having misters come on for 'x' seconds, multiple times a day.
I don't have humidity sensors setup, so just want to manually set misters to come on for say 10 seconds, 2-3 times a day. Using Timer mode, I have no idea how to do this.
I could set ONE operating window up (as per screenshot), but how do I repeat this so there are multiple of these events in a day?
[image]

theyosh, 2023-06-20:
Ah, may I use you as a guinea pig? ;) Try the following page: https://theyosh.github.io/TerrariumPI/scenarios/ There you can enter some values. I made an example for you:

[image]

You have 3 'layers' where you can control the time.

First: Set the operating window. This is the period in which the relays are allowed to be activated. I picked 08:00 and 20:00 hours, which is a 12 hours operating window.

Second: I set the On and Off duration. As you wanted 2-3 times, I took 4 times 'a day' which is in the operating period of 12 hours. So than you use for On 15 minute, and Off 165 minutes (180 - 15). Because 180 minutes is 3 hours, and 12 / 3 is 4 times a day.

Third: I set the power on time to 15 which is 15 seconds on, and 855 seconds of settle time (15 * 60 - 15). As you have only a 15 minute on duration in which the relays are allowed to toggle on, you specify here, that it should be 15 seconds on and wait for 855 seconds for a next run. But then the 15 minute period is over, and will the relay wait for 165 minutes before it is allowed to go on again.

So, you first specify the operating window, then you specify the relay timer window and finally you specify the time it should be on, and with the settle you make it as long as the relay timer window.

On the page, the red lines are the period the relays are actually on. In the green and blue area, they are off. Play with the values to match your setup, and use the values in the form of TP4.

jezzalencko, 2023-06-21:
Hi @theyosh this makes MUCH more sense now thanks. I think I was getting a bit confused about the nomenclature of the 'on duration' and 'off duration'. This wording makes me think that the relay is ON in the 'on duration; but actually it is a CHILD sub-operating window that you can define in order to force multiple events within a PARENT operating window.
The power on time is then the active time the relay is on WITHIN the CHILD sub-operating window.

closed 2023-07-07 by theyosh