theyosh/TerrariumPI #811: [FEATURE]:

cjhudlin, 2023-06-21:
**Describe the solution you'd like**
Possiblity of using a snle relay multipl times based on conditions

**Additional context**
when using, for example, a fan, i would like this fan relay to turn on when two other relays turn on, but the two other relays relays turn at different times, so the fan should turn on when ither relay 2 or relay 3 come on, but this isnt possiblre as the relay can only be used once per condition

cjhudlin, 2023-07-12:
Hi, thank you for your reply and efforts!
my thoughts wre to have the possibility to be able to use the same relay but in other areas, so for example my cooling fan is attached to my cooling unit, so i would need to turn it on when the cooler is on, but i would like to be able to turn on the fan to us as a circulatory fan as and when id like to. in this case, i would have a cooling area and a circulatory area, but the same relay would need to be used in both areas, which urently isnt possible.

i dont expect a depending relay to come on when anther comes on, this can be done with th areas manually, and as y mntioned, probablky quite difficult

theyosh, 2023-07-31:
I have some ideas. I need some time to work them out.

oldmanbluntz, 2023-08-03:
Sweet! I just started using TPi, and was trying something similar. I have a fogger that is used to maintain humidity in a Veiled Chameleon enclosure using a relay and humidity sensors, but I also was to have the fogger turn on for a few minutes twice a day to add water droplets to the Pothos plant in there for the Cham to drink from. I was going to post an issue about this, but I see someone else wanted to do the same. If you need someone to test the new feature, i'm able to help test it as well.

I'll be posting some shots of the build this weekend!

theyosh, 2023-08-03:
Ok, I updated the software to support some kind of depending on other relays, not areas.

[image]

In the screen shot above I have selected a Heater relay, which will only go on when **one** of the two depending relays is on.
So, if you create an area with a timer setup of 8 - 10 hours a day, and then select the relays you want to depend on, the relay in the area will only go on when the dependency is met.

These depending relays can be part of an area or not. So you could also add a relay which you trigger manually, to start this area.

A few issues to take in account:
1. The order of areas is not changed so it can happen that depending relays will be switched as latest. That means that the depending area will go on the next run after 30 seconds.
2. That the relays that are depending on, should stay on for about a minimum of 60 seconds. Any shorter can be to short to trigger the relaying logic.

Code is not well tested yet. It looks promising, but there can be still some bugs.

theyosh, 2023-08-03:
This can be tested by updating the code with `git pull` and restart the software. Make sure you reload the gui with clear cache. Else you will not see the new options.

Using docker is not handy for testing.

oldmanbluntz, 2023-08-04:
I did a manual Install. I'm old school like that. Plus, i have heard of many headaches with Docker and Octoprint.

I will give it a try tomorrow after work and see how it works, and see if i can break it, so it can be fixed.

oldmanbluntz, 2023-08-05:
Yeah, thats not quite what I am looking for, and also what the OP is looking for.

When you have an area setup, and that area uses a relay, you CANNOT use that relay in any other area you try to setup, an error pops up and says that relay is in use in another area.

This is counter intuitive action. With digital timers, you can have one object, like a relay, function on multiple schedules.

Example:

I have an area setup. It uses humidity sensors to maintain humidity in the lower half of my enclosure, and when the humidity drops lower than a set percentage, it will engage the relay to operate a fogger to increase humidity. 

I want to setup a different "area". its not an area persay, as areas are just schedules in regards to terrarium pi. That "area" is just a 5 minute timer at 630 in the morning that fires off the relay that runs the fogger.

Why can you not use the same relay in multiple areas? Because your solution does not allow you to run multiple relays still. It still comes up with the same error.

oldmanbluntz, 2023-08-05:
The issue faced is when you select the relay in "relays" i n area settings, you can't select the same one in "depends on relays"....

I need to use the same relay in multiple areas. I don't need the one relay to dependent on other relays being active.

So, basically, using the same relay in multiple area schedules. I really don't know any other way to describe this.

theyosh, 2023-08-11:
The 'timers' I use are bit different. The difference is that I enforce the relay during a period. Which means, I do not only switch at a certain time, but also make sure that it stays in the correct state.

And if you start just after the timer time, the relay will not toggle in a normal timer setup. In my setup, I will enforce the relay to the correct state, even when the timer time is past.

And this logic is for me more important, than just switching a relay. I want to make sure the relay is in the correct state. And that is way you cannot use a relay twice, as a relay cannot have more than 1 state.

And yes, we can argue that this is different, but I am using it now this way for almost 10 years. And the consequence is that you can use a relay only once.

cjhudlin, 2023-08-12:
> The 'timers' I use are bit different. The difference is that I enforce the relay during a period. Which means, I do not only switch at a certain time, but also make sure that it stays in the correct [...]

Thanks for the clarification,
But while it is not possible to have the relay in more than one state at any one time, in such a case as the above requested, the relay wouldn't be a state conflict because Even if a timer were to set the relay into the same state as another area or off state, the on state of whatever is set to "on" at the time would take priority because the off state is "0" or "no signal", so it wouldn't make a difference if there were more than one area or timer used at the same time for a particular relay (in theory). The only issue I could see is that the gpio controller might not accept this, in which case it would be a case of running while loop / if statement for the areas and timers using the same gpio to make sure the correct state is set

oldmanbluntz, 2023-08-13:
> The 'timers' I use are bit different. The difference is that I enforce the relay during a period. Which means, I do not only switch at a certain time, but also make sure that it stays in the correct [...]

Would there be a way to add multiple on and off times for a single relay within a single timer program? That seems like it would be possible with the way you have the timers setup, as the single timer program would know when the relay was on and off. Maybe for ease, add the ability to hard set the amount of times the relay can cycle in a timer program?

Like if i needed the fogger to activate 3 times a day for 5 minutes, i could set a hard limit of 3 full cycles per full 24hr timer cycle, and that would add the option to add multiple on and off times for the relay. Not sure if that would be possible at all. I haven't had the time to put the code in pycharm and have a look what would be possible to do.

theyosh, 2023-08-17:
@oldmanbluntz You can create such a timer: https://theyosh.github.io/TerrariumPI/scenarios/

Play with the duration fields. Some are in minutes, other are in seconds.

An example of 3 times 5 minutes fogging a day:
[image]

cjhudlin, 2023-08-22:
> Ok, I updated the software to support some kind of depending on other relays, not areas.
>
> [image]
>
> In the screen shot above I have selected a Heater relay, which will only go on when **one** of the [...]

I just re-read what you wrote here... So this means that if one relay is on, it can also turn on the other relay too?

I like this idea, I can use this for my humidifier, when the humidifier relay is on, I can trigger the cooler fan to circulate the humidity too. That's cool

Still, I'd like to be able to use the fan for circulation though, but I won't be able to do that, unless, (if I understood what you wrote here) I create a "dummy" relay and when this relay is triggered (on a timer) then it will trigger the fan. This could also work, but I have to create a dummy relay.

Can you confirm that I understood what you said here? If this is the case, I can work with this

cjhudlin, 2023-08-22:
I also noticed that you've added a manual mode for the relays. I tried this just now with a relay that is in an area, but there is no way of knowing that it in manual mode, or the possibility of switching it back. How do I make it back to "area mode"?

theyosh, 2023-08-25:
And while I post the above message, I get an idea. It is a bit of a hassle, but I think it can be done.

But I depends an what kind of relay you are using for the fan. Can you tell me?

Because what you can do, is create 2 'script' [relays](https://theyosh.github.io/TerrariumPI/hardware/relay/script-relay/) 

Each script relay needs a custom script file on disk, which toggle the relay. You can not use the same script twice, that would fail, but with a small trick you can symlink it to a different name. And each relay script returns the value of `-1.0` a 'current' state, which means there is no current state, and **assume** that we are in the correct state.

So when you add a script relays named 'Fan normal' and 'Fan condition' which are two different scripts on disk, but both the same code and controlling the same relay, you can then have 2 areas with the 'same' fan.

So if you can tell what kind of hardware you are using, I can maybe produce the relay script files you can use.

cjhudlin, 2023-08-25:
thank you for the response. i am testing your already made feature (depends on relay). im testing the fan with the humidifier. and in theory it should also come on when the temperature is low. my next test would be to create a timed area and make a dummy relay. when this dummy relay is triggered it should also turn on the fan (hopefully). this is a work around that i can deal with. the relays i use are gpio inverse 4-relay modules

theyosh, 2023-08-31:
I created a FAQ for relay duplication. https://theyosh.github.io/TerrariumPI/faq/how-to-use-a-relay-multiple-times/

This involves creating custom scripts and using them a relay script type relays

closed 2023-09-10 by theyosh