theyosh/TerrariumPI #812: [BUG]: fix_db.py script causes db to become malformed

cjhudlin, 2023-06-22:
**Setup:**
 - TerrariumPI version: 4.6.3
 - Install: Docker
 - Raspberry PI: 3a+

**Describe the bug**
When trying to fix corrupt DB the tool makes the db malformed

**To Reproduce / Installation steps**
use the fix_db.py tool

**Expected behavior**
repair database

**Additional context**
eyboardInterrupt: 
Traceback (most recent call last):
  File "/opt/venv/lib/python3.7/site-packages/yoyo/backends/base.py", line 341, in _insert_lock_row
    {"when": datetime.utcnow(), "pid": pid},
  File "/opt/venv/lib/python3.7/site-packages/yoyo/backends/base.py", line 388, in execute
    cursor.execute(sql, params)
sqlite3.DatabaseError: database disk image is malformed
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/TerrariumPI/terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/TerrariumPI/terrariumEngine.py", line 95, in __init__
    init_db(self.version)
  File "/TerrariumPI/terrariumDatabase.py", line 39, in init
    with backend.lock():
  File "/usr/local/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "/opt/venv/lib/python3.7/site-packages/yoyo/backends/base.py", line 324, in lock
    self._insert_lock_row(pid, timeout)
  File "/opt/venv/lib/python3.7/site-packages/yoyo/backends/base.py", line 346, in _insert_lock_row
    "SELECT pid FROM {}".format(self.lock_table_quoted)
  File "/opt/venv/lib/python3.7/site-packages/yoyo/backends/base.py", line 388, in execute
    cursor.execute(sql, params)
sqlite3.DatabaseError: database disk image is malformed
KeyboardInterrupt
2023-06-22T00:10:12Z
Exception ignored in: <module 'threading' from '/usr/local/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.7/site-packages/gevent/monkey.py", line 912, in _shutdown
    orig_shutdown()
  File "/usr/local/lib/python3.7/threading.py", line 1307, in _shutdown
    lock.acquire()
  File "/opt/venv/lib/python3.7/site-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
  File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 442, in gevent._gevent_c_abstract_linkable.AbstractLinkable._AbstractLinkable__wait_to_be_notified
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:

theyosh, 2023-06-23:
Euh, I am not sure if this will work with Docker setups. I have not tested that.

But I hardly believe that my db fix script will corrupt a database. I have tested it much times and used it a couple of times. And it always worked.

But, I think in your case it is easier to clear the database and start over. As you also have some issues with the variation setup, it is easier to start over.

Just clear the data folder, which is according to the example at `/opt/terrariumpi/data`. Make sure the Docker container is stopped. Else it will go wrong again. Clear files with the command `rm /opt/terrariumpi/data/*`.

cjhudlin, 2023-06-24:
thank you, yes i did this and i started again with the system, its working now with the full sync option in the db env file and it is much more stable. but i did try this fix scripot twice and both times i got this exact error

theyosh, 2023-06-28:
Do you have the database by any chance? If so, could you mail it to github@theyosh.nl ? So I can have a try. Because it never happened to me. So maybe there is something strange with your database, and does my script not fix that.

cjhudlin, 2023-07-12:
Hi, sorry for the late reply, i have been working t build the terrarium, i will reply to the ther comments you made later, as im currently doing some work. but to answer this question, no, i dont have it, sorry. but it was only a case of running the script and then i had this issue. im not certain why

theyosh, 2023-07-13:
Than I close this issue now. I can't reproduce and therefore nothing to fix for me.

closed 2023-07-13 by theyosh