theyosh/TerrariumPI #813: [BUG]: Raspberry pi camera v3 not working

Jerrys-modz, 2023-07-04:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.6.3
 - Install: Manual]
 - Raspberry PI: 4

**Describe the bug**
The new raspberry pi camera v3 requires Bullseye and terrariumpi runs on buster.  Is there anyway to use the v3 camera on buster?

theyosh, 2023-07-07:
Hi,

I see it want libcamera2 software which I do not support yet. But it could be that you can also use it as a USB webcam. 

I am not sure, but I thought you could also use a /dev/videoX with the usb webcam. Than you have at least an still image webcam.

But I also need to upgrade the hole system in order to work on a Bullseye system fully supported. At this moment, it does not. Also the Docker container needs an overhaul.. So that will take some time.

Or, if you are handy, you can also make a small script which saves an image from the camera to disk and that you use the file base webcam in TP4. That is also a solution.

theyosh, 2023-07-15:
Euh, I did some testing again, but I cannot get a V1 or V2 camera to work with the new libcamera library. I do not know why.

By using the legacy way I can use the camera as a webcam. So you can try to use Bullseye OS and try to install TP4.
When adding a webcam, try a USB webcam (still). Use address `/dev/video0`. That could make it work.

Bullseye is not officially supported, but I think it should work. But the camera could be a bit of a hassle to get to work.

theyosh, 2023-07-31:
I cannot do any more than the above. So the only thing I can say is to try it out with different OS-es and using legacy camera which my installed already is setting up. If that does not work, then I cannot do much about it.

theyosh, 2023-08-03:
Can we close this issue? I cannot do more at the moment.

closed 2023-08-11 by theyosh