theyosh/TerrariumPI #814: Cannot add Sensors, Webcams, Areas again

oOoWatercooledoOo, 2023-07-14:
When trying to add sensors the selection form resets certain fields and won't accept changes.

**Setup:**
 - TerrariumPI version: 4.6.3
 - Install: Manual
 - Raspberry PI: 3B

**Describe the bug**
Select box resets choice

**To Reproduce / Installation steps**
Try to add sensor or webcam

**Expected behavior**
Sensor selection should accept changes and not reset previous fields

Apparently this bug was previously found to be caused by 3rd party libraries and was closed on Jan. 23

theyosh, 2023-07-14:
Somehow the package was upgraded which should not be the case. So I fixed that.

Do a `git pull` and clear your browser cache. It should be fixed again.

oOoWatercooledoOo, 2023-07-14:
Awesome.  Think this fixed it.  I was able to get my Pi camera added just have to figure out why there's no picture.  I was also able to add a sensor.

Thanks

theyosh, 2023-07-15:
Black image could be White balance related. Put white balance to auto or grayworld and it should work.

oOoWatercooledoOo, 2023-07-15:
Yeah I thought that was it also and had it set to auto,  Changed to Greyworld and it's working perfectly now.  Thanks for the help really appreciate the time and effort you have put into this project.  When I get it all setup I will post pics with my Blue Azureus Dart frogs.

theyosh, 2023-07-15:
Nice! I like those frogs... they are funny somehow...

But I can close this issue?

closed 2023-07-17 by theyosh