theyosh/TerrariumPI #816: [BUG]: Invalid behavior of relay and timer functionality

Awakawaka, 2023-07-15:
**Setup:**
 - TerrariumPI version: 4.6.3
 - Install: [Docker]
 - Raspberry PI: [4]

Hi
I was trying to set up a relay to switch on/off the main lights of the enclosure by timer. Here is my area setting page:
[image]
Note: Begin time and End time are just example values while I was trying to figure out myself.

Unfortunately, the relay does not react to area settings and, for some reason, is always on. Even if I turn it off on the 'Relay' page, it will be turned on automatically after some time. There is no other background code to control relays, so the system is doing it.

Also, for some reason, the time of the timer in the 'Enclosure' preview window is always one hour above the selected time in the 'Area' settings page:
[image]

**To Reproduce / Installation steps**

1. Add GPIO Relay
2. Setup the area as in the screenshot above

**Expected behavior**
Relay is reacting to the operating window according to area settings.

**Screenshots**
[image]

**Additional context**
Relay hat used: https://www.waveshare.com/wiki/RPi_Relay_Board

Awakawaka, 2023-07-15:
In addition, for some reason, when the relay is off hardware wise, the system thinks it is on, and vice versa
[image]
[image]
Led1 is not glowing which means the relay is 'off' state but the system treats it as the relay is in 'on' state

theyosh, 2023-07-15:
There is a `reverse GPIO` relay which will work for you. This is an old legacy bug....

About the area time difference, did you configure the time zone on your Pi correctly? And is your time zone on your laptop/desktop correct? Because you are the first one with this issue. I guess a wrong time zone on the PI

Awakawaka, 2023-07-15:
Yes, you're right. I've re-installed the system completely(including deleting volumes) and changed TZ. Also, previously reverse GPIO didn't work as expected for me, which is why I've added this issue, as I've run out of options. But just like with the TZ case now everything seems to work nicely. Thank you.

theyosh, 2023-07-15:
nice! So I can close this issue now?

Awakawaka, 2023-07-17:
Sorry, didn’t see your comment. Yes sure, go ahead. After two days I can surely say everything is working:)
Thank you for your effort and this system! I didn’t had enough time to develop back something like that in 2020 my own.  Planning to moving on RPI Zero 2w.

theyosh, 2023-07-17:
I would stay at a Pi4. I do not know the specs of a Pi Zero 2W. But I noticed that my software takes a lot of CPU, specially when using webcams. Also, I am not sure if they work totally the same as an RPI3 or RPI4. As the RPI Zero is different, and I need a tweaks in the installer to make it work. So it could be that it will not work fully out of the box. Simple issues I can maybe fix, but I do not own every piece of hardware.

closed 2023-07-17 by theyosh