theyosh/TerrariumPI #817: [BUG]:

cjhudlin, 2023-07-18:
Setup:

 - TerrariumPI version: 4.6.3
 - Install: Docker
 - Raspberry PI: 3a

**Describe the bug**
AM2303 / DHT22 sensor randomly drops out and is hard to get back

**To Reproduce / Installation steps**
Random

**Expected behavior**
work consistantly

**Additional context**
Sometimes the AM2302 sensor drops out ramdomly. this is a common issue with these sensors, and i have replaced the sensor recently but still get the same issue. doing some reading, ive heard some mention that the sensor goes into a "halted" state when too many requests are being read from the sensor. maybe the sensor polling could be adjusted from its standard 2 seconds to maybe 2.5, with a 2.5 timeout instead of 0.5, or using a different library. usually the only way to fix this is to disconnet the sensor power for a minute or so, and it comes back, but this doesnt always work and the PI itself must be restarted, restarting docker isnt enough

theyosh, 2023-07-20:
This is a bit a known issue. They are not that stable.

There is a somewhat power management option where you can shutdown the sensor after readout. This requires an extra GPIO pin which is powering the DHT sensor: https://theyosh.github.io/TerrariumPI/hardware/#power-saving

But I am not sure if the sensor needs two seconds to get the first readout. Because the timeout after powering up is 1 second. So that could be to short.

So mostly I advise the BME280 sensor. It uses I2C and it for me pretty stable for now at least 5 years.

cjhudlin, 2023-07-21:
thank you for the info, ill see about getting one. for now, i put it on a relay that restarts the sensor :D

theyosh, 2023-08-03:
Can we close this issue? I cannot do much about this.

cjhudlin, 2023-08-04:
yes sure

closed 2023-08-04 by cjhudlin