theyosh/TerrariumPI #821: [BUG]: Cannot Save Area

biohazard827, 2023-07-20:
I did a clean install of 4.6.3 and cannot seem to save a simple timer light relay. I am given an error that my enclosure, type, and mode are not set. I have setup many instances of the software and have never seen this issue. I have included a screen capture of me setting the values along with the error. I checked network traffic to see if there was some issue in the JSON being passed to be saved, but I'm not seeing anything so this must be JS form validations.

[image]

[image]

theyosh, 2023-07-20:
Strange. But did you save your old database? You can restore the database if needed.

But what kind of browser are you using?

I have to look into this. This is pretty strange...

biohazard827, 2023-07-20:
I do not have a backup, in fact my issue previously was that the DB just gave up or became corrupt. I was getting "sqlite3.operationalerror: disk i/o error" when looking at the service logs or when accessing the DB directly via console using sqlite3. I figured since there was a new update I would just start fresh.

theyosh, 2023-07-20:
I can add new areas without any issues. Using Firefox browser. So I get the feeling that clearing the browser cache should do the trick.

theyosh, 2023-07-20:
Also, there is a fix for corrupted databases: https://theyosh.github.io/TerrariumPI/faq/sqlite3-databaseerror-database-disk-image-is-malformed/

biohazard827, 2023-07-20:
It was a caching issue. Thanks for the help!

closed 2023-07-20 by biohazard827

theyosh, 2023-07-20:
No problem.