theyosh/TerrariumPI #822: [FEATURE]:

Konrad84, 2023-07-28:
Question 1# - My USB Camera caused issues so i decided to disconnect it from the raspberry pi. Is it possible to implement WLAN Stream Cameras like "NEDIS Smartlife Outdoor Cam" for example towards to the Rasp CSI connected camera in the shown site/overview webcams?

Question 2# - Is it possible to get the stream of the CSI camera implemented on an homepage maybe over no-ip service and the connection?

Question 3# - Is it possible to show for example the temperature data of a sensor on a separate homepage via API or someway else? 

I would like to make a Homepage with HTML where people see the stream and data (i would tell it is powered by terrariumpi ;-) in underline) but don´t get access to the login the relais etc. .. . Have you any suggestions to start this?    

Greets from Bavaria (Turtle PI)

theyosh, 2023-07-31:
> Question 1# - My USB Camera caused issues so i decided to disconnect it from the raspberry pi. Is it possible to implement WLAN Stream Cameras like "NEDIS Smartlife Outdoor Cam" for example towards to the Rasp CSI connected camera in the shown site/overview webcams?

I think it could be possible. According to page https://www.ispyconnect.com/camera/nedis it should be possible using RTSP. If that is the case, I can make a live streaming webcam of it. Which is not taken much time. But I do not own this kind of webcams to test it. So if you can see if you can watch the camera using VLC and a RTSP stream. If that is the case, I can incorporate it in TerrariumPI

> Question 2# - Is it possible to get the stream of the CSI camera implemented on an homepage maybe over no-ip service and the connection?

Not sure what you mean with CSI camera. But I need IP connected webcams for live streaming. If the camera is storing just still images, you need to get those images to a location where the Pi can read them. Then you can use a file based webcam.

> Question 3# - Is it possible to show for example the temperature data of a sensor on a separate homepage via API or someway else?

Yup, There is a complete API which can provide the information you need: https://theyosh.github.io/TerrariumPI/api/ But I think you need a NGINX webserver infront of TerrariumPI. There is a example config in the contrib folder. And you need to allow CORS rules in NGINX. Else you cannot show data from side A on side B.
It depends how you make the page, because all API calls are just available. But those API calls that make changes, needs a password. So it should be safe enough.

cjhudlin, 2023-08-02:
@Konrad84 , for wifi streaming, you can use the RTSP link that is provided by your camera to stream to the webcam view. as for noip, you can do the same here too. just allow the access through your router to the address of the rtsp link, though i would personally not do this unless youre working over a vpn.

theyosh, 2023-08-03:
Live RTSP camera is added to TP. It is untested as I could not find a rtsp stream to test that works at my place.

Konrad84, 2023-08-04:
Thank you i ll Test it. Have to get my VNC working first

theyosh, 2023-08-25:
Any updates? I would like to close this issue.

Konrad84, 2023-08-25:
Thank you for Patience my VNC doesnt work i tried to Install Desktop Version as Update but IT didnt Work. I ll try to Install VNC again after vacation ;) you can close the issue. Thank you

theyosh, 2023-08-25:
Why do you need VNC??? That is not handy with TerrariumPI. You only need SSH. As my code will not work with Desktop version of Raspberry PI.  https://www.tomshardware.com/reviews/raspberry-pi-headless-setup-how-to,6028.html

So VNC is not handy and I should not try to use that.

Konrad84, 2023-08-25:
Hmm ahh i was wrong don't need VNC i only should add the Camera directly xD OK inll try this soon. You can close the issue and No need for VNC your right.

closed 2023-08-25 by theyosh