theyosh/TerrariumPI #823: RPI Zero installation

Awakawaka, 2023-07-31:
Hi. Well I’m currently trying to run TerrariumPi on RPI Zero W 1.1V. I know it’s not supported properly but I’ve seen in closed issues that there are people actually using it so I decided to give it a try. Manual installation have succeeded but after reboot of pi (by the installation script) the TerrariumPi service is not starting. And under the debug process it I have the issues described here https://github.com/theyosh/TerrariumPI/issues/631. I’m pulling from default branch and in installation scripts I can see the fixes you have provided for the mentioned issue. But it seems it’s not relevant anymore. 
I’m using Legacy OS Lite so that is not also a case.

Awakawaka, 2023-07-31:
Hmmmm.. after installing `python-gevent` there is another error. Never seen this before:
[image]

Awakawaka, 2023-07-31:
Service is trying to start but failing:
[image]

Awakawaka, 2023-07-31:
Additional attempts to start. However, after the restart env is  not set with `source venv/bin/active`
[image]

theyosh, 2023-07-31:
This is a somewhat known issue with `gevent` python library and Pi Zeros. What kind of OS did you install? You need the old lite image which contains `Buster` OS. Else the installer script will probably not work correctly.

You can always do a re-install by running `sudo ./install.sh` which should just try again to install it all. After that, you can try to run it in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

Awakawaka, 2023-07-31:
I’m using Raspberry Pi OS Legacy Lite. The same is mentioned in your installation guid. Should I call ‘sudo -i’ before re-installing to be from root user perspective?

Awakawaka, 2023-07-31:
I’m using this one
[image]
Installed by using official raspberry pi imager.

Awakawaka, 2023-07-31:
In addition is it okay to use default branch for the installation? Or should I switch to another one?

theyosh, 2023-07-31:
All ok. You have the correct image. Just use the main branch from Github. So that is all normal following the installation.

You do not need to run `sudo -i`. By running the command `sudo ./install.sh` will do the same.

So, try the `sudo ./install.sh` command and then in debug mode running: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

Awakawaka, 2023-07-31:
I’ll try and get back to you with the result

Awakawaka, 2023-07-31:
Well, at least it is running:)
[image]
The actual issue was with `bcrypt` version 4.0.1 caused the error 'illegal instruction'. So I've installed version 3.2.2.
[image]

I'll start checking the workability of the hardware I'm using with RPI 4 and will let you know.

theyosh, 2023-08-01:
He, thanks for finding this. I have updated the installer script to fix this. So a next clean install should now work out of the box.

You can start it normally with: https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/

And you should good to go.

Awakawaka, 2023-08-02:
Hi @theyosh. So I just installed all hardware I'm using except for the camera. I will let it work for half the day and let you know the results.

theyosh, 2023-08-02:
Nice! Do you use a bleutooth sensor?

Awakawaka, 2023-08-02:
Unfortunately no as I don't have any, I'm using dc18b20 by one-wire-connection.

Awakawaka, 2023-08-03:
@theyosh, it's me again. After almost 24h working time I can say I'm satisfied with the results:
[image]
Note: In the graph, during the day, I accidentally unplugged my system, which is why you can see a sudden drop in temperature:)

The RPI Zero has not triggered a fan during the day, so the temperature was never too high (60 degrees). The only drawback is that, for some reason, I cannot edit the already added environment, but I can live with that by simply adding a new one. All other functionality works well and as expected. After I get my hands on the cable to connect RPI Zero to the camera, I'll run additional tests with added webcam to the existing environment and motion detection enabled. But so far, the RPI Zero can be supported. Maybe in the future, you'd like to introduce TerrariumPi Lite. Those small micro-computers would be an ideal candidate. On weekends I'll re-install the system using your updated installation code. 

Also, maybe I'll get my hands on RPI Zero 2w:)

theyosh, 2023-08-25:
Hi, any updates that I should know? And you should be able to edit an existing environment. But what do you mean with that? You can change an environment only its name, description, doors and webcams.
Each area can be changed by itself.

But I would like to close this issue if there are no further updates.

Awakawaka, 2023-08-25:
Hi, I’ve been using RPI Zero 2w for a while now with connected camera. I can say for sure it can be supporters officially. RPI Zero is a good low cost candidate for TerrariumPI but without a camera. I’ll provide additional detailed details here later today or on weekends.

Link to my set up hosted via ngrok: terrarirum.ngrok.io

theyosh, 2023-09-10:
I see it is still working. So it looks good. I will update the readme file to update the supported Raspberry Pis.

So can we close this issue? As it looks like everything is working.

Awakawaka, 2023-09-11:
I haven’t check updated installation script yet. But I believe it should all good, so yes please close the ticket. Hopefully this will be helpful for other guys too.

closed 2023-09-11 by theyosh