theyosh/TerrariumPI #825: Changing static IP address

jezzalencko, 2023-08-26:
Hi @theyosh, I've tried changing the IP address and rebooting the tPi, however it won't change for some reason.
Any idea how I can force a static IP?

theyosh, 2023-08-28:
Changing IP outside support of TerrariumPI. But a quick google search results: https://raspberrypi-guide.github.io/networking/set-up-static-ip-address

And keep the address `0.0.0.0` in the GUI. Because this means listen to the IP of the Pi. On change this value when you have multiple network devices on your PI

theyosh, 2023-09-10:
Does the above help? There is not much I can do more.

Would like to close this issue.

jezzalencko, 2023-09-10:
Haven’t had a chance to workout how to do this, but you can close the issue.

closed 2023-09-10 by theyosh