theyosh/TerrariumPI #827: [BUG]:  TP wont bootup since Sensor has been deleted

sni89, 2023-09-04:
I changed some sensors and deleted the old ones. I guess i forgot that i have some dependencies going with relays/areas and the deleted sensors. TP is now traped in loading during bootup.

could you please take a look at my output during debug mode? :) 

[image]
[image]

theyosh, 2023-09-04:
I made an update which should skip this error. But I am afraid a lot will happen still.. I can't tell what will happen when sensors are deleted when they are still being used.

So you can do now a `git pull` and run in debug again. See if it continues further.

sni89, 2023-09-04:
well its up and i can delete the areas ! ty a LOT!

theyosh, 2023-09-04:
Nice, so we where very **lucky** :P

closed 2023-09-04 by sni89