theyosh/TerrariumPI #828: [FEATURE]: Recover Telegram Bot & notifications

cmonicob, 2023-09-11:
**Is your feature request related to a problem? Please describe.**
I would like to recover the part of the notifications through a telegram bot since at the time it was very practical to know the status of the terrarium

**Describe the solution you'd like**
I would also be interested in adding logic to the module not only to obtain notifications, to be able to make real-time status queries of the terrarium, to have captures of the cameras upon request, to interact with the relays if necessary,...

**Additional context**
All this part of the logic to create an extensive Telegram bot that adapts to the functionalities of the project could take charge of its programming, and its future requests.

theyosh, 2023-09-12:
Hi,

I can add the Telegram part in Notifications, but I am not able to test. So if you can test it, I can add it back to TP4.

The part about requesting real time status, is a bit hard. Because that can introduce some security issues.

What I suggest is to create your own TelegramBot, and use the TerrariumPI api https://theyosh.github.io/TerrariumPI/api/ to get your info and make relays toggle.

But if you are able and willing to test the Telegram integration, I can try to add it.

cmonicob, 2023-09-23:
Perfect
I have the bot that I created for TP3, I can carry out the tests, also for security reasons we can filter by a list of IDs, and the idea is to maintain the consistency of the system is to interact as you mentioned through the API.
At the time I had the following commands implemented:
/getcam [cameraID] -> send an image of the camera
/getstatus -> current system status summary
/getsensors -> current sensor status summary
/getswitch -> current status summary of the switches

cmonicob, 2023-10-05:
[image]

theyosh, 2023-10-05:
Hmm this is a strange error. What will happen when you restart TP4 now in debug mode? Does it also show an error during startup?

cmonicob, 2023-10-05:
no error on startup

The problem is on => self.telegram_bot = Application.builder().token(self.setup["token"]).build()
If change self.setup["token"] for 'botToken' direct in code, run telegram bot

theyosh, 2023-10-05:
So, when you hardcode your token in the code it will work? Could it be that your token is starting with a number?

And what is now happening when you try to edit your Telegram service in the GUI? Does it still show your token and allowed users?

and if it is running, can you communicate with it?

cmonicob, 2023-10-05:
token telegram is compose 'number:str'

user id not work, if remove verification allowed users in code and edit funcions connect bot OK

```
    async def _connect(self):
        try:
            await self.telegram_bot.initialize()
            await self.telegram_bot.start()
            logger.info(f"Connected to Telegram")
            await self.telegram_bot.updater.start_polling()
        except InvalidToken as ex:
            logger.error(f"Error starting Telegram bot: {ex}")
```

And

```
    def stop(self):
        async def _stop():
            if self.telegram_bot is not None:
                try:
                    await self.telegram_bot.updater.stop()
                    await self.telegram_bot.stop()
                except Exception as ex:
                    logger.error(f"Error stopping Telegram bot: {ex}")
                finally:
                    await self.telegram_bot.shutdown()

                self.telegram_bot = None

        data = self._async.run(_stop())

        if self.__thread is not None:
            self.__thread.join()

        logger.info(f"Disconnected from Telegram")
```

cmonicob, 2023-10-05:
Sorry User List is OK, my problem is send codeId not NameUser

theyosh, 2023-10-05:
Thanks for the information and code changes. That helps a lot!

So I updated the code on a few places.

1. Fixed the GUI so that the bot token is now stored correctly.
2. The bot token and allowed users where stored in plain text in the database. Now they will be encrypted when added/updated in the GUI (so I suggest you update at least once the Telegram settings, so you token and allowed users will be encrypted)
3. Fixed the connect and stop code based on your code updates.

So, I think you have to reverse the `terrariumNotification.py` file using `git checkout terrariumNotification.py`.
Than do a `git pull` and restart in debug mode (just during testing)
Reload the GUI and make sure you have cleared the cache or hold shift while reloading.

Now you should be able to update your token, which is probably wrong, So update it with the correct token and save. I hope it will than all start to work.

theyosh, 2023-10-05:
> Sorry User List is OK, my problem is send codeId not NameUser

I am not sure what is easier to enter. Your username or userid? I do not use Telegram, so I am not sure what people finds easier.

theyosh, 2023-10-06:
Did you had time to test the latest code?

cmonicob, 2023-10-07:
For the user issue, the most practical thing is the Telegram name, which requires the user to have it configured, for this reason I usually use the Telegram ID, also at the time of computation it is more efficient via ID.
I normally allow both things and when I work by name I transform it to ID.  

The part of the /sensor and /webcam functions shows an error, I'm looking at how to pass parameters to the call

theyosh, 2023-10-07:
Do you get an error in de python debug screen? Or an error message from the bot in Telegram?

Does the /start and /help work?

/sensors should also work. It should show a list of all sensors.

Without more info, I do not know what is happening. I do not have Telegram

cmonicob, 2023-10-10:
How can I get the name of the webcams?
```
for webcam in self.engine.webcams:
    webcam.name
```
it does not work

theyosh, 2023-10-10:
I think it should be:

```
for webcam in self.engine.webcams.values():
    webcam.name
```

It is a dictionary, not a plain list

theyosh, 2023-10-11:
So, how much code do you want to add? I see you have also added buttons and relays. And fixing some stupid `trim` vs `strip` calls ;) Thanks.

But there is only with webcams a select option. Is this as an example? Because I think that way of selecting a sensor or relay is pretty nice. And I can make that code based on yours. Or did you have a different reason for the difference?
Because I can try to make the select option a bit more general and use it for all the sensors, relays and buttons.

cmonicob, 2023-10-12:
At the moment my idea is to add the necessary functions to be able to control the status of the terrarium via telegram.
With the basic functions that you could perform through the GUI without authentication.

The webcam option I chose to do it this way because in the other functions without device selector command shows a summary of all, but in the webcam you have to select one if or if.
Now I'm working to see if instead of ID you can query by name, because the for function I used in the webcam I don't like it.

```
for webcam in self.engine.webcams.values():
    if webcam_id == webcam.name:
         return await self.webcam(update, context, webcam.id)
```

I'm also reviewing how I could do a validation in the future, for example to allow extra actions in the relay in manual mode, but for this I think I would have to create a separate module of telegram and review a lot the consistency of the code to not interfere in the programming. For this reason I think that at the moment with what we have done is phenomenal to be able to control the terrarium in a fast way and all the other necessary actions are already done through VPN.

theyosh, 2023-10-12:
Ok, that is clear. So what if I update some of your code. Because I think I can optimize it a bit, which will probably also make the code look better.

Fur the future: If you want also to make actions like toggling a relay, it could be an option to have an /authentication command that expects a secret string. When correct, you can updated the connected user so that you can see it is authorized to make other actions. But for now,  lets see if we can update the code a bit more, and I will try to add the status command also.

theyosh, 2023-10-12:
So that would mean that I accept this PR now, update the code, and you can than sync your repo with mine again. Correct?

cmonicob, 2023-10-12:
Now I think I would have all the necessary functions for a correct functioning with the telegram bot.

If you can optimize the code it would be perfect

theyosh, 2023-10-13:
So I have merged the code and adjusted a bit. The most changes are in the webcamSelect function. I hope I made the correct changes. I rather do not use `not` to much. So I reversed some if else logic.

Also after seeing the code, I get the feeling that a command has always the text `/command uuid`. So that means that I can split it always on spaces, and pick the second entry (nr 1) for the uuid. If that does not exists, the id stays `None` and will trigger the listing function.

And all string concatenation are changed to array appending actions and a final join with new line.

I hope it all still works... could you test it?

theyosh, 2023-10-20:
Euh, what is the status? You changed some code? Does my optimizations work? Cause I would able to merge this in the main branch, so I can work to a new release.

theyosh, 2023-10-23:
> Currently the only thing that fails is the notification part of the system that when trying to add the following error request.json['services'].split(',') raises AttributeError: 'int' object has no attribute 'split'.

Crap, that is my fault. I changed some form processing code, and I did not test it well enough.... So I will fix that.

theyosh, 2023-10-23:
So, my develop branch should now be fixed. And you should now be able to enable messages through telegram.

theyosh, 2023-10-30:
I have done some more general updates and think we have incorporated Telegram communication for now.

Do you still have some issues left? If nor, I would like to close this issue and thank you a lot for helping!

theyosh, 2023-11-03:
Hi, I don't want to push to much, but did you had time to test the latest version? Can either be the main or develop branch. Both should be in sync. I am pretty close with my release and would be nice to know if the Telegram code is still working correctly.

cmonicob, 2023-11-04:
Updated to the latest code and the Telegram part works perfectly.

theyosh, 2023-11-04:
Cool! Thanks for the confirmation

closed 2023-11-04 by theyosh