theyosh/TerrariumPI #829: [BUG]: Database Clean-Up script appears to be broken

G4KCM, 2023-09-12:
Hi,

Just upgraded to v4.7.0 this morning all went fine. Thank you.
I decided to try the database clean-up script, after stopping TerrariumPI I followed the instructions and the script failed, screenshot attached showing output.

Regards 

[image]

theyosh, 2023-09-12:
Hmm, could you upload your database to: https://cloud.theyosh.nl/s/ZJTj4pAjcKgpe3C
All password are encrypted, so I cannot read them out.

Because I do not get this error. I guess some data is not correct in your database. So with you DB I can do some testing and debugging.

G4KCM, 2023-09-15:
Hi, just uploaded the terrarium.db file for you.  Rgds Clive

theyosh, 2023-09-15:
Thanks, got the file, and did the db_cleanup and got the same error. So I can do some testing now. Thanks for uploading.

theyosh, 2023-09-15:
Damm, that was easy... you do not use buttons, so there is no data.... Never thought about that :P 

So do a `git pull` and you should be able to clean your database.

G4KCM, 2023-09-16:
Brilliant!  Thank you!!
I will run it again next week when I have time to keep an eye on it.
Rgds

closed 2023-09-16 by G4KCM