theyosh/TerrariumPI #831: [FEATURE]: Ability to chose between NC and NO buttons

chakachim, 2023-09-19:
**Is your feature request related to a problem? Please describe.**
Not directly

**Describe the solution you'd like**
It would be convenient in certain scenarios to be able to flip the detected state of buttons. I'm trying to use ball tilt sensors with my setup and it would like to flip the configuration to where '0' is actually closed, and '1' is open.

theyosh, 2023-10-03:
The code is now also in the main branch. So do a `git pull` and restart. Then reload in the browser and you should have an inverse option for all buttons. Does this work for you?

chakachim, 2023-10-04:
Ran a git pull and rebooted. I now see the option, but when I go to click it the mouse changes to a red circle with a slash through it. 

https://github.com/theyosh/TerrariumPI/assets/21134725/afd45652-6315-458b-823a-c55ca192d380

theyosh, 2023-10-04:
Crap, my mistake. Do another `git pull` and just reload your browser. Make sure you have cleared the cache, or hold shift key while reloading.

theyosh, 2023-10-07:
Did you had time to test again? It should not work

chakachim, 2023-10-08:
Yes, the button now works, I still have permanately affix my sensor, but
this should do the trick.  Thank you for your quick response

closed 2023-10-09 by theyosh