theyosh/TerrariumPI #832: [BUG]:

DominicHartman, 2023-09-23:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.7.0
 - Install: Manual
 - Raspberry PI:  4B

**Describe the bug**
Hello, I am experiencing problems with the AM2302 sensor. Sometimes I see the temperatures coming and then it disappears again, the same applies to the humidity sensors, connecting to other ports on the Pi works the same, restarting has no effect either.

**To Reproduce / Installation steps**
I installed it as the site indicated

**Expected behavior**
I installed these sensors while building the terrarium. They cannot be replaced, but I am sure that the sensors are good. I expected that it would transmit the measurements.

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):

**Additional context**
Add any other context about the problem here.

[terrariumpi_logfile.txt](https://github.com/theyosh/TerrariumPI/files/12706872/terrariumpi_logfile.txt)

[image]
[image]
[image]

theyosh, 2023-09-25:
Hmm this is not the complete solution, but your sensors are configured incorrectly:

> 2023-09-23 18:21:16,869 - ERROR   - terrariumEngine       - Measurement for sensor am2302 humidity named 'hot zone' at address '23' of 65.00% is outside valid range 22.00% to 23.00%. Skipping this update.
2023-09-23 18:21:16,872 - ERROR   - terrariumEngine       - Measurement for sensor am2302 temperature named 'hotzone' at address '23' of 25.10C is outside valid range 27.00C to 32.00C. Skipping this update.
2023-09-23 18:21:16,875 - ERROR   - terrariumEngine       - Measurement for sensor am2302 humidity named 'cold zone' at address '7' of 67.60% is outside valid range 22.00% to 23.00%. Skipping this update.
2023-09-23 18:21:16,878 - ERROR   - terrariumEngine       - Measurement for sensor am2302 temperature named 'coldzone' at address '7' of 23.70C is outside valid range 27.00C to 32.00C. Skipping this update.

All the measured values are outside the min and max value. And therefore counted as invalid. Could it be that you mixed up the min and max limit values with the alarm values?

For the min and max **limit** value use 0 and 100 to start with. And set the alarm values to the values you had set in the limit min/max fields.

And I can't see if the other errors where there in the past. My software can work with one update every 5 minutes. So as long as the sensor is giving one value within every 5 minutes, it should work.

theyosh, 2023-10-03:
Did the above help? I can not find anything else what is wrong

DominicHartman, 2023-10-09:
I changed the value to 0/100, they are now visible more often, but how do you calibrate the sensors because the temperatures are not correct?

[image]
[image]

theyosh, 2023-10-09:
What do you mean with: "because the temperatures are not correct"

Some values are pretty low, which could just be the sensor that is having issues. In order to get rid of them, you could try to increase the lower limit value to about 15. So some measurements will then dropped. This could make the graphs a bit nicer. But this will not change the average graphs on the frontpage.

theyosh, 2023-10-24:
Any updates? I cannot understand the remark about incorrect sensor values. The only thing I can see, is that the sensor is not that stable. That is something I cannot fix.

theyosh, 2023-11-03:
Closing due to no repsonse

closed 2023-11-03 by theyosh