theyosh/TerrariumPI #835: [BUG]: Can't add relay

Jexkin, 2023-10-02:
**More info**
I can't add my LED Warrior 18 relays other channel. It has 2 channels, I set up the one when doing all the tests and it's still working. I was just about to set up the other channel but ti gives me an error.

**Setup:**
 - TerrariumPI version: 4.7.0
 - Install: [Manual]
 - Raspberry PI: 4

**Expected behavior**
Adding the relay to my dashboard.

**Screenshots**
[image]

**Additional context**
The other channel uses the Address `2,0x20` and so far it did not result in any error message when updating TPI. I'm not sure why I can't just add the other channel rn.
I also tried to use `2,0x20` for the new channel but it gave me the same error message.

theyosh, 2023-10-02:
Could it be that the remaining fields needs the value 0 ? They should be optional, but the error is indicating that a number value is missing.

Jexkin, 2023-10-02:
I just tested it and putting the values `0` in the empty fields worked!
I haven't added a relay in a long time and I assumed it'd automatically use `0` as a value if no value is given, especially because it's optional.
My fault and thanks for the quick help! :)

closed 2023-10-02 by Jexkin

theyosh, 2023-10-02:
Not your fault. The code should handle that. Software should make things easier :P . But nice that it solves the issue for now.