theyosh/TerrariumPI #838: [BUG]: Docker permanently "starting"; web interface not loading

threedom, 2023-10-07:
Generally super happy with this software, thanks again, theyosh!
After upgrading to 4.7 I ran into some issues and it seems like most of the debugging info is for the manual installation, not for docker, so I am looking for some advice here.

**Setup:**
 - TerrariumPI version: 4.7
 - Install: Docker
 - Raspberry PI: 4

**Describe the bug**
Docker image remains in "starting" state indefinitely. Web interface cannot be accessed.
Generally the controls appear to be working, but it seems odd that it is starting indefinitely and that the web interface cannot be accessed at all.
I should also say that I had this issue yesterday and after numerous restarts it started working and this morning the issue came up again.

Some related info:
- The docker container appears to be using no memory
- Occasionally the container  hits above 100% CPU
- At some point I was getting a 137 error

**To Reproduce / Installation steps**
Upgraded from 4.6.x to 4.7 via `docker compose up -d`, performed no other changes.
Then tried to log into web interface with no success.

**Troubleshooting so far**
- Stopped and restarted the docker container
- Tried to pull and update the container again
- Rebooted the Raspberry
- Reviewed logs via `docker compose logs -f`
- Tried to adjust healthcheck interval and timeout time

**Logs**

> terrariumpi-terrariumpi-1  | Traceback (most recent call last):
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumPI.py", line 19, in <module>
> terrariumpi-terrariumpi-1  |     terrariumEngine = terrariumEngine(__version__)
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumEngine.py", line 158, in __init__
> terrariumpi-terrariumpi-1  |     self.__load_existing_enclosures()
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumEngine.py", line 1212, in __load_existing_enclosures
> terrariumpi-terrariumpi-1  |     new_enclosure.update()
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumEnclosure.py", line 109, in update
> terrariumpi-terrariumpi-1  |     area_states[area_id] = self.areas[area_id].update(read_only)
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumArea.py", line 1256, in update
> terrariumpi-terrariumpi-1  |     super().update(read_only)
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumArea.py", line 659, in update
> terrariumpi-terrariumpi-1  |     difference = float(self.setup["day_night_difference"]) * (-1.0 if self.state["is_day"] else 1.0)
> terrariumpi-terrariumpi-1  | TypeError: float() argument must be a string or a number, not 'NoneType'
> terrariumpi-terrariumpi-1  | KeyboardInterrupt
> terrariumpi-terrariumpi-1  | 2023-10-07T16:37:20Z
> terrariumpi-terrariumpi-1  | Exception ignored in: <module 'threading' from '/usr/local/lib/python3.7/threading.py'>
> terrariumpi-terrariumpi-1  | Traceback (most recent call last):
> terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/gevent/monkey.py", line 912, in _shutdown
> terrariumpi-terrariumpi-1  |     orig_shutdown()
> terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/threading.py", line 1307, in _shutdown
> terrariumpi-terrariumpi-1  |     lock.acquire()
> terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/gevent/thread.py", line 121, in acquire
> terrariumpi-terrariumpi-1  |     acquired = BoundedSemaphore.acquire(self, blocking, timeout)
> terrariumpi-terrariumpi-1  |   File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
> terrariumpi-terrariumpi-1  |   File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
> terrariumpi-terrariumpi-1  |   File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
> terrariumpi-terrariumpi-1  |   File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
> terrariumpi-terrariumpi-1  |   File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
> terrariumpi-terrariumpi-1  |   File "src/gevent/_abstract_linkable.py", line 442, in gevent._gevent_c_abstract_linkable.AbstractLinkable._AbstractLinkable__wait_to_be_notified
> terrariumpi-terrariumpi-1  |   File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
> terrariumpi-terrariumpi-1  |   File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
> terrariumpi-terrariumpi-1  |   File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
> terrariumpi-terrariumpi-1  |   File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
> terrariumpi-terrariumpi-1  | KeyboardInterrupt:
> terrariumpi-terrariumpi-1  | 2023-10-07 09:38:23,892 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.7.0 on a Raspberry Pi 4 Model B Rev 1.4 ...
> terrariumpi-terrariumpi-1  | 2023-10-07 09:38:23,956 - INFO    - terrariumEngine       - Loaded 29 settings in 0.06 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 09:38:23,957 - INFO    - terrariumEngine       - Loading weather data from source https://openweathermap.org/city/5391959
> terrariumpi-terrariumpi-1  | 2023-10-07 09:38:23,960 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://openweathermap.org/city/5391959' is not valid! Please check your source
[... 522 lines cut ...]
>             "GlobalIPv6PrefixLen": 0,
>             "IPAddress": "",
>             "IPPrefixLen": 0,
>             "IPv6Gateway": "",
>             "MacAddress": "",
>             "Networks": {
>                 "host": {
>                     "IPAMConfig": null,
>                     "Links": null,
>                     "Aliases": null,
>                     "NetworkID": "5d48089b93c43a15af7afe519979863f8120e9b37eef64c282fcc203b21ee423",
>                     "EndpointID": "9798646b211be159ec5cc0b0da43d5fef78d88517981b8aa7f755d4edb7b3bff",
>                     "Gateway": "",
>                     "IPAddress": "",
>                     "IPPrefixLen": 0,
>                     "IPv6Gateway": "",
>                     "GlobalIPv6Address": "",
>                     "GlobalIPv6PrefixLen": 0,
>                     "MacAddress": "",
>                     "DriverOpts": null
>                 }
>             }
>         }
>     }
> ]

theyosh, 2023-10-07:
Hmm, strange, but I think I fixed the error. If you d a `docker compose pull` I hope you will get a new image. Try that image, and I hope it will work again.

threedom, 2023-10-07:
I just did a full clean up of the docker image and pulled it. appears to have the same issue still. I'll retry again just in case we were overlapping in timing.
Are there any steps other than performing a docker rmi on the image and a docker rm on the container to perform a clean reinstall?

threedom, 2023-10-07:
Ok yes, I am still getting a 137 error.

> docker inspect 726b516a28b9
[
    {
        "Id": "726b516a28b9318d7399783c99962076d7c6baf188341a69a7cbefd83e22aad7",
        "Created": "2023-10-07T19:15:28.673721273Z",
        "Path": "/bin/bash",
        "Args": [
            "/TerrariumPI/run.sh"
        ],
        "State": {
            "Status": "running",
            "Running": true,
            "Paused": false,
            "Restarting": false,
            "OOMKilled": false,
            "Dead": false,
            "Pid": 1640,
            "ExitCode": 0,
            "Error": "",
            "StartedAt": "2023-10-07T19:20:21.897978344Z",
            "FinishedAt": "2023-10-07T19:20:01.411065747Z",
            "Health": {
                "Status": "starting",
                "FailingStreak": 0,
                "Log": [
                    {
                        "Start": "2023-10-07T20:19:58.833066724+01:00",
                        "End": "2023-10-07T20:19:59.769180645+01:00",
                        "ExitCode": 137,
                        "Output": "Restarting unhealty docker 2023-10-07 12:19:59.363806\n"
                    }
                ]
            }
        },
        "Image": "sha256:59c96882a30356795e0f83b6f6cd3a18a97787b9a8ab176b545530ea0e1be55e",
        "ResolvConfPath": "/var/lib/docker/containers/726b516a28b9318d7399783c99962076d7c6baf188341a69a7cbefd83e22aad7/resolv.conf",
        "HostnamePath": "/var/lib/docker/containers/726b516a28b9318d7399783c99962076d7c6baf188341a69a7cbefd83e22aad7/hostname",
        "HostsPath": "/var/lib/docker/containers/726b516a28b9318d7399783c99962076d7c6baf188341a69a7cbefd83e22aad7/hosts",
        "LogPath": "/var/lib/docker/containers/726b516a28b9318d7399783c99962076d7c6baf188341a69a7cbefd83e22aad7/726b516a28b9318d7399783c99962076d7c6baf188341a69a7cbefd83e22aad7-json.log",
        "Name": "/terrariumpi-terrariumpi-1",
        "RestartCount": 1,
        "Driver": "overlay2",
        "Platform": "linux",
        "MountLabel": "",
        "ProcessLabel": "",
        "AppArmorProfile": "",
        "ExecIDs": null,
        "HostConfig": {
            "Binds": [
                "/dev:/dev:rw",
                "/opt/terrariumpi/data:/TerrariumPI/data:rw",
                "/opt/terrariumpi/media:/TerrariumPI/media:rw",
                "/opt/terrariumpi/webcam-archive:/TerrariumPI/webcam/archive:rw",
                "/opt/terrariumpi/DenkoviRelayCommandLineTool:/TerrariumPI/3rdparty/DenkoviRelayCommandLineTool:rw",
                "/boot/config.txt:/boot/config.txt:rw",
                "/etc/modules:/etc/modules:rw",
                "/opt/terrariumpi/logs:/TerrariumPI/log:rw",
                "/opt/terrariumpi/scripts:/TerrariumPI/scripts:rw",
                "/boot/cmdline.txt:/boot/cmdline.txt:rw"
            ],
            "ContainerIDFile": "",
            "LogConfig": {
                "Type": "json-file",
                "Config": {}
            },
            "NetworkMode": "host",
            "PortBindings": {},
            "RestartPolicy": {
[... 239 lines cut ...]
            "GlobalIPv6PrefixLen": 0,
            "IPAddress": "",
            "IPPrefixLen": 0,
            "IPv6Gateway": "",
            "MacAddress": "",
            "Networks": {
                "host": {
                    "IPAMConfig": null,
                    "Links": null,
                    "Aliases": null,
                    "NetworkID": "5d48089b93c43a15af7afe519979863f8120e9b37eef64c282fcc203b21ee423",
                    "EndpointID": "251d9e86822a3d63df95d894bf9fc518ee2bc8ca763d8f91908309c6c52d19bf",
                    "Gateway": "",
                    "IPAddress": "",
                    "IPPrefixLen": 0,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "",
                    "DriverOpts": null
                }
            }
        }
    }
]

threedom, 2023-10-07:
[image]

theyosh, 2023-10-07:
Above image and docker inspect is not helping. It is a code issue. So only `docker compose logs` is needed.

But, is it possible to upload your database to: https://cloud.theyosh.nl/s/ZJTj4pAjcKgpe3C So I can have a look at your area setup. I think I had fixed it in code, but apparently not. If you can share your db, it is easier for me to debug. All passwords are encrypted which I cannot read out.

threedom, 2023-10-07:
Just uploaded the db file.

theyosh, 2023-10-07:
Thanks. I made a new container image. Can you do another `docker compose pull`. It should pull some new layers. And try to start it. I hope I have tackled the python bug.

threedom, 2023-10-07:
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:16,909 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.7.0 on a Raspberry Pi 4 Model B Rev 1.4 ...
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:16,975 - INFO    - terrariumEngine       - Loaded 29 settings in 0.06 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:16,976 - INFO    - terrariumEngine       - Loading weather data from source https://openweathermap.org/city/5391959
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:16,979 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://openweathermap.org/city/5391959' is not valid! Please check your source
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:17,850 - INFO    - terrariumEngine       - Loading existing sensors from database.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:18,777 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'LT - RHP2' at address '28-01204fba80d8' with value 31.31F in 0.91 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:33,700 - WARNING - terrariumEngine       - Measurement for sensor LYWSD03MMC temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31' of 29.14F is outside valid range 30.00F to 120.00F during startup in 14.92 seconds. Will be updated in the next round.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:57,967 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with value 75.00% in 24.26 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:50:57,973 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with value 31.62F in 0.00 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:12,718 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'LT - Cold Side' at address 'a4:c1:38:8b:0b:31' with value 90.00% in 14.74 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:12,725 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'LYWSD03MMC bluetooth sensor measuring temperature' at address 'a4:c1:38:8b:0b:31' with value 29.13F in 0.00 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:27,739 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'Living Room' at address 'a4:c1:38:e2:f6:49' with value 50.00% in 15.01 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:27,747 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'Living Room' at address 'a4:c1:38:e2:f6:49' with value 26.80F in 0.01 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:27,751 - INFO    - terrariumEngine       - Scanning for new sensors ...
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:37,970 - INFO    - terrariumEngine       - Loaded 8 sensors in 80.12 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 12:51:37,971 - INFO    - terrariumEngine       - Loading existing relays from database.
> terrariumpi-terrariumpi-1  | Traceback (most recent call last):
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumPI.py", line 19, in <module>
> terrariumpi-terrariumpi-1  |     terrariumEngine = terrariumEngine(__version__)
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumEngine.py", line 137, in __init__
> terrariumpi-terrariumpi-1  |     self.__load_existing_relays()
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumEngine.py", line 811, in __load_existing_relays
> terrariumpi-terrariumpi-1  |     relay.id, relay.hardware, relay.address, relay.name, callback=self.callback_relay
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/hardware/relay/__init__.py", line 87, in __new__
> terrariumpi-terrariumpi-1  |     known_relays = terrariumRelay.available_hardware
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/terrariumUtils.py", line 47, in __get__
> terrariumpi-terrariumpi-1  |     return classmethod(self.fget).__get__(None, owner)()
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/hardware/relay/__init__.py", line 63, in available_hardware
> terrariumpi-terrariumpi-1  |     imported_module = import_module("." + file.stem, package="{}".format(__name__))
> terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
> terrariumpi-terrariumpi-1  |     return _bootstrap._gcd_import(name[level:], package, level)
> terrariumpi-terrariumpi-1  |   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
> terrariumpi-terrariumpi-1  |   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
> terrariumpi-terrariumpi-1  |   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
> terrariumpi-terrariumpi-1  |   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
> terrariumpi-terrariumpi-1  |   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
> terrariumpi-terrariumpi-1  |   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
> terrariumpi-terrariumpi-1  |   File "/TerrariumPI/hardware/relay/kasa_relay.py", line 11, in <module>
> terrariumpi-terrariumpi-1  |     from kasa import Discover, SmartStrip, SmartPlug
> terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/kasa/__init__.py", line 16, in <module>
> terrariumpi-terrariumpi-1  |     from kasa.discover import Discover
[... cut ...]
> terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/kasa/modules/antitheft.py", line 2, in <module>
> terrariumpi-terrariumpi-1  |     from .rulemodule import RuleModule
> terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/kasa/modules/rulemodule.py", line 29, in <module>
> terrariumpi-terrariumpi-1  |     class Rule(BaseModel):
> terrariumpi-terrariumpi-1  |   File "pydantic/main.py", line 186, in pydantic.main.ModelMetaclass.__new__
> terrariumpi-terrariumpi-1  | TypeError: Argument 'bases' has incorrect type (expected list, got tuple)
>

theyosh, 2023-10-07:
Not sure what is going on. I suggest going back to the latest 4.6 version. I do not know why you have all those errors. That needs more debugging from my side.

On the other side, you are the first with these errors. So that is strange.

In order to clean docker setup, make sure all docker containers are stopped. Then run `docker system prune -a` which should clear the images. You date should stay. To make sure make a backup of the data folder.

threedom, 2023-10-07:
So I tried that to no success. Sharing the log here but will be offline for a few hours. 
Thanks for the quick support. I may try to perform a clean new setup of the RPI when I get back home. 

> 
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:40,483 - INFO    - terrariumEngine       - Starting TerrariumPI 4.6.2 ...
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:40,547 - INFO    - terrariumEngine       - Loaded 29 settings in 0.06 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:40,549 - INFO    - terrariumEngine       - Loading weather data from source https://openweathermap.org/city/5391959
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:40,551 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://openweathermap.org/city/5391959' is not valid! Please check your source
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:41,296 - INFO    - terrariumEngine       - Loading existing sensors from database.
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:42,184 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'LT - RHP2' at address '28-01204fba80d8' with value 30.44F in 0.88 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:47,561 - ERROR   - hardware.sensor       - Sensor LYWSD03MMC bluetooth sensor temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31' has exception: Failed to connect to peripheral A4:C1:38:8B:0B:31, addr type: public
terrariumpi-terrariumpi-1  | 2023-10-07 13:13:47,569 - WARNING - hardware.sensor       - Invalid reading from sensor LYWSD03MMC bluetooth sensor temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31', retrying in 0.5 seconds...
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:06,751 - WARNING - terrariumEngine       - Measurement for sensor LYWSD03MMC temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31' of 28.61F is outside valid range 30.00F to 120.00F during startup in 24.57 seconds. Will be updated in the next round.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:21,533 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with value 75.00% in 14.78 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:21,540 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with value 30.73F in 0.00 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:38,912 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'LT - Cold Side' at address 'a4:c1:38:8b:0b:31' with value 88.00% in 17.37 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:38,918 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'LYWSD03MMC bluetooth sensor measuring temperature' at address 'a4:c1:38:8b:0b:31' with value 28.62F in 0.00 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:58,504 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'Living Room' at address 'a4:c1:38:e2:f6:49' with value 52.00% in 19.58 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:58,510 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'Living Room' at address 'a4:c1:38:e2:f6:49' with value 26.87F in 0.00 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:14:58,515 - INFO    - terrariumEngine       - Scanning for new sensors ...
terrariumpi-terrariumpi-1  | 2023-10-07 13:15:08,668 - INFO    - terrariumEngine       - Loaded 8 sensors in 87.37 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 13:15:08,670 - INFO    - terrariumEngine       - Loading existing relays from database.
terrariumpi-terrariumpi-1  | 2023-10-07 13:15:12,517 - ERROR   - hardware.relay.kasa_relay - Unable to connect to the device: 192.168.1.251: [Errno 113] Connect call failed ('192.168.1.251', 9999)
terrariumpi-terrariumpi-1  | Traceback (most recent call last):
terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/kasa/protocol.py", line 131, in _query
terrariumpi-terrariumpi-1  |     await self._connect(timeout)
terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/kasa/protocol.py", line 81, in _connect
terrariumpi-terrariumpi-1  |     self.reader, self.writer = await asyncio.wait_for(task, timeout=timeout)
terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
terrariumpi-terrariumpi-1  |     return fut.result()
terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/asyncio/streams.py", line 77, in open_connection
terrariumpi-terrariumpi-1  |     lambda: protocol, host, port, **kwds)
terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/asyncio/base_events.py", line 962, in create_connection
terrariumpi-terrariumpi-1  |     raise exceptions[0]
terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/asyncio/base_events.py", line 949, in create_connection
terrariumpi-terrariumpi-1  |     await self.sock_connect(sock, address)
terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 473, in sock_connect
terrariumpi-terrariumpi-1  |     return await fut
terrariumpi-terrariumpi-1  |   File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 503, in _sock_connect_cb
terrariumpi-terrariumpi-1  |     raise OSError(err, f'Connect call failed {address}')
terrariumpi-terrariumpi-1  | OSError: [Errno 113] Connect call failed ('192.168.1.251', 9999)
[... cut ...]
terrariumpi-terrariumpi-1  | 2023-10-07 13:15:49,957 - WARNING - hardware.relay        - Error changing relay Kasa Smart relay named 'Strip 2 - Snake Sun' at address '192.168.1.251,2' to state 0.0. Error: Unable to connect to the device: 192.168.1.251: [Errno 113] Connect call failed ('192.168.1.251', 9999), retrying in 0.5 seconds...
terrariumpi-terrariumpi-1  | 2023-10-07 13:15:53,079 - WARNING - hardware.relay        - Error changing relay Kasa Smart relay named 'Strip 2 - Snake Sun' at address '192.168.1.251,2' to state 0.0. Error: Unable to connect to the device: 192.168.1.251: [Errno 113] Connect call failed ('192.168.1.251', 9999), retrying in 0.5 seconds...

theyosh, 2023-10-07:
Could it be that some network devices (tplink kasa) are changed? Now I see errors about wrong IP addresses.

Also you weather url is wrong. You should change it.

theyosh, 2023-10-07:
And some settings are not correct?

> terrariumpi-terrariumpi-1 | 2023-10-07 13:14:06,751 - WARNING - terrariumEngine - Measurement for sensor LYWSD03MMC temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31' of 28.61F is [...]

theyosh, 2023-10-07:
Hmm, I am sorry that I do not have a clean solution. But I found the issue with 4.7.0 container and the error

> terrariumpi-terrariumpi-1 | 2023-10-07 12:51:37,971 - INFO - terrariumEngine - Loading existing relays from database.
terrariumpi-terrariumpi-1 | Traceback (most recent call last):
terrariumpi-terrariumpi-1 | File "/TerrariumPI/terrariumPI.py", line 19, in
terrariumpi-terrariumpi-1 | terrariumEngine = terrariumEngine(version)
terrariumpi-terrariumpi-1 | File "/TerrariumPI/terrariumEngine.py", line 137, in init
terrariumpi-terrariumpi-1 | self.__load_existing_relays()
terrariumpi-terrariumpi-1 | File "/TerrariumPI/terrariumEngine.py", line 811, in __load_existing_relays
terrariumpi-terrariumpi-1 | relay.id, relay.hardware, relay.address, relay.name, callback=self.callback_relay
terrariumpi-terrariumpi-1 | File "/TerrariumPI/hardware/relay/init.py", line 87, in new
terrariumpi-terrariumpi-1 | known_relays = terrariumRelay.available_hardware
terrariumpi-terrariumpi-1 | File "/TerrariumPI/terrariumUtils.py", line 47, in get
terrariumpi-terrariumpi-1 | return classmethod(self.fget).get(None, owner)()
terrariumpi-terrariumpi-1 | File "/TerrariumPI/hardware/relay/init.py", line 63, in available_hardware
terrariumpi-terrariumpi-1 | imported_module = import_module("." + file.stem, package="{}".format(name))
terrariumpi-terrariumpi-1 | File "/usr/local/lib/python3.7/importlib/init.py", line 127, in import_module
terrariumpi-terrariumpi-1 | return _bootstrap._gcd_import(name[level:], package, level)
terrariumpi-terrariumpi-1 | File "", line 1006, in _gcd_import
terrariumpi-terrariumpi-1 | File "", line 983, in _find_and_load
terrariumpi-terrariumpi-1 | File "", line 967, in _find_and_load_unlocked
terrariumpi-terrariumpi-1 | File "", line 677, in _load_unlocked
terrariumpi-terrariumpi-1 | File "", line 728, in exec_module
terrariumpi-terrariumpi-1 | File "", line 219, in _call_with_frames_removed
terrariumpi-terrariumpi-1 | File "/TerrariumPI/hardware/relay/kasa_relay.py", line 11, in
terrariumpi-terrariumpi-1 | from kasa import Discover, SmartStrip, SmartPlug
terrariumpi-terrariumpi-1 | File "/opt/venv/lib/python3.7/site-packages/kasa/init.py", line 16, in
terrariumpi-terrariumpi-1 | from kasa.discover import Discover
terrariumpi-terrariumpi-1 | File "/opt/venv/lib/python3.7/site-packages/kasa/discover.py", line 9, in
terrariumpi-terrariumpi-1 | from kasa.smartbulb import SmartBulb
terrariumpi-terrariumpi-1 | File "/opt/venv/lib/python3.7/site-packages/kasa/smartbulb.py", line 9, in
terrariumpi-terrariumpi-1 | from .modules import Antitheft, Cloud, Countdown, Emeter, Schedule, Time, Usage
terrariumpi-terrariumpi-1 | File "/opt/venv/lib/python3.7/site-packages/kasa/modules/init.py", line 3, in
terrariumpi-terrariumpi-1 | from .antitheft import Antitheft
terrariumpi-terrariumpi-1 | File "/opt/venv/lib/python3.7/site-packages/kasa/modules/antitheft.py", line 2, in
terrariumpi-terrariumpi-1 | from .rulemodule import RuleModule
terrariumpi-terrariumpi-1 | File "/opt/venv/lib/python3.7/site-packages/kasa/modules/rulemodule.py", line 29, in
terrariumpi-terrariumpi-1 | class Rule(BaseModel):
terrariumpi-terrariumpi-1 | File "pydantic/main.py", line 186, in pydantic.main.ModelMetaclass.new
terrariumpi-terrariumpi-1 | TypeError: Argument 'bases' has incorrect type (expected list, got tuple)

So I made another container with the tag 4.7.0 and latest. So do another `docker composer pull` and see if it helps.... I hope it will at least start up again.

threedom, 2023-10-08:
> And some settings are not correct?

So the settings are weird - I am located in the US and so all my probes and limits are in Fahrenheit. However, it seems like since the issue with the container (aka since this morning) the values from the probes are in Celsius while TerrariumPI still thinks that the units for both probes and limits are Fahrenheit. So it reads something like 30F which is actually 30C and 86F. Idk where that error is coming from.

threedom, 2023-10-08:
> Could it be that some network devices (tplink kasa) are changed? Now I see errors about wrong IP addresses.
> 
> Also you weather url is wrong. You should change it.

I'm not really using the weather data but fair point.

Regarding the IPs - all the devices were working before but they are relatively new. I had replaced my individual Kasa power outlets with one singular connected power strip with 6 outlets. but it was functioning fine.

threedom, 2023-10-08:
The good news is, your update got it to start the container again!
Everything seems to be working just as intended for now. I'll keep an eye on it and will update if any issues pop up. 

Here is the log after:

> 
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:04,302 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.7.0 on a Raspberry Pi 4 Model B Rev 1.4 ...
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:08,242 - INFO    - terrariumEngine       - Loaded 29 settings in 3.94 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:08,245 - INFO    - terrariumEngine       - Loading weather data from source https://openweathermap.org/city/5391959
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:08,249 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://openweathermap.org/city/5391959' is not valid! Pleas                                                                                                                                                                                                                                              e check your source
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:15,595 - INFO    - terrariumEngine       - Loading existing sensors from database.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:16,504 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named 'LT - RHP2' at address '28-01204fba80d8' with value 30.00F in                                                                                                                                                                                                                                               0.90 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:32,956 - WARNING - terrariumEngine       - Measurement for sensor LYWSD03MMC temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31'                                                                                                                                                                                                                                               of 28.55F is outside valid range 30.00F to 120.00F during startup in 16.45 seconds. Will be updated in the next round.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:37,881 - ERROR   - hardware.sensor       - Sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' ha                                                                                                                                                                                                                                              s exception: Failed to connect to peripheral a4:c1:38:5c:68:44, addr type: public
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:37,886 - WARNING - hardware.sensor       - Invalid reading from sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address '                                                                                                                                                                                                                                              a4:c1:38:5c:68:44', retrying in 0.5 seconds...
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:46,180 - ERROR   - hardware.sensor       - Sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' ha                                                                                                                                                                                                                                              s exception: Failed to connect to peripheral a4:c1:38:5c:68:44, addr type: public
> terrariumpi-terrariumpi-1  | 2023-10-07 17:52:46,184 - WARNING - hardware.sensor       - Invalid reading from sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address '                                                                                                                                                                                                                                              a4:c1:38:5c:68:44', retrying in 0.5 seconds...
> terrariumpi-terrariumpi-1  | 2023-10-07 17:53:05,640 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with value 84.00% in 32.68 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:53:05,720 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with value 30.00F in 0.08 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:53:18,441 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'LT - Cold Side' at address 'a4:c1:38:8b:0b:31' with value 95.00% in 12.72 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:53:18,448 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC temperature named 'LYWSD03MMC bluetooth sensor measuring temperature' at address 'a4:c1:38:8b:0b:31' with value 28.52F in 0.00 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:53:38,129 - INFO    - terrariumEngine       - Loaded sensor LYWSD03MMC humidity named 'Living Room' at address 'a4:c1:38:e2:f6:49' with value 54.00% in 19.68 seconds.
[... cut ...]
> terrariumpi-terrariumpi-1  | 2023-10-07 17:54:11,708 - INFO    - terrariumEngine       - Loaded 1 enclosures in 2.06 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:54:11,831 - INFO    - terrariumEngine       - TerrariumPI is up and running at address: http://192.168.1.75:8090 in 129.32 seconds.
> terrariumpi-terrariumpi-1  | 2023-10-07 17:54:11,000 - INFO    - terrariumWebserver    - Running webserver at 192.168.1.75:8090
> terrariumpi-terrariumpi-1  | 2023-10-07 17:55:28,485 - WARNING - terrariumEngine       - Engine update took 76.39 seconds. That is 46.39 seconds short.
>

threedom, 2023-10-08:
Just in the interest of sharing info to possibly fix things that may hit other people, in going through the logs I noticed a few things:

- the logs in the web interface and in the console give different sets of information (copied below): Logs in the web interface show accurate readings with a few errors whereas my `docker compose logs -f` are littered with errors and warnings
- While the interface seems to be working fine in terms of regulating the vivarium, the bluetooth sensors seem to be rather unreliable with their updates (I believe I saw you point that out before, theyosh)
- Currently, it seems like the overall update takes significantly longer than the intended 30s. In terms of resources the RPI has plenty headroom, so I am not sure why it takes so long to perform the refresh
- Startup of the container until it reaches healthy state takes a while (easily 3-5 minutes) if it is even successful. I have to restart the container (or the whole pi) a few times until it works so it doesn't feel fully reliable yet

(and to be clear, no immediate issue on my side from what I see but I thought I'd share it in case it helps with the search for bugs)

Thank you @theyosh 

Web interface: 

```
2023-10-07 18:40:08,009 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC temperature named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with new value 86.07F in 0.02 seconds.
2023-10-07 18:40:07,885 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' with new value 85.00% in 17.80 seconds.
2023-10-07 18:39:54,459 - WARNING - hardware.sensor       - Invalid reading from sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44', retrying in 0.5 seconds...
2023-10-07 18:39:54,452 - ERROR   - hardware.sensor       - Sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' has exception: Failed to connect to peripheral a4:c1:38:5c:68:44, addr type: public
2023-10-07 18:39:49,978 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC temperature named 'LT - Cold Side' at address 'A4:C1:38:8B:0B:31' with new value 83.21F in 14.88 seconds.
2023-10-07 18:39:37,111 - INFO    - terrariumEngine       - Updated relay gpio relay named 'ST - Lamp' at address '33' with new value 0.00 in 0.01 seconds.
2023-10-07 18:39:36,993 - INFO    - terrariumEngine       - Updated relay gpio relay named 'ST - Matt' at address '32' with new value 0.00 in 0.01 seconds.
2023-10-07 18:39:36,876 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Strip 6' at address '192.168.1.251,6' with new value 100.00 in 0.01 seconds.
2023-10-07 18:39:36,759 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Strip 5' at address '192.168.1.251,5' with new value 0.00 in 0.01 seconds.
2023-10-07 18:39:36,642 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Strip 4 - Growlights' at address '192.168.1.251,4' with new value 100.00 in 0.01 seconds.
2023-10-07 18:39:36,526 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Strip 3 - Waterfall' at address '192.168.1.251,3' with new value 0.00 in 0.01 seconds.
2023-10-07 18:39:36,410 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Strip 2 - Snake Sun' at address '192.168.1.251,2' with new value 100.00 in 0.01 seconds.
2023-10-07 18:39:36,291 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Strip 1 - Channel Snake Heat Panel' at address '192.168.1.251,1' with new value 0.00 in 1.32 seconds.
2023-10-07 18:39:34,992 - INFO    - terrariumEngine       - Updated sensor 1wire temperature named 'LT - RHP2' at address '28-01204fba80d8' with new value 86.00F in 0.90 seconds.
2023-10-07 18:39:34,079 - INFO    - terrariumEngine       - Starting a new update round with 8 sensors, 8 relays, 0 buttons and 0 webcams.
2023-10-07 18:39:34,078 - WARNING - terrariumEngine       - Engine update took 79.82 seconds. That is 49.82 seconds short.
2023-10-07 18:39:33,950 - INFO    - terrariumEngine       - Updated Enclosure LT with 4 areas in 0.03 seconds.
2023-10-07 18:39:33,942 - INFO    - terrariumArea         - Updated area Heating named LT Heat Panel in sensors mode at enclosure LT in 0.01 seconds.
2023-10-07 18:39:33,942 - INFO    - terrariumArea         - Relays for Heating named LT Heat Panel period low are not switched on because at least one of the depending areas is in an alarm state.
2023-10-07 18:39:33,941 - INFO    - terrariumArea         - Depending area Cold Side is in alarm state for area Heating named LT Heat Panel. Current: 83.23F
2023-10-07 18:39:33,933 - INFO    - terrariumArea         - Updated area Moisture named Waterfall Pump in timer mode at enclosure LT in 0.00 seconds.
2023-10-07 18:39:33,932 - INFO    - terrariumArea         - Relays for Moisture named Waterfall Pump period low are not switched on because we have to wait for 1699.0 more seconds of the total settle time of 3000.0 seconds.
2023-10-07 18:39:33,931 - INFO    - terrariumArea         - Updated area Heating named Cold Side in sensors mode at enclosure LT in 0.01 seconds.
2023-10-07 18:39:33,921 - INFO    - terrariumArea         - Updated area Lighting named LT Sunlight in timer mode at enclosure LT in 0.00 seconds.
[... cut ...]
terrariumpi-terrariumpi-1  | 2023-10-07 19:06:31,153 - ERROR   - hardware.sensor       - Sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44' has exception: Failed to connect to peripheral a4:c1:38:5c:68:44, addr type: public
terrariumpi-terrariumpi-1  | 2023-10-07 19:06:31,159 - WARNING - hardware.sensor       - Invalid reading from sensor LYWSD03MMC bluetooth sensor humidity named 'LT - Warm Side' at address 'a4:c1:38:5c:68:44', retrying in 0.5 seconds...

```

threedom, 2023-10-08:
And not to overload this thread but:
I just tried to change the name for one of my sensors (in the web interface) and it was apparently too much for the container, it shut down, and restarted.

Logs

```
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,618 - INFO    - terrariumEngine       - Stopped Kasa Smart relay named 'Strip 2 - Snake Sun' at address '192.168.1.251,2'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,619 - INFO    - terrariumEngine       - Stopped Kasa Smart relay named 'Strip 3 - Waterfall' at address '192.168.1.251,3'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,619 - INFO    - terrariumEngine       - Stopped Kasa Smart relay named 'Strip 4 - Growlights' at address '192.168.1.251,4'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,619 - INFO    - terrariumEngine       - Stopped Kasa Smart relay named 'Strip 5' at address '192.168.1.251,5'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,620 - INFO    - terrariumEngine       - Stopped Kasa Smart relay named 'Strip 6' at address '192.168.1.251,6'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,621 - INFO    - terrariumEngine       - Stopped GPIO devices relay named 'ST - Matt' at address '32'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,622 - INFO    - terrariumEngine       - Stopped GPIO devices relay named 'ST - Lamp' at address '33'
terrariumpi-terrariumpi-1  | 2023-10-07 19:16:49,623 - INFO    - terrariumEngine       - Stopped TerrariumPI 4.7.0 after running for 0:13:02. Bye bye.
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=19>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=20>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=21>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=22>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=23>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=24>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=25>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=26>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=27>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | /usr/local/lib/python3.7/asyncio/selector_events.py:668: ResourceWarning: unclosed transport <_SelectorSocketTransport fd=28>
terrariumpi-terrariumpi-1  |   source=self)
terrariumpi-terrariumpi-1  | ResourceWarning: Enable tracemalloc to get the object allocation traceback
terrariumpi-terrariumpi-1  | 2023-10-07 19:17:16,543 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.7.0 on a Raspberry Pi 4 Model B Rev 1.4 ...
terrariumpi-terrariumpi-1  | 2023-10-07 19:17:16,607 - INFO    - terrariumEngine       - Loaded 29 settings in 0.06 seconds.
terrariumpi-terrariumpi-1  | 2023-10-07 19:17:16,609 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=[San Francisco],[USA]&appid=[fa759460a68282328fc58122412a9e2b]
terrariumpi-terrariumpi-1  | 2023-10-07 19:17:16,611 - ERROR   - terrariumEngine       - Loading weather exception: Weather url 'https://api.openweathermap.org/data/2.5/weather?q=[San Francisco],[USA]&appid=[fa759460a68282328fc58122412a9e2b]' is not valid! Please check your source
terrariumpi-terrariumpi-1  | 2023-10-07 19:17:17,488 - INFO    - terrariumEngine       - Loading existing sensors from database.
[... cut ...]
terrariumpi-terrariumpi-1  |   File "/opt/venv/lib/python3.7/site-packages/kasa/protocol.py", line 141, in _query
terrariumpi-terrariumpi-1  |     f"Unable to connect to the device: {self.host}: {ex}"
terrariumpi-terrariumpi-1  | kasa.exceptions.SmartDeviceException: Unable to connect to the device: 192.168.1.119: [Errno 113] Connect call failed ('192.168.1.119', 9999)
terrariumpi-terrariumpi-1  | 2023-10-07 19:19:13,841 - WARNING - hardware.relay        - Error getting new data from relay Kasa Smart relay named 'Channel Office L' at address '192.168.1.119'. Error: unknown, retrying in 0.5 seconds...
```

theyosh, 2023-10-08:
The system will restart when it takes to much time to update. Due to the kasa relay problems, there will be a 10 sec delay for very relay. This can case the updates to be to slow and it is trying to restart.

A thought with the bluetooth sensors, replace batteries. Here also, the max timeout is 10 seconds before terminating the reading. And this will be tried 3 times. So in worse case that can case a delay of almost 30 seconds, which will trigger other warnings and counters.

I do not know why this is happening. Did you reboot the Pi self? Something that will solve bluetooth issues. 

Because the last errors of `OSError: [Errno 113] Connect call failed ('192.168.1.119', 9999)` are looking more network related. But this is just a guess. And I do understand that you only updated the container, and this should not happening.
Could also be that docker network is causing issues. Again, that I cannot see from here.

It is hard to explain all docker commands to clear stuff. Google is your friend. A single way to clear all is `docker system prune -a` which should clear everything. Make sure the docker container is stopped.

threedom, 2023-10-08:
For the sake of trying, I did `docker system prune -a` and a manual TPI 4.7 install yesterday and imported the data files.
I believe my data files may be corrupted which may contribute to the issue. I am getting the above mentioned 404 error when I try to edit any sensor (no issue editing relays) and I still get a bunch of errors from the relays during updating.
I'll likely set up a fresh environment today and will not perform the DB import to avoid that issue.

threedom, 2023-10-09:
Closing since I did a complete reinstall.

closed 2023-10-09 by threedom

theyosh, 2023-10-09:
And after the re-install all was working again? That is strange. And sorry that it happened. It should not be the case.

threedom, 2023-10-09:
No worries - bugs are inevitable 😄 
I faced a few minor issues but now everything is running smoothly again. I couldn't reuse the existing DBs for whatever reason and even after the full reset the engine update takes more than the desired 30s but it doesn't seem to have a negative impact on stability.

It seems to me that the wireless sensors I am using are just not ideal for the setup so I excluded them in any logic and rely only on the wired sensors I have. Will probably put them on the ignore list going forward. 
Currently on docker 4.6.2 and will try updating to 4.7 again but I need a few days of smooth running before I want to deal with (potential) troubleshooting again.

Thanks a lot for all the help @theyosh!

Adding extract from logs below for context:

> 2023-10-08 23:33:07,444 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC temperature named 'Yuzu - Warm Side - Wireless' at address 'a4:c1:38:5c:68:44' with new value 88.86F in 0.01 seconds.
> 2023-10-08 23:33:07,325 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC humidity named 'LYWSD03MMC bluetooth sensor measuring humidity' at address 'a4:c1:38:5c:68:44' with new value 72.00% in 15.06 seconds.
> 2023-10-08 23:32:53,608 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named '4 - Growlights' at address '192.168.1.251,4' with new value 0.00 in 0.01 seconds.
> 2023-10-08 23:32:53,491 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named '3 - Waterfall' at address '192.168.1.251,3' with new value 0.00 in 0.01 seconds.
> 2023-10-08 23:32:53,375 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named '2 - Sun' at address '192.168.1.251,2' with new value 0.00 in 0.01 seconds.
> 2023-10-08 23:32:53,258 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named '1- RHP' at address '192.168.1.251,1' with new value 100.00 in 2.14 seconds.
> 2023-10-08 23:32:52,162 - INFO    - terrariumEngine       - Updated sensor 1wire temperature named 'Yuzu - Cold Side - Wired' at address '28-01204fc0d569' with new value 74.75F in 0.92 seconds.
> 2023-10-08 23:32:51,137 - INFO    - terrariumEngine       - Updated sensor 1wire temperature named 'Yuzu - Warm Side - Wired' at address '28-01204fba80d8' with new value 86.11F in 0.91 seconds.
> 2023-10-08 23:32:50,215 - INFO    - terrariumEngine       - Starting a new update round with 8 sensors, 4 relays, 0 buttons and 0 webcams.
> 2023-10-08 23:32:50,213 - WARNING - terrariumEngine       - Engine update took 59.80 seconds. That is 29.80 seconds short.
> 2023-10-08 23:32:50,128 - INFO    - terrariumEngine       - Updated Enclosure Z - Bottom Terrarium with 1 areas in 0.01 seconds.
> 2023-10-08 23:32:50,124 - INFO    - terrariumArea         - Updated area Lighting named Growlight in timer mode at enclosure Bottom Terrarium in 0.00 seconds.
> 2023-10-08 23:32:50,122 - INFO    - terrariumEngine       - Updated Enclosure Yuzu's Paradise with 4 areas in 0.03 seconds.
> 2023-10-08 23:32:50,116 - INFO    - terrariumArea         - Updated area Heating named RHP in sensors mode at enclosure Yuzu's Paradise in 0.01 seconds.
> 2023-10-08 23:32:50,108 - INFO    - terrariumArea         - Updated area Moisture named Waterfall in timer mode at enclosure Yuzu's Paradise in 0.00 seconds.
> 2023-10-08 23:32:50,107 - INFO    - terrariumArea         - Relays for Moisture named Waterfall period low are not switched on because we have to wait for 613.0 more seconds of the total settle time of 1710.0 seconds.
> 2023-10-08 23:32:50,106 - INFO    - terrariumArea         - Updated area Cooling named Cooling in sensors mode at enclosure Yuzu's Paradise in 0.01 seconds.
> 2023-10-08 23:32:50,098 - INFO    - terrariumArea         - Updated area Lighting named Main Lights in timer mode at enclosure Yuzu's Paradise in 0.00 seconds.
> 2023-10-08 23:32:49,958 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC temperature named 'Living Room' at address 'a4:c1:38:e2:f6:49' with new value 74.10F in 0.02 seconds.
> 2023-10-08 23:32:49,836 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC humidity named 'LYWSD03MMC bluetooth sensor measuring humidity' at address 'a4:c1:38:e2:f6:49' with new value 58.00% in 12.93 seconds.
> 2023-10-08 23:32:36,800 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC temperature named 'Yuzu - Cold Side - Wireless' at address 'a4:c1:38:8b:0b:31' with new value 78.53F in 0.01 seconds.
> 2023-10-08 23:32:36,680 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC humidity named 'LYWSD03MMC bluetooth sensor measuring humidity' at address 'a4:c1:38:8b:0b:31' with new value 98.00% in 28.17 seconds.
> 2023-10-08 23:32:20,755 - WARNING - hardware.sensor       - Invalid reading from sensor LYWSD03MMC bluetooth sensor humidity named 'LYWSD03MMC bluetooth sensor measuring humidity' at address 'a4:c1:38:8b:0b:31', retrying in 0.5 seconds...
> 2023-10-08 23:32:20,748 - ERROR   - hardware.sensor       - Sensor LYWSD03MMC bluetooth sensor humidity named 'LYWSD03MMC bluetooth sensor measuring humidity' at address 'a4:c1:38:8b:0b:31' has exception: Failed to connect to peripheral a4:c1:38:8b:0b:31, addr type: public
> 2023-10-08 23:32:08,403 - INFO    - terrariumEngine       - Updated sensor LYWSD03MMC temperature named 'Yuzu - Warm Side - Wireless' at address 'a4:c1:38:5c:68:44' with new value 88.70F in 0.01 seconds.
[... cut ...]
> 2023-10-08 23:31:52,401 - INFO    - terrariumEngine       - Updated sensor 1wire temperature named 'Yuzu - Cold Side - Wired' at address '28-01204fc0d569' with new value 74.86F in 0.92 seconds.
> 2023-10-08 23:31:51,379 - INFO    - terrariumEngine       - Updated sensor 1wire temperature named 'Yuzu - Warm Side - Wired' at address '28-01204fba80d8' with new value 85.89F in 0.95 seconds.
> 2023-10-08 23:31:50,411 - INFO    - terrariumEngine       - Starting a new update round with 8 sensors, 4 relays, 0 buttons and 0 webcams.
> 2023-10-08 23:31:50,410 - WARNING - terrariumEngine       - Engine update took 52.01 seconds. That is 22.01 seconds short.