theyosh/TerrariumPI #842: [FEATURE]: Toggle to enable/disable notifications in the "edit button" screen

chakachim, 2023-10-09:
**Is your feature request related to a problem? Please describe.**
I have started using the notification feature to alert me whenever the door is opened on the Terrarium. However it looks like there is no way (that I've found) to disable a button from notifying. As a result I get notified everytime the light sensor is triggered. 

**Describe the solution you'd like**
toggle button on the button edit screen that allows me to disable notifications.

**Additional context**
Add any other context or screenshots about the feature request here.

theyosh, 2023-10-17:
Yeah, I think this can be done. Need some time to add it.

theyosh, 2023-11-01:
Hi,

I added this option to sensor, relays and buttons. In order to test/use you need to update the code with `git pull` and rerun the installer. `sudo ./install.sh` as there new libraries being used for other new features.

After that, you can edit the button you want to disable from notification by disabling the notification setting.

theyosh, 2023-11-01:
Oh, and make user TerrariumPI is stopped before updating.

theyosh, 2023-11-13:
Any update on this? The new release is out, and this is released. So it should work. Could you confirm this?

chakachim, 2023-11-13:
Just updated and tested. Everything is working great. Thank you for adding
this feature!

theyosh, 2023-11-13:
Thanks for the feedback.

closed 2023-11-13 by theyosh