theyosh/TerrariumPI #844: [BUG]: Upgraded from v4.0.0 to v4.7.0 - traceback thrown, WebUI not reachable

duenni, 2023-10-11:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.7.0
 - Install: [Manual] 
 - Raspberry PI: 3B

**Describe the bug**
I upgraded from v4.0.0 to 4.7.0 before I did that I also updated all OS packages with apt. Install went smooth without error. After a reboot the WebUI is not reachable. When starting TerrariumPi in debug mode I get the following:
```
python terrariumPI.py 
2023-10-11 11:58:36,684 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.7.0 on a Raspberry Pi 3 Model B Rev 1.2 ...
2023-10-11 11:58:36,762 - INFO    - terrariumEngine       - Loaded 30 settings in 0.07 seconds.
2023-10-11 11:58:37,951 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-10-11 11:58:38,558 - INFO    - terrariumEngine       - Loaded sensor am2302 humidity named 'sensor1_humidity' at address '11' with value 76.20% in 0.59 seconds.
2023-10-11 11:58:38,564 - INFO    - terrariumEngine       - Loaded sensor am2302 temperature named 'sensor1_temp' at address '11' with value 21.00C in 0.00 seconds.
2023-10-11 11:58:39,105 - INFO    - terrariumEngine       - Loaded sensor am2302 humidity named 'sensor2_humidity' at address '18' with value 73.10% in 0.54 seconds.
2023-10-11 11:58:39,111 - INFO    - terrariumEngine       - Loaded sensor am2302 temperature named 'sensor2_temp' at address '18' with value 21.40C in 0.00 seconds.
2023-10-11 11:58:41,142 - INFO    - terrariumEngine       - Loaded sensor miflora temperature named 'MiFlora bluetooth sensor measuring temperature' at address 'c4:7c:8d:6a:98:d6' with value 21.40C in 2.03 seconds.
2023-10-11 11:58:41,150 - INFO    - terrariumEngine       - Loaded sensor miflora light named 'MiFlora bluetooth sensor measuring light' at address 'c4:7c:8d:6a:98:d6' with value 39.00lux in 0.01 seconds.
2023-10-11 11:58:41,156 - INFO    - terrariumEngine       - Loaded sensor miflora moisture named 'MiFlora bluetooth sensor measuring moisture' at address 'c4:7c:8d:6a:98:d6' with value 1.00% in 0.01 seconds.
2023-10-11 11:58:41,163 - INFO    - terrariumEngine       - Loaded sensor miflora fertility named 'MiFlora bluetooth sensor measuring fertility' at address 'c4:7c:8d:6a:98:d6' with value 0.00µS/cm in 0.01 seconds.
2023-10-11 11:58:41,168 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-10-11 11:58:50,372 - INFO    - terrariumEngine       - Loaded 8 sensors in 12.42 seconds.
2023-10-11 11:58:50,374 - INFO    - terrariumEngine       - Loading existing relays from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 137, in __init__
    self.__load_existing_relays()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 811, in __load_existing_relays
    relay.id, relay.hardware, relay.address, relay.name, callback=self.callback_relay
  File "/home/pi/TerrariumPI/hardware/relay/__init__.py", line 87, in __new__
    known_relays = terrariumRelay.available_hardware
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 47, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/pi/TerrariumPI/hardware/relay/__init__.py", line 63, in available_hardware
    imported_module = import_module("." + file.stem, package="{}".format(__name__))
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/TerrariumPI/hardware/relay/kasa_relay.py", line 11, in <module>
    from kasa import Discover, SmartStrip, SmartPlug
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/kasa/__init__.py", line 16, in <module>
    from kasa.discover import Discover
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/kasa/discover.py", line 9, in <module>
    from kasa.smartbulb import SmartBulb
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/kasa/smartbulb.py", line 9, in <module>
    from .modules import Antitheft, Cloud, Countdown, Emeter, Schedule, Time, Usage
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/kasa/modules/__init__.py", line 3, in <module>
    from .antitheft import Antitheft
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/kasa/modules/antitheft.py", line 2, in <module>
    from .rulemodule import RuleModule
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/kasa/modules/rulemodule.py", line 29, in <module>
    class Rule(BaseModel):
  File "pydantic/main.py", line 186, in pydantic.main.ModelMetaclass.__new__
TypeError: Argument 'bases' has incorrect type (expected list, got tuple)

```
I also noticed that the MOTD still says that 4.0.0 is installed. I rebooted several times. terrariumpi service is started.

**To Reproduce / Installation steps**
Upgrade from 4.0.0 to 4.7.0

[image]

[image]

theyosh, 2023-10-11:
Hmm, I thought this was fixed. But the quickest way is to do the following:

1. `sudo service terrariumpi stop`
2. `cd /home/pi/TerrariumPI`
3. `source venv/bin/activate`
4. `pip install pydantic==1.10.9`
5. `deactivate`
6. `sudo service terrariumpi start`

And this should fix it.

duenni, 2023-10-12:
That fixed it. Thank you, much appreciated. 😃

closed 2023-10-12 by duenni

threedom, 2023-10-17:
@theyosh is there a fix like this for the docker version? I have what seems to be the same issue with the docker version.

theyosh, 2023-10-17:
Just made new docker images. So a `docker pull` should pull a new image, and hopefully solve your issue.

threedom, 2023-10-17:
Did a pull of 4.7 (latest) and seems to be working right now.