theyosh/TerrariumPI #845: [FEATURE]: "All sensors" menu in version 4.7.0

duenni, 2023-10-12:
Hi,

in older versions of TerrariumPi one could set an entry to settings.cfg to have an "All sensors" overview menu.
Is it possible to get that menu back in v4.7.0? I searched a bit in the Settings and Wiki but couldn't find an option for it.
Thanks for considering!

Regards

theyosh, 2023-10-12:
You are right. It should be there, but I did not code it correctly. So I have updated the code and now it should work.

It is the last option 'All gauges on a single page' which should be toggled on. Not sure if a page reload is needed, but there should be a new menu item at the `home` section.

So, do a `git pull` and just reload your browser. Make sure cache is cleared. Of holding the shift key while reloading.

duenni, 2023-10-13:
Nice one! It's there now. Thanks!

closed 2023-10-13 by duenni