theyosh/TerrariumPI #846: [BUG]: I2C 4Channels LED AC dimmer doesn't save I2C address, only channel.

1ucket, 2023-10-15:
**Setup:**
 - TerrariumPI version: 4.7.0
 - Install: Manual
 - Raspberry PI: 4

**Describe the bug**
When changing the address value of the I2C 4Channels LED AC dimmer the IC2 part of the address will not be saved, only the channels before the comma.

**To Reproduce / Installation steps**
1. Add I2C 4Channels LED AC dimmer
2. Change address
3. Edit relais to verify
Thus the relais cannot be controlled, it will stay at 100% all the time.

**Expected behavior**
The I2C address is saved.

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):
When changing it:
[image]
When reopening the config:
[image]

1ucket, 2023-10-15:
I checked the post request body, and it is already missing the address. So this seems to be a front end issue.
{"hardware":"i2c_4ch-dimmer","address":"3","name":"Test","wattage":100,"flow":0,"calibration":{}}
I verified it with both Chrome and Firefox.

theyosh, 2023-10-15:
Hmm, could you try to do a `git pull` and reload the browser. There is a possibility that you have found a bug that is already fixed. By pulling the latest code it would solve it. If not, I have to look into the form.

1ucket, 2023-10-15:
> Hmm, could you try to do a `git pull` and reload the browser. There is a possibility that you have found a bug that is already fixed. By pulling the latest code it would solve it. If not, I have to look into the form.

I'm on the latest software (did a pull because of issue #840 and again today). So it seems to still be present.

closed 2023-10-16 by 1ucket